realsenseai/librealsense #7251: pyrealsense2 has no attribute 'context'

malapatiravi, 2020-09-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.06 | 
| Operating System & Version |   Ubuntu20 | 
| Kernel Version (Linux Only)    |   5.4.0-42                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.37.0                       | 
| Language                            | python                         | 
| Segment			|  Robot                          | 

### Issue Description
`import pyrealsense2 as rs
import numpy as np
import cv2

ctxt = rs.context()

dev_list = ctxt.query_devices()                                                                                                                                                                                     
n = len(dev_list)
print("number of devices : "+str(n))
`
The above program throws an error as follows
`Traceback (most recent call last):
  File "test.py", line 5, in <module>
    ctxt = rs.context()                                                                                                                                                                                        
AttributeError: module 'pyrealsense2' has no attribute 'context'`

The C/C++ application works without any problem. 
May i know what is the fix for this issue?

MartyG-RealSense, 2020-09-02:
Hi @malapatiravi   It is standard to use **ctx** instead of ctxt.  for example:

**ctx = rs.context()
dev_list = ctx.query_devices()**

malapatiravi, 2020-09-03:
@MartyG-RealSense I changed the code to ctx and i still get error and i think the issue is not with standard and something to do with how we compile if i am not wrong.

MartyG-RealSense, 2020-09-03:
I note that on your first line you have an apostrophe at the start of the line that should not be there:

`import pyrealsense2 as rs
^

Can you check if that is present in your script please and remove it if it is?

MartyG-RealSense, 2020-09-13:
Hi @malapatiravi  Do you require further assistance with this case, please?   Thanks!

malapatiravi, 2020-09-13:
@MartyG-RealSense None of the above solutions worked. Yes we need support. the "`"  is not part of code and its how it is pasted in the github. The issue still exist with librealsense SDK.

MartyG-RealSense, 2020-09-14:
I have researched this question extensively and not been able to find an obvious answer, unfortunately.  If you create a simple test script then this should establish if the attribute error is occurring because there is something in your pyrealsense2 wrapper installation that is preventing **rs** from being accessed.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

MartyG-RealSense, 2020-10-02:
Case closed due to no further comments received.

closed 2020-10-02 by MartyG-RealSense

MikeHallettUK, 2021-01-22:
Try "import pyrealsense2.pyrealsense2 as rs"

This works for me.

BiomechatronicsRookie, 2025-01-08:
Hi, I'm facing the same problem. 

I built pyrealsense from source on Jetson Orin Nano following [this](https://forums.developer.nvidia.com/t/connecting-intelrealsense-depth-camera-d455-to-jetson-orin-nano-8gb/296753/5) specifically the solution from DaneLLL on Jun 18th. 

I considered theadvice given on the official documentation for librealsense [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) where it is said that  you should follow a specific guideline to install the library on Jetson Orin. However, I am using an Intel Realsense D415 without an IMU, so I don't really need the drivers for it, as that would also downgrade my jetpack form 6.1 to 6.0 again.

Is there anyone facing the same issues?