realsenseai/librealsense #7252: “Frame didn't arrive within 15000” after long run with D435 or D435i

lipeize-pku, 2020-09-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435/D435i | 
| Firmware Version                    | 5.12.05.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)   |  4.9.140-tegra    | 
| Platform                                   | NVIDIA Jetson TX2 |
| SDK Version                            |  2.35.2                   | 
| Language                                | C++  | 
| Segment			          |  Robot  | 

### Issue Description

This is a follow-up of #7065. Memory leak in #7065 is solved. But camera still keeps cutting out.

I am using gstreamer to produce a video stream of RGB frames from D435/D435i, while some point cloud algorithms running in the background, but the stream keeps getting cut out after running for hours. 

This did not happen when I was using 2.13.0 version of the SDK, but only happens after I upgraded to 2.35.2.

### Trouble Shooting Attempts

Monitored RAM usage. Normal. Far from filling up when stream cuts out.

Monitored CPU usage. Normal when stream cuts out.

Used a cooling fan. Lowered temperature from 42 to 35 degree. Still cuts out.

Unable to get hardware reset to work properly. Camera cuts out a few frames after reset.

MartyG-RealSense, 2020-09-02:
Hi @lipeize-pku  I considered your new question very carefully and all of the possiilities for what the problem could be.

The SDK version may not be involved in this particular question.  If you were getting the 'frame didn't arrived' error almost immediately after beginning streaming then you could say that the version number may be involved.  However, SDK 2.35.2 apparently works fine at first, up until the point hours later when the RGB stops.

May I confirm whether it is only the RGB stream that stops?  Does the point cloud (presumably using a depth stream) continue processing after the 'frames didn't arrived within 15000' error appears?

lipeize-pku, 2020-09-03:
The whole programme would stuck at `wait_for_frames` and then exit. So RGB stream and point cloud both stops.

I just found out that the camera would show up in realsense-viewer as USB 2.1 after the programme crashes. Could it be that the camera switched from USB 3.2 to USB 2.1 during running, causing the programme to crash?

If so, how can I test this theory? For example, how can I get USB state of the camera in C++?

MartyG-RealSense, 2020-09-03:
If the camera went into USB 2.1 mode then some stream modes would become inaccessible because they are not supported on a USB 2.1 connection due to USB bandwidth limitations.  So I could see a potential problem if a program was using a particular resolution only available on USB 3 and then the connection was detected as USB 2.

In librealsense, you can access a text readout of the port's detected USB type using an instruction called **RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR **.  The type descriptor is accessible through the **rs-enumerate-devices** tool.

[image]

You should also be able to access it individually on its own by calling RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR, perhaps on a loop so you can continually see its status whilst your program is running.  There is an tutorial program called **FluentRealSense** that shows how to print out the various RS2_CAMERA_INFO statistics.

https://www.codeproject.com/Articles/1233892/FluentRealSense-The-First-Steps-to-a-Simpler-RealS

A list of the RS2_CAMERA_INFO categories can be found here:

https://intelrealsense.github.io/librealsense/doxygen/rs__sensor_8h.html#aa607d93c1bac40af3ec1c9b0345f59af

gaopeipeiok, 2020-09-04:
How can I force the USB to 3.0? I find it will change to 2.0 from 3.0 after my programme stoped by ctrl+c sometimes, then I plug and pull the camera several times, the USB cannot become 3.0. I have to reboot the TX2, it can connect to USB3.0.

MartyG-RealSense, 2020-09-04:
Hi @gaopeipeiok   Are you using the official short 1 meter USB cable that was supplied with the camera or a longer cable of your own choice, please?

You could also check whether inserting into the USB port with a quick, firm motion instead of a slow insertion improves the number of times that it is detected as USB 3.

gaopeipeiok, 2020-09-04:
I  use the official short 1 meter USB cable that was supplied with the camera。
I use normal speed to insert the USB，I don't feel slow。It was USB3.0 at first，and work fine for several hours, but after I restart my programme it change to 2.0 somtimes。

MartyG-RealSense, 2020-09-04:
@gaopeipeiok   If you are using a program that you have written yourself, you could try inserting a **hardware_reset()** instruction near the top of your script to reset the camera hardware when the program starts if you are not using this function already.   The link below has a C++ example.  If you are using another language such as Python, please let me know and I will find a compatible code example.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

gaopeipeiok, 2020-09-04:
Thanks for your reply,I will have a try now.

gaopeipeiok, 2020-09-08:
@MartyG-RealSense   My programme run 4days, and then cannot get frame from pipe_rs.wait_for_frames(500) without any error log,. After try 5 times, I stop the pipe by pipe.stop(),but the stop failed, my programme was choked.

MartyG-RealSense, 2020-09-08:
Hi @gaopeipeiok   Does your program use **set_real_time(false)** ?  This has been known to cause a freeze when stopping the pipeline, but does not occur when using set_real_time(true).  

If you are using set_real_time(false) and it is not practical to use (true), another approach may be to initiate a hardware_reset() instruction after 5 tries instead of using pipe.stop().

gaopeipeiok, 2020-09-08:
Thanks for your reply。
I donnot call set_real_time。
I will try to call hardware_reset instead stop。

hustwb, 2020-09-09:
Hi @lipeize-pku Do you solve this problem,I  have this problem too

gaopeipeiok, 2020-09-11:
@MartyG-RealSense  when I cannot get frame with the error "Frame didn't arrive within 500", I will retry 5 times, if cannot get frame too, I will call hardware_reset with the code "dev.hardware_reset(); dev = hub.wait_for_device();", but it failed with "failed to set power state" . If I call pipe.stop() before the dev.hardware_reset, my programme will be choked. Then I call exit(), top exit my programme,  it will be chocked too. Finally I have to call abort() to exit my programme.So I think the camera or the SDK is dead clocked.

MartyG-RealSense, 2020-09-11:
@gaopeipeiok   Is the program able to stop if you do not include the loop to keep retrying after you get an error, please?

gaopeipeiok, 2020-09-16:
@gaopeipeiok when I get the error log，I  did not retry get frame. And call pipe.stop()，my program choked too.

MartyG-RealSense, 2020-09-16:
@gaopeipeiok  As long as your program is able to keep retrying, I assume that there is processing occurring in the background and the program is not frozen.  The problem seems to occur when it stops retrying after a certain number of tries and then tries to carry out an action, such as a hardware_reset() or a close of the pipeline.

After 5 attempts have been made, you could try putting the CPU to sleep for a period first to give the CPU % usage a chance to settle down so that it isn't in a "busy" state. due to high activity.   Perhaps put the CPU to sleep for 5 seconds as a good test-value.  Then after 5 seconds, stop() the pipeline.  

An example of a sleep instruction is provided below:

**while(true) {
std::this_thread::sleep_for(std::chrono::milliseconds(10));
}**

gaopeipeiok, 2020-09-17:
@MartyG-RealSense  I didnot retry to get frame this time, and the problem occur too. I get the error log:"Frame didn't arrive within 2000",But I did not get the "stop success" log.My code is as the following :
try{
			data_rs = pipe_rs.wait_for_frames(2000);
		}catch(const rs2::error & e){
std::cout<< e.whate() << std::end;
std::cout<< "start stop pipe" << std::end;
pipe_rs.stop();
std::cout<< "start stop pipe success" << std::end;
}

MartyG-RealSense, 2020-09-17:
Going back to your original comment about using gstreamer ... I wonder if you could test with a simpler script that does not involve gstreamer and see whether the stream still goes wrong.  That may indicate whether gstremer is a factor in the problem.

gaopeipeiok, 2020-09-17:
I have try do not use gstreamer， it get the error too。

发自我的iPhone

MartyG-RealSense, 2020-09-17:
Are you using hardware sync of two cameras at the same time (D435 and D435i), with them joined by hardware sync cables so that one 'slave' camera syncs to a 'master' camera?  I ask this because if the hardware sync cable is long and does not have electrostatic discharge protection components built into it, if an ESD discharge occurs then it can cause the frame counters to reset and the RGB can freeze.  If you go to the link below and scroll upward a little, this phenomenon is explained in the multiple camera white paper.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-2-multi-camera-considerations

Also, is there a chance that the camera's view is being exposed to a flash of strong light such as a burst of sunlight?  This could cause the IR imager component to become saturated and no longer function correctly until the camera is reset, typically in past cases with a USB unplug-replug.

gaopeipeiok, 2020-09-18:
@MartyG-RealSense  There is only one camera. And it is not being  exposed to a flash of strong light such as a burst of sunlight.It is indoor. When I get the error "Frame didn't arrive within 2000", I did not reset the camera nor unplug-replug, It can be normal after I abort my program and restart it.

gaopeipeiok, 2020-09-18:
@MartyG-RealSense  Can you tell me the possible reason why I get the error. is it  the  bug of firmware or the SDK?

MartyG-RealSense, 2020-09-18:
There are a wide range of reasons why **frames didn't arrive** time-out errors may be generated, which is why the process of asking questions is important in order to narrow down the number of possible causes.  I appreciate that it can be a frustrating process.

Does the error always occur around the same time after start or at different hours (5 hours, 11 hours, 16 hours, etc).

Information on setting SDK logging can be found here:

https://dev.intelrealsense.com/docs/troubleshooting#section-enable-lib-real-sense-log

gaopeipeiok, 2020-09-18:
It can run more than 16 hours now, the longest time is more than 4 days.

MartyG-RealSense, 2020-09-18:
4 days tends to be a typical maximum running time of a long-run session before a problem occurs, if problems do occur.  So that is consistent with past cases.  

This is not necessarily due to a problem with the camera hardware though, as the 400 Series cameras can run indefinitely so long as their operating temperature remains within recommended tolerances.  When a problem does occur after that time period, it is more likely to be related to a problem in the USB cable or USB port, the computer hardware or the computer operating system.

gaopeipeiok, 2020-09-18:
But why the call pipe.stop(), my program will be  choked?

MartyG-RealSense, 2020-09-18:
Reading your very first comment again, I would just add that if you are cooling your camera by blowing air directly into its ventilation grille, you should not be doing this.  By all means direct cool air in the general direction of the casing from a distance to keep it cool.  But forcing air in through an electronic device's ventilation grille at close range disturbs the air flow inside the device, a flow is designed in a way that the hot air is exhausted out through the grille.

Also, if the overall CPU usage is showing as normal on your computer operating system's memory monitor, it does not necessarily mean that the librealsense program has not crashed, but that the rest of the computer and whatever programs are running in the background are still working okay and using the CPU.  More accurate information comes from monitoring the status of an individial program's processes.  For example, if a program shows frequently changing **disk activity** and then those values become near zero, that can indicate that the program has frozen but not crashed and exited.

If your program can run for between 16 hours and 4 days and you are not observing memory leaks, the chances are high that the program code itself is okay and there is a problem elsewhere in the USB or the computer hardware / software.

Do you have a different computer that you can try the program with?

gaopeipeiok, 2020-09-18:
I did not use fan  to cool it these days.
I have try four Tx2 and one windows computer. 
I have try the realsense-viewer and my program.

MartyG-RealSense, 2020-09-18:
I see, you joined this case later and were not its creator - lipeize-pku created it - so the fan and gstreamer comments do not apply to your situation.   I do apologise for the confusion in previous comments.  

Durng further research, I located a case on the Nvidia forum about TX2 where a freeze was occurring with D435 after a certain amount of time when the program had run fine up until then.

https://forums.developer.nvidia.com/t/realsense-d435-with-jetson-tx2-depth-stop-working-freezes-after-time/83911/8

The user in that case specified what worked for them at the end of the case.

https://forums.developer.nvidia.com/t/realsense-d435-with-jetson-tx2-depth-stop-working-freezes-after-time/83911/10

gaopeipeiok, 2020-09-19:
@MartyG-RealSense  Thanks for your help. But I use Jetpack 4.3 now which is newer than 4.2.2. And my issue seems  not like that user.

MartyG-RealSense, 2020-09-19:
@gaopeipeiok   During research I came across a couple of cases where the camera was stopped by stopping the sensor instead of the pipeline.  So they did pipeline.start() and **sensor.stop()**.

There is an example script in the link below that uses sensor.stop():

https://github.com/IntelRealSense/librealsense/issues/4024#issuecomment-494603782

Another mention of a successful use of this method is here:

https://github.com/IntelRealSense/librealsense/issues/6459#issuecomment-648795212

In another example, sensor.stop() and then sensor.close() is used to shut the stream off.

https://github.com/IntelRealSense/librealsense/issues/4158#issuecomment-505559462

MartyG-RealSense, 2020-09-22:
@gaopeipeiok   Though each case is individual, it is worth mentioning the case of another RealSense user who was experiencing the **Frames didn't arrived within 15000** error after a 24 hour period at the same time that you were experiencing your problem, and solved it by reinstalling the latest SDK (2.38.1 at the time of writing this).

https://github.com/IntelRealSense/librealsense/issues/7288

gaopeipeiok, 2020-09-23:
Does the latest SDK support Tx2? When I use apt update, did not find the version 2.38.1

MartyG-RealSense, 2020-09-23:
My understanding is that the Jetson package updates may lag behind the release date of a new SDK because the Jetson packages have to be built by Intel manually instead of generated automatically.

gaopeipeiok, 2020-09-27:
@MartyG-RealSense  after I the "Frame didn't arrive within 2000", I sleep 10 seconds then I get the following error:
uvc streamer watchdog triggered on endpoint: 130
 uvc streamer watchdog triggered on endpoint: 132
 uvc streamer watchdog triggered on endpoint: 131

MartyG-RealSense, 2020-09-27:
@gaopeipeiok   There was a past case with the **uvc streamer watchdog triggered on endpoint** error occurring with a TX2 and D435i.  It happened on their Tx2 but worked fine on a colleague's TX2.  They found that the difference between the boards was that his TX2 had the factory-default settings and his colleague had performed OS and kernel updates.  So they flashed their TX2 board and then it worked.

https://github.com/IntelRealSense/realsense-ros/issues/1015#issuecomment-564411882

gaopeipeiok, 2020-09-27:
@MartyG-RealSense Do you mean I need to update my kernel，But my program can run some hours and my OS version is ubuntu 18.04 and kernel 4.9.140， not like the issue of 1015.

MartyG-RealSense, 2020-09-27:
@gaopeipeiok   I reviewed your case again carefully from the beginning but could find no obvious leads about how to progress.  @hustwb - mentioned earlier in this case - solved their particular long-run problem with this error with an SDK update.

https://github.com/IntelRealSense/librealsense/issues/7288#issuecomment-696457483

@hustwb  Could you describe for @gaopeipeiok please the SDK update method that you used that led to your success please?   Thanks!

MartyG-RealSense, 2020-10-05:
Hi @gaopeipeiok   Do you still require assistance with this case, please?   Thanks!

gaopeipeiok, 2020-10-09:
@MartyG-RealSense Thanks for your help，I update the SDK and firmware to the latest. My program has run 12 days. I think the issue is solved. Thanks again!

MartyG-RealSense, 2020-10-09:
That's great news @gaopeipeiok  - thanks for the update   :)

closed 2020-10-09 by MartyG-RealSense