realsenseai/librealsense #7255: How can i cut from rgb or depth frames  to  create pointcloud？

syly666, 2020-09-02:
I means cut from rgb or depth frames firstly,
then create  pointcloud  instead of "create pointcloud firstly  ,then cut ".

Because  only a few points need to be dealt     if  cut firstly  .

MartyG-RealSense, 2020-09-02:
Hi @syly666  A depth image can be made more sparse (more black gaps) if the value of the **Laser Power** option is reduced, though it does not give you any control over the parts of the image that have their detail reduced.

You can also apply a 'depth clamp' to the data to control how far into the background the camera can observe.  The **rs-align-advanced** example program (which aligns depth with color) demonstrates this, with a slider that progressively removes and brings back the background as the slider is increased and decreased. 

https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced

On the depth image alone (not aligned with color), you can also set a minimum and maximum observable distance for the camera using a post-processing filter called a **Threshold Filter** (for example, setting a minimum distance of 0.7 meters and a maximum of 3 meters, and excluding all depth data outside of that range.  You can set the filter either in the **Post Processing** section of the RealSense Viewer's options side-panel or with scripting.

syly666, 2020-09-02:
I mean : if rgb frame is 1920*1080  ,it can be created  2,073,600 points . 
If i  crop it to 500*300 ，it  only can be created 150000 points.
So how can i crop rgb or depth firstly  ，then create  pointclouds？
I mean from
[image]
to
[image]

MartyG-RealSense, 2020-09-02:
You can crop the XY planes of an image by defining a 'bounding box' that excludes data from the image that it outside of the box.  A comment in the link below discusses the subject of defining a bounding box.

https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

syly666, 2020-09-02:
Thank you for your answer，but  it is not crop rgb or depth frame firstly ，then create pointclouds.
it just creat pointcloud, then crop pointcloud.
[image]

MartyG-RealSense, 2020-09-02:
Bounding boxes are not just for point clouds.   they can be defined for 2D images too.

https://github.com/IntelRealSense/librealsense/issues/2769

https://support.intelrealsense.com/hc/en-us/community/posts/360033272013/comments/360008045254

syly666, 2020-09-02:
[image]
Thank you for your reply.
So it must " pc.calculate(depth_frame)"  firstly,then  crop?
I want to crop rgb frame (this rgb frame has been aligned to depth frame)firstly，then " pc.calculate(depth_frame)" . 
Because  " pc.calculate(depth_frame)"  firstly will create too many points.

MartyG-RealSense, 2020-09-02:
Bounding box programming is a subject that is outside of my experience, unfortunately.  I will seek guidance from another RealSense team member.

@dorodnic   Do you have some advice that you could provide about bounding box creation regarding the question above, please?   Thanks!

MartyG-RealSense, 2020-09-09:
@dorodnic Do you have some advice that you could provide about bounding box creation regarding the question above, please? Thanks!

MartyG-RealSense, 2020-09-18:
I apologise that you did not receive further comments about your question @syly666  - did you manage to resolve it in the meantime, please?

MartyG-RealSense, 2020-09-26:
Case closed due to no further comments received.

closed 2020-09-26 by MartyG-RealSense

syly666, 2020-09-26:
//At present, I only generate the point cloud near the ROI region of the image by this method
//Where  "float minx,float miny,float minz,float maxx,float maxy,float maxz"  is the three-dimensional size of the image ROI
The three-dimensional size of the image ROI
typedef pcl::PointCloud<RGB_Cloud> point_cloud;
typedef point_cloud::Ptr cloud_pointer;
cloud_pointer PCL_Conversion(const rs2::points& points, const rs2::video_frame& color,float minx,float miny,float minz,float maxx,float maxy,float maxz) {

	// Object Declaration (Point Cloud)
	cloud_pointer cloud(new point_cloud);

	// Declare Tuple for RGB value Storage (<t0>, <t1>, <t2>)
	std::tuple<uint8_t, uint8_t, uint8_t> RGB_Color;

	//================================
	// PCL Cloud Object Configuration
	//================================
	// Convert data captured from Realsense camera to Point Cloud
	auto sp = points.get_profile().as<rs2::video_stream_profile>();

	cloud->width = static_cast<uint32_t>(sp.width());
	cloud->height = static_cast<uint32_t>(sp.height());
	//cloud->width = static_cast<uint32_t>(sp.width())*static_cast<uint32_t>(sp.height());
	//cloud->height = 1;
	cloud->is_dense = false;
	cloud->points.resize(points.size());

	auto Texture_Coord = points.get_texture_coordinates();
	auto Vertex = points.get_vertices();

	// Iterating through all points and setting XYZ coordinates
	// and RGB values
	for (int i = 0; i < points.size(); i++)
	{
		//===================================
		// Mapping Depth Coordinates
		// - Depth data stored as XYZ values
		//===================================
		if (Vertex[i].x<minx || Vertex[i].x>maxx || Vertex[i].y > maxy || Vertex[i].y < miny || Vertex[i].z<minz || Vertex[i].z>maxz) continue;
		cloud->points[i].x = Vertex[i].x;
		cloud->points[i].y = Vertex[i].y;
		cloud->points[i].z = Vertex[i].z;

		// Obtain color texture for specific point
		RGB_Color = RGB_Texture(color, Texture_Coord[i]);

		// Mapping Color (BGR due to Camera Model)
		cloud->points[i].r = get<2>(RGB_Color); // Reference tuple<2>
		cloud->points[i].g = get<1>(RGB_Color); // Reference tuple<1>
		cloud->points[i].b = get<0>(RGB_Color); // Reference tuple<0>

	}

	return cloud; // PCL RGB Point Cloud generated
}

syly666, 2020-09-26:
@MartyG-RealSense

MartyG-RealSense, 2020-09-26:
Is the code above your solution please @syly666  or do you need further assistance with this case?

syly666, 2020-09-26:
> 请您了解您的解决方案@syly666，或者您是否需要有关此案例的进一步帮助？

Thank you very much， I think my problem has been solved.

MartyG-RealSense, 2020-09-26:
Thanks very much for the confirmation!