realsenseai/librealsense #7258: T265 Known Issues and Robustness

cimbar, 2020-09-03:
Hi,
There are 3 main known issues about T265 and it stays there at each new release.

**Known Issues**
#4261 - [T265] Add ability to open multiple devices from different processes.
#4518 – [T265] Pose data produces NaNs. Can still occur in some cases. If detected, please attempt to make a raw data (images + IMU) recording using the recorder tool, and attach a link to it in the github issue, to assist our resolution.
[T265][Mac] - Start after stop is not working on Mac with the T265 camera

Is realsense developer team working on these known issues? It looks that there is not much interest on fixing T265 issues. This camera developed for robotics application but it is still not possible to develop robust system with this camera since it was released. Users always face an issue about connection, stability or accuracy. I wonder that is there a roadmap about these known issues and will there be major update in soon?

mpottinger, 2020-09-03:
For me it is accuracy and relocalization that is the problem. In a handheld application it has nowhere near as good tracking stability and recovery as ARCore and ARKit despite having two fisheye cameras dedicated to tracking.

stevemartinov, 2020-09-03:
I gave up on this camera while ago, this is an absolute piece of crap. It constantly gave me wrong readings (usually -0.5m offset within 10 meter range) and relocalization/pose_jumping simply do not work (even though it can jump to a correct position but all previous nodes are not optimized and therefore right after the jump, if you move backwards to n-1 node, it will jump back again! What a crap!)

mpottinger, 2020-09-04:
> Iand relocalization/pose_jumping simply do not work

Yes this is a problem for me. It does not work like on mobile phone AR SDK's, where if it loses tracking it jumps back to the correct pose relatively quickly. If the T265 goes off, it doesn't recover very easily/quickly at all, sometimes not at all. Meanwhile I could wave my ARCore phone around like crazy and pose will recover extremely quickly and reliably.

msadowski, 2020-09-14:
According to the reply I've received [here](https://github.com/IntelRealSense/librealsense/issues/4518#issuecomment-691242254) it doesn't seem like there will be any work done on T265, which is a pity as the idea behind this sensor was so interesting!

RealSenseSupport, 2020-09-15:
T265 is not discontinued or retired, and is available. We have moved our focus to our next generation of products and consequently, we will not be addressing issues/new features in the T265.

arjunsengupta98, 2020-09-21:
> According to the reply I've received [here](https://github.com/IntelRealSense/librealsense/issues/4518#issuecomment-691242254) it doesn't seem like there will be any work done on T265, which is a pity as the idea behind this sensor was so interesting!

Agreed :(

lishanggui, 2020-09-23:
Hi  @RealSenseSupport  my problem at https://github.com/IntelRealSense/librealsense/issues/7199.  can any people help me？ thanks！

rlabs-oss, 2020-12-21:
> T265 is not discontinued or retired, and is available. We have moved our focus to our next generation of products and consequently, we will not be addressing issues/new features in the T265.

Translation/Understanding: It was abandoned. 

Moving on: 

What alternative products (that are supported) are available for translation and IMU pose output ? 

For my T265 rs-enumerate-devices returns "No device detected" - yet the rs-pose example works fine.

closed 2026-10-23 by MartyG-RealSense