realsenseai/librealsense #7260: D455 IR images are darker?

rwchu, 2020-09-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.06 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0.2034                       | 

### Issue Description
I've attached snapshots taken from the RealSense Viewer of the same environment with both the D435i and D455. The D455 image is noticeably dimmer than the D435i. Both are Y8 IR images and were taken with exposure = 8500, gain = 16, and laser power = 150. 

I'm not sure why the D455 is dimmer when (to my knowledge) they have the same imagers. Please advise?

D435i:
[image]

D455:
[image]

agrunnet, 2020-09-03:
The D455 is actually a color sensor. It will appear dimmer than the d435.

rwchu, 2020-09-10:
Thanks for your reply. Is there an updated datasheet with all the D450 module specs? The latest one I can find is from June 2020, which is missing some specs.

RealSenseCustomerSupport, 2020-09-14:
----------------------------------------------

Hi

Currently the datasheet of the June version is the latest one. Usually it doesn't tell the details about the spec of the sensors.

Thanks

** Please be informed that we will close this ticket in seven (7) days if we have not heard back from you. You're encouraged to close the ticket if you think the issue/problem is solved.

rwchu, 2020-09-15:
Hi @agrunnet, could you provide any more info on the difference between the IR sensors for the 455 and 435? I always believed they were the same OV9782 sensor. Is there an additional filter added for the 455? The datasheet is not very clear on this. 

- Ryan

agrunnet, 2020-09-15:
D435 uses monochrome OV9282
D455 uses color OV9782.

closed 2020-09-15 by agrunnet