realsenseai/librealsense #7262: Cannot have color overlay on 3D pointcloud

carstenpanch, 2020-09-04:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0210969|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003abef98ac|
|**Firmware Update Id**|00000003abef98ac|

If importing a file (bag) with everything enabled - depth + RGB + motion. Then nothing shows up on the 3D  point cloud. Only if you disable RGB camera.

ev-mp, 2020-09-06:
@carstenpanch, there is a tracked issue on RGB with OpenVINO plugin.
Try to switch off RGB post-processing to reenable texture mapping in 3D.
Duplicate of #7175, #6992

carstenpanch, 2020-09-07:
It didn't work, and also there is no OpenVINO face detection that can be switched on/off in the RealSense viewer.
Below is the console output I just found - maybe that helps? These 5 lines loop for each frame:

```
ros_file_format.h:87 - Failed to convert source: rgb8 to matching rs2_format
ros_file_format.h:65 - Failed to convert source: Raw Frame Size to matching enum rs2_frame_metadata_value
ros_file_format.h:87 - Failed to convert source: mono16 to matching rs2_format
ros_file_format.h:65 - Failed to convert source: Raw Frame Size to matching enum rs2_frame_metadata_value
ros_file_format.h:87 - Failed to convert source: 8UC1 to matching rs2_format
```

ev-mp, 2020-09-07:
@carstenpanch , thanks for the feedback - apparently it wasn't useful.
Can you share a part of the bag file?

RealSenseSupport, 2020-09-18:
Hi @carstenpanch 

Is there anyway you can provide bag file for this one?  Have you also tried the latest FW and LRS releases?

*If we don’t hear from you in 7 days, this issue will be closed.

carstenpanch, 2020-09-23:
Hi,
Sorry for the delay. I have placed a link to the bag here: [Link](https://we.tl/t-3HwU3HruU3)
With the lateste viewer 2.38.1, if I drag the file into the viewer, the window is empty - but if I then choose some depth texture, then it shows up.

RealSenseSupport, 2020-10-02:
Hi @carstenpanch 

The link to the bag file wasn't available.  Can you share again?  Also we just released 2.39.0, maybe worth a try with this latest.

*If we don’t hear from you in 7 days, this issue will be closed.

carstenpanch, 2020-10-05:
Hi,
I have made it available again here: [https://we.tl/t-P911gFW4VV](https://we.tl/t-P911gFW4VV)
I tried the bag with the 2.39.0 software - that didn't change anything. But maybe you meant that I should try and record a new sequence with the new 2.39.0 software?

Best regards,
Carsten

RealSenseSupport, 2020-10-07:
Hi @carstenpanch 

Thanks for the download again.  And yes I was meaning for you to record a new sequence with the new 2.39.0 release.

closed 2021-02-15 by RealSenseSupport