realsenseai/librealsense #7263: Curious why rs-post-processing uses 2 ques and polls for frames?

arothenberg, 2020-09-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description

I'm just curious what rs-post-processing gains from a separate thread queuing the original frame and the filtered frame? I'm glad the code is there as an example of realsense enqueue but I'm missing something about the case use. It's probably obvious.

MartyG-RealSense, 2020-09-05:
Hi @arothenberg   A section in Intel's post-processing documentation explains why there the original data is preserved:

"The filters preserve the original data and always generate a new (filtered) frame to pass on.  The re-generation of new frames allows to share frame among different consumers (thread) without the risk of the data being overwritten by another user".

https://dev.intelrealsense.com/docs/post-processing-filters#section-design-and-implementation

dorodnic, 2020-09-06:
Hi @arothenberg 
There is no specific reason to do so. You can absolutely run post-processing on the same thread, similar to how you apply align and other filters. We simply wanted to introduce more advanced concepts in different examples. 
Doing processing on a background thread is a frequent question, and post-processing example is sufficiently advanced to demo it, without being too complicated. At least that was the thinking.

arothenberg, 2020-09-07:
Thanks dorodonic. I was definitely happy to see the enqueue code included. I'll be using that. 
If you enqueued the  original before processing the filter maybe it could show (in the view) the fps difference 
between original and filtered.  I'll test myself it but does filtering take significant time?

Marty, I'll close this in a few days. Thanks.

arothenberg, 2020-09-07:
Thanks again. On reflection separate threads for filtered and original would be tricky since wait for frames is blocking. I'd need to queue frames outside the thread and allow the threads to poll the frames but not release the frame from the queue until both 
filtered and original were processed on the frame.  

Can the enqueue cause too much memory use if the frames are not released from the queue? I know this might be a dumb question.

closed 2020-09-07 by arothenberg