realsenseai/librealsense #7264: Discrepancy between raw depth frame values when multiplied by units and getDistance function

blakat360, 2020-09-06:
I'm trying to understand the way the raw depth values are stored. This is in order for me to be able to bind the depth texture to a shader and then later query it for distance values.

I ran this test script but after a few pixels, the assert starts failing

```c++
    auto [width, height] = cam.getResolution();

    rs2::depth_frame frame = cam.getDepthFrame();
    auto units = frame.get_units();
    auto arr = reinterpret_cast<const uint16_t*>(frame.get_data());

    for (int row = 0; row < height; row++) {
        for (int col = 0; col < width; col++) {
            std::cerr << frame.get_distance(row, col) << " " << arr[row + row * col] << '\n';
            assert(frame.get_distance(row, col) == arr[row + row * col] * units);
        }
    }
```

The asserts are always off by ~1cm. Is this because get distance linearises the depth values before converting them to a distance? If so, how do I get the near and far planes of the realsense camera?

MartyG-RealSense, 2020-09-06:
Hi @blakat360   The original depth pixel value is the **uint16_t** value.  the pixel value is then multiplied by the **depth unit scale** to convert the pixel value to meters.  

https://github.com/IntelRealSense/librealsense/issues/2348#issuecomment-418152047

blakat360, 2020-09-06:
I am taking it as a uint_16, note: ```auto arr = reinterpret_cast<const uint16_t*>(frame.get_data());```
I've tested and depth scale is the same as the units given by querying ```rs2::depth_frame::get_units()```

I'm confused as to why cumulative error does not stay at 0. I'm assuming I'm indexing into the image incorrectly but can't seem to come up with something that works.

MartyG-RealSense, 2020-09-06:
RealSense 400 Series cameras have a phenomenon called **RMS Error** where the amount of error in measurements increases over distance.  Error is less than 1% of the distance of the object from the camera.  So for example, if the camera is 1 meter away then an accuracy of 2.5 mm to 5 mm is expected.

blakat360, 2020-09-06:
I understand that - I'm wondering why I get a difference in measurement when I pull the distance out of the raw data using 2 different methods.

MartyG-RealSense, 2020-09-06:
I am trying to understand your script, so thank you for your patience.  It looks as though you are calculating the distance in two ways - through uint16 and through frame.get_distance() - and subtracting one from the other.  And that you expect that if both methods read the same distance then the result should be zero (e.g 1.5 - 1.5 = 0).  

Is my interpretation correct, please?

blakat360, 2020-09-06:
That is correct!

This line;
```auto arr = reinterpret_cast<const uint16_t*>(frame.get_data());```
gets a pointer to the start of the depth frame (through ```frame.get_data()```) and casts this to an array of unsigned 16 bit integers (raw depth values). I also grab the amount I need to multiply these by to get distances using ```rs2::depth_frame::get_units()```.

I then loop over the entire frame using the nested for loops and attempt to calculate the distance by 
a) calling rs2::depth_frame::get_distance
b)  getting the raw value, and multiplying that by units myself

I expect the value to be the same. However, it is not. This leads me to believe I'm indexing into the frame the wrong way.

I have tested get_depth_scale (through querying the device and then accessing its streams) and in this case it is the same as ```get_units``` (0.001). This leads me to believe that this is not the problem. Furthermore, the errors are uncorrelated in each iteration of the loop are uncorrelated which further suggests that the multiplier is not the issue.

ev-mp, 2020-09-06:
@blakat360 , can you take a second look into the code?
```c++
for (int row = 0; row < height; row++) {
        for (int col = 0; col < width; col++) {
            cumulative_error += std::abs(frame.get_distance(col, row) - (double)arr[col + row * col] * units);
        }
    }
```

<details>
  <summary>Spoiler</summary>
  
>arr[col + row * ~~col~~ _**width**_] * units
  
</details>

blakat360, 2020-09-06:
I'm an idiot, thank you @ev-mp!

closed 2020-09-06 by blakat360