realsenseai/librealsense #7266: L515 not detected in realsense-viewer v2.38.1 with OS Ubuntu 18.04

gauvainrobert, 2020-09-06:
Hello,

I bought an L515 recently. The camera is detected by my OS Ubuntu 18.04 but when I am running realsense-viewer (v 2.38.1) the camera is not detected. Before this camera, I was using a D435 camera and it was working perfectly.

`dmesg -wT` returns

> [dim. sept.  6 17:50:27 2020] usb 1-3: new high-speed USB device number 43 using xhci_hcd
> [dim. sept.  6 17:50:28 2020] usb 1-3: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
> [dim. sept.  6 17:50:28 2020] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [dim. sept.  6 17:50:28 2020] usb 1-3: Product: Intel(R) RealSense(TM) 515
> [dim. sept.  6 17:50:28 2020] usb 1-3: Manufacturer: Intel(R) RealSense(TM) Camera
> [dim. sept.  6 17:50:28 2020] usb 1-3: SerialNumber: 00000000F0090415

The thing is that I use a usb3.0 cable and a usb3.0 port. As I said, it was working with the previous camera D435.

I tried some things :
    

- using cmake with : 

        cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DFORCE_LIBUVC=ON`
        make -j2
        then use ./tools/enumerate-devices/rs-enumerate-devices but it didn't work

- using

`./scripts/setup_udev_rules.sh` 

    didn't work too

    
    

| Required Info                         |                                            |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    | 5.4.0-42-generic                  | 
| Platform                                 | PC  |
| SDK Version                            | 2.0                           |

ev-mp, 2020-09-06:
@gauvainrobert, the root-cause is that the camera is recognized as USB2 device only.
>[dim. sept. 6 17:50:27 2020] usb 1-3: new high-speed USB device number 43 using xhci_hcd

Make sure that the USB3 cable is inserted firmly into USB3 port - slow plugin may cause device to be enumerated as USB2. Other factors are loose cable/connector and/or insufficient power.
Duplicate of #7224

francisc0garcia, 2020-09-14:
Hi,

I am currently having the same issue. I also have the the same setup: L515 -  [ubuntu 18.04] - kernel [5.4.0-47].

I have already tested at least 5 different combination of USB3 cables and PCs without success.

With some cables I got the device detected as USB2 device (just like the original post), and with other cables it does not recognize the camera at all (meaning: [dmesg] does not show anything - and windows 10  [device manager] does not report new hardware ).

The camera worked properly the first week, and suddenly the problem appeared.

RealSenseSupport, 2020-09-14:
Hi @gauvainrobert - Did your situation change based on the information @ev-mp provided?

Hi @francisc0garcia - For L515, we do not support USB2 at the moment.  You've mentioned that you've attempted multiple USB3 cables and I'm assuming these are certified cables.  You have also tried and verified that the cables are seated properly into the device and into the host system.  

Can you let us know what FW you have on the camera?  When you setup on Win 10 system, is there any time where you're able to have the camera recognized and able to run RealSense Viewer to stream the camera streams?

You may need to look into RMA the camera.

francisc0garcia, 2020-09-15:
Hi @RealSenseSupport, thanks for your answer,  Yes, I have verified the cables and proper connections. Sadly, I can not get the FW of the camera (camera is not recognized) and the only feedback that the camera returns is:

```
[ 1532.586663] usb 1-4: new high-speed USB device number 16 using xhci_hcd
[ 1532.739351] usb 1-4: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[ 1532.739354] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1532.739356] usb 1-4: Product: Intel(R) RealSense(TM) 515
[ 1532.739358] usb 1-4: Manufacturer: Intel(R) RealSense(TM) Camera
[ 1532.739359] usb 1-4: SerialNumber: 00000000F0190480
```

Regarding Win10, no, the camera is not detected now on Windows, but i did not test it at the beginning when the camera was working on ubuntu (first week). 

Thanks, I will look into RMA the camera.

RealSenseSupport, 2020-09-30:
Hi @gauvainrobert @francisc0garcia  

Assuming everything with regards to insertion, cables, power, etc is taken care of and still not being recognized, then RMA may need to be looked into.

Let's close the ticket until RMA is handled if you so choose.

*If we don’t hear from you in 7 days, this issue will be closed.

alexlitz, 2020-09-30:
I believe [this code block](https://github.com/IntelRealSense/librealsense/blob/0adceb9dc6fce63c348346e1aef1b63c052a1db9/src/l500/l500-factory.cpp#L128) is likely the problem, commenting it out resolved this issue for me.

RealSenseSupport, 2020-10-07:
Thanks @alexlitz for the information.

closed 2020-10-07 by RealSenseSupport