realsenseai/librealsense #7267: Cannot get the exposure time of color frame

Kevinlaw95, 2020-09-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |05.12.05.00 | 
| Operating System & Version |    Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  SDK 2.0}                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
how can i get the exposure time of the color frame? I want to calibrate the color camera response function, so I need to  get the exposure time of the camera in real time. This is my code:

` rs2::pipeline pipe;
  rs2::config pipe_config;
  pipe_config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
  pipe_config.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
  rs2::pipeline_profile profile = pipe.start(pipe_config);
  rs2::frameset data;
  rs2::frame depth, color;
  while (ros::ok())
  {
    data = pipe.wait_for_frames();
    color = data.get_color_frame();
    if(color.supports_frame_metadata(RS2_FRAME_METADATA_ACTUAL_EXPOSURE)){
      exposure_time = color.get_frame_metadata(RS2_FRAME_METADATA_ACTUAL_EXPOSURE); //us
      std::cout<<"expose time: "<<exposure_time<<"us"<<std::endl;
    }
  }
`

I find that if I use "set_option(RS2_OPTION_EXPOSURE,exposure_time)" to set the exposure time and i can get the exposure time from the code above. If not, I can't get. So how I can get the real time exposure time when the camera is in auto exposure mode?
Thanks!

MartyG-RealSense, 2020-09-07:
Hi @Kevinlaw95   I hope that the link below will provide useful guidance about how to get the actual exposure values when using auto-exposure.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

ev-mp, 2020-09-07:
@Kevinlaw95 , you cannot read the actual exposure for RGB sensor when Auto-Exposure is turned on.
It is a device limitation.
Duplicate of #5596.

Kevinlaw95, 2020-09-08:
I tried to get exposure time by metadata. Maybe because of device limitation I could't get it. Thank you for your help!@ev-mp and @MartyG-RealSense .

MartyG-RealSense, 2020-09-08:
Thank you @Kevinlaw95 - do you need any further assistance, please?

Kevinlaw95, 2020-09-09:
Hello @MartyG-RealSense , today I tried to reinstall the realsense-SDK through the source [code.](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) . I find that the first step make my system get the latest  stable kernel(5.4.0-42-generic). And some error always occurs at this step of patch-arch.sh. 
But I got the exposure time through the above code, although the exposure time is always a constant value. 

So, I want to know which version of the kernel or which version of the realsense I can use to obtain the automatic exposure time. Maybe I can solve this problem by changing a system! Thanks!

MartyG-RealSense, 2020-09-09:
Hi @Kevinlaw95  I considered your question very carefully, but @ev-mp will likely be able to provide a more accurate response on this particular subject.  If the RGB actual exposure metadata value cannot be updated due to actual exposure readings only being supported for the depth sensor - as https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920 asserts - then I do not see a way to work around that to get an actual-value color exposure update.

closed 2020-09-29 by MartyG-RealSense