realsenseai/librealsense #7268: SR300 depth camera image gets distorted

GilVolovich, 2020-09-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300} | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.36                           | 
| Language                            |  C++                           | 
| Segment			|  Robot                          | 

Hi,

While streaming depth camera video, every few seconds the image gets distorted. Due to this issue, when i am trying to extract ply from a distorted frame, the ply is corrupted.
[image]

1. Why might be the root cause for this issue ?
2. Is there any way to detect a corrupted image and retry the ply extraction?

Thanks,
Gil

MartyG-RealSense, 2020-09-07:
Hi @GilVolovich  I ran extensive tests in the Viewer with an SR300 but was unable to replicate what you experienced.

Though the Viewer's log in your kindly provided image shows the warning that 'the user was holding onto frames after the stream stopped', this does not seem to be relevant to your problem as I found that this warning occurred on my own computer whenever the SR300 stream was deactivated at any FPS speed.

Could you try going into the Viewer's 2D mode please using the '2D' button in the top corner of the Viewer and start the depth stream.  Then left-click on the 'i' info icon at the top of the depth stream's panel to display real-time information about the stream, including its FPS.  This should tell you whether a constant FPS is being maintained or whether it is falling and rising in sync with the changes in the image.

[image]
 
If you just need one good frame in your ply, you could click on the **Pause** icon at the top of the center panel in 3D point cloud mode to pause the live-stream on a good frame, and then click the Export button to export that good frame.  

[image]

On Windows, if the Microsoft **3D Viewer** application is installed (it can be got from the Microsoft Store for free if it is not), then once the ply file is created then you can double-click on the file to open it in Microsoft's 3D Viewer app and compare it to the original point cloud frame in the RealSense Viewer.

[image]

GilVolovich, 2020-09-07:
Hi MartyG-RealSense,

Thank you for the quick and detailed response!
I checked what you asked for - The FPS seems stable around 29.94.
In regards to your suggestion for pausing on a specific frame, i need to capture the frame from c++ code, without any human interaction. Any suggestions?

Thanks,
Gil

MartyG-RealSense, 2020-09-07:
I can see the difficulty of your situation.  If you need to export without human involvement then you want to have confidence that the ply is going to be consistently good quality so that whatever is captured as a ply, there is a good chance of it being a high quality frame.

Before getting into C++, can you perform one more test with your 3D mode please?  Activate the RGB stream as well as the depth stream.  then go to the **Texture** option at the top of the center panel and select from its drop down menu **Intel RealSense SR300 Color**.  This should map color to the point cloud instead of using depth as the texture source.  Does the point cloud behave more stably when rendering it that way?

[image]

GilVolovich, 2020-09-07:
Hi,

No, its still unstable, exactly like when i am using depth as the texture source.

Thanks,
Gil

MartyG-RealSense, 2020-09-07:
Ok, thank you for trying.  Does it still fluctuate if you leave the camera stationary (e.g leave it on a table and do not move it around with the hand)

GilVolovich, 2020-09-07:
hi,

The camera is attached to a robotic arm.
i will describe the flow, maybe it will help:
1. Turn on the camera
2. Move the camera (attached to the robotic arm) to a predefined position. Stop movement (Camera is now stationary).
3. Extract ply.
4. repeat 2-3, for 4 more predefined points.
5. Turn off camera.

Thanks,
Gil

MartyG-RealSense, 2020-09-07:
Thanks very much for the very useful information.   In the animated image in your first comment, it looks as though the camera is moving around by small amounts.  Is this showing the robot arm moving between the 5 points?

If so, I wonder if the changes in the image are occurring because the distance between the camera and the observed object is changing by small amounts as the camera moves.  The SR300's depth sensing range is around 1.5 meters (compared to the 10 meters on the 400 Series and L515 models), so it is easier for objects to pass out of sensing range with the SR300 when the camera is being moved.

GilVolovich, 2020-09-08:
Hi,

The camera is not moving while the image is captured. specifically the animated image i've uploaded was not taken from the actual flow, but from the RealSense viewer - The camera was stationary at this time.

Thanks,
Gil

MartyG-RealSense, 2020-09-08:
Can you confirm please that you have no other RealSense cameras located near the SR300 (such as mounted on the robot arm above or below the SR300).  Multiple SR300s intefere with each other if their pattern projections overlap, and 400 Series cameras can interfere with the IR of other cameras (but do not experience interference themselves).

GilVolovich, 2020-09-08:
Hi,

There might be a camera 3-4 meters away attached to other robots. It might be SR300 or F200.

Thanks,
Gil

GilVolovich, 2020-09-13:
Hi Marty,

Can you update? do you need to try anything else?

Thanks,
Gil

MartyG-RealSense, 2020-09-13:
I re-researched your case but was not able to find much new information.

Did you determine whether or not nearby SR300 cameras on your other robots might be influencing the image quality, please?

Does the observed object have a reflective surface with a light source cast on it?  I wondered whether the breaks in the image might be related to a pattern of light reflections on the object.

GilVolovich, 2020-09-21:
hi Marty,

i checked what you asked for. there is no reflection of light and no nearby cameras.
Let's assume there is a HW problem in the camera itself. How can identify a bad image?

Thanks,
Gil

MartyG-RealSense, 2020-09-21:
The SR300 did not have a calibrator tool created for it like the one that the R200 stereo camera from the same generation had.  There was though a user-created Windows utility called **IVCAM** for adjusting laser parameters.  It was originally designed for the SR300's predecessor the F200 but because the F200 and SR300 work very similarly, SR300 users were able to use it too.  It had to be set to **Windows 8** in its Compatibility Mode settings though.  And it is not a RealSense SDK tool, so its main usefulness is in testing the hardware as any changes made will not affect a RealSense SDK program.

https://software.intel.com/es-es/comment/1862567

To set the Compatibility Mode, you right-click on the launch icon, select **Properties** from the menu and then select the **Compatibility** tab of the Properties window.

[image]

GilVolovich, 2020-09-21:
Hi Marty,

Sorry, i don't understand how this is answering my question.
I want to know if its possible to identify a bad image in real time, then perform another scan.

Thanks,
Gil

MartyG-RealSense, 2020-09-21:
It was suggested as a means of testing whether there was a potential hardware problem, outside of the RealSense SDK software (thus eliminating the SDK as a cause of the problem if the same problem manifested in IVCAM).

There is the option of checking the camera in the **Depth Quality Tool**, though I did not suggest it because essentially you seem to have a good depth image in the moments when it is not fluctuating and so the Depth Quality Tool may not provide useful feedback.

Another way to test for a bad image would be to perform a check on the depth value of the pixels.  If, for example, there are more than 50% of pixels in the image that have a depth value of zero then take this as an indicator of a bad frame and take the scan again.

GilVolovich, 2020-09-22:
hi Marty,

i am looking at the frame data, most i get ~60%-70% of zero values, on good frames as well...
Is that may be because of the filters? 

thanks,
Gil

GilVolovich, 2020-09-22:
sorry, the filters are only performed after i check the frame data... but i can see that there are more zeros when the object is far from the camera. When i reach ~40cm there is more then 50% loss.

MartyG-RealSense, 2020-09-22:
I would expect more of the pixels in the image to become zero as the camera moves further away from the observed scene and the camera can no longer read depth detail from that part of the scene (bearing in mind the ~1.5 meter depth sensing range of the SR300).  Small objects may be especially vulnerable to becoming unreadable before the full 1.5 meter range is reached.

 There was an old 3D scanning trick with that generation of cameras where placing a "horizon" (e.g a white board background) behind the object could help the camera to 'lock on' to detail in the scene.  Placing objects around the main object could also help with lock-on, though this may not be practical for your project if you only want to capture the main object.

MartyG-RealSense, 2020-10-02:
Hi @GilVolovich  Do you still require support with this case, please?

closed 2020-10-12 by MartyG-RealSense