realsenseai/librealsense #7270: downgrading to Realsense SDK 2.36/2.37

padhupradheep, 2020-09-07:
I downloaded Realsense SDK 2.38 and now I want to downgrade to 2.37, as the official support for ROS package is 2.37. I tried to do the installation from the source, but when I run the step 

`./scripts/patch-realsense-ubuntu-lts.sh`

I get an error stating that 

> ubuntu-xenial-hwe folderf atal: unable to connect to kernel.ubuntu.com:kernel.ubuntu.com[0: 91.189.94.216]: errno=Connection timed out

What needs to be done?

ev-mp, 2020-09-07:
@padhupradheep , please check that you're not behind a firewall that blocks ubuntu.com ports.
When running the script on a corporate network you need to make sure to set http and https proxy servers

padhupradheep, 2020-09-07:
I do not think I'm behind a firewall, in the file I mentioned above I changed it from git to http and still it is not working. Maybe some additional material would help here!

ev-mp, 2020-09-07:
@padhupradheep , I just checked the script on Ubuntu 16 and both the git://kernel.ubuntu.com/ and its IP (91.189.94.216) are fully functional. So it is definitely a host-side issue that you need to investigate.

There was a similar issue reported a long time ago that can be reviewed as well #1630.

padhupradheep, 2020-09-07:
Thanks! You are right! The problem is with the network! :)

closed 2020-09-07 by padhupradheep

padhupradheep, 2020-09-08:
"Clone
--
git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial
git+ssh://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial
https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial"

Maybe these are other links that could be helpful for people who will have a similar issue to get it done. 

Source: https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/xenial