realsenseai/librealsense #7271: Intel Realsense

lrobinson746, 2020-09-07:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

MartyG-RealSense, 2020-09-07:
Hi @lrobinson746  Your computer's processor may be of an age that makes it incompatible with these tools since SDK 2.36.0 onward.  You can resolve this problem by installing standalone versions of the tools instead of using the ones that are installed by the Windows SDK installer.  

For example, to download the standalone Viewer for SDK 2.38.1, you can go to that version's **Assets** list on the SDK Releases web-page and download the file called **Intel.RealSense.Viewer.exe**.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

[image]

Further details can be found in the link below:

https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292

MartyG-RealSense, 2020-09-18:
Case closed due to solution provided and no further comments received.

closed 2020-09-18 by MartyG-RealSense