realsenseai/librealsense #7273: The use of align_to_depth.process in example, makes the picture very stuttered. How to solve it.

hebowu, 2020-09-08:
When running rs-align, the screen is very stuttered. I know through the test that it is the statement align_to_depth.process (frameset) that causes stutter in the main loop. How can I solve it?
Looking forward to reply.

dorodnic, 2020-09-08:
Hi @hebowu 
What platform and camera are you using? I assume by stutter you mean framerate drop. Are you using pre-built SDK or building from source?

hebowu, 2020-09-08:
Hi @dorodnic  Thank you very much for your reply.
My computer system is win10,CPU is Intel Xeon Gold6134,GPU is 1080Ti. 
I built it from the source. This problem was found in example rs-align after CMake compiled the source code.

hebowu, 2020-09-08:
Hi @dorodnic
I solved this problem myself, because the realsense2d.lib, I linked earlier, in fact, I should have linked to realsense2.lib.
Thank you very much for your help.

dorodnic, 2020-09-08:
Glad you found a solution. This was the next thing I was about to ask. Its easy to mix up since CMake default is Debug.

RealSenseCustomerSupport, 2020-09-17:
----------------------------------------------

Hi @hebowu, Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-10-05 by RealSenseCustomerSupport