realsenseai/librealsense #7274: How do color and depth images align

sunbin1357, 2020-09-08:
How do color image and depth image align using realsense D435i？

MartyG-RealSense, 2020-09-08:
Hi @sunbin1357  If you need to know about both **depth to color** alignment and about **color to depth** alignment, Intel's white-paper document about projection, texture mapping and occlusion provides a useful reference in the link below.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-3-4-stream-alignment

The opening section of the documentation for the **Align** sample program provides further useful information.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align#overview

MartyG-RealSense, 2020-09-18:
Hi @sunbin1357   Did the information above answer your question pleae?  Thanks!

closed 2020-09-26 by MartyG-RealSense