realsenseai/librealsense #7275: D435 RAW16 to RGB8 problem

zpwork, 2020-09-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                          | 
| Segment			|  Robot                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

When using opencv python to convert raw16 data to rgb8,  I did that with the following code:

         rawframes = pipeline.wait_for_frames()
         raw_frame = rawframes.get_color_frame()
         imgRAW = np.asanyarray(raw_frame.get_data())
         imgRAW = np.uint8( imgRAW / 256.0)
         imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BAYER_GR2BGR)

but the output image is greener and darker than the images I took directly in RGB color space.

Any suggestion for converting the RAW16 data (uint16 bayer patern) to RGB8? Thanks in advance.

MartyG-RealSense, 2020-09-08:
Hi @zpwork   Is the converted image closer to your RGB images if you instead use **COLOR_BGR2RGB** instead of COLOR_BAYER_GR2BGR.  For example:

 imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BGR2RGB)

zpwork, 2020-09-09:
> Hi @zpwork   Is the converted image closer to your RGB images if you instead use COLOR_BGR2RGB instead of COLOR_BAYER_GR2BGR.  For example:
> imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BGR2RGB)

Thabk you for help.
But, RAW16 data is bayer patterned image which is packed in uint16 (h, w) numpy array. It means, in RAW16 image, each pixel has only 1 channel of color(R or G or B). while BGR or RGB data will be (h, w, 3) numpy array which contains 3 channel of color in each pixels. 
When you use cv2.COLOR_BGR2RGB, it requires (h, w, 3) data as input. It will throw an arg error.

MartyG-RealSense, 2020-09-09:
The OpenCV documentation link below has a list of Bayer color codes to try at the bottom of the page in the section headed **Bayer > RGB**

https://docs.opencv.org/3.4/de/d25/imgproc_color_conversions.html

zpwork, 2020-09-10:
I have tried all 8 parameters at the bottom of the page, but the color is still greener and darker than it should be.
[image]

zpwork, 2020-09-10:
I have tried all 8 parameters at the bottom of the page you mentioned, but nothing's changed, still darker and greener than it should be.
[image]

MartyG-RealSense, 2020-09-10:
I am not sure what to suggest.  The link below describes the complexities of CV color conversions:

https://github.com/opencv/opencv/issues/13066

If you are accessing a 16-bit image, I was wondering why you are using uint8 instead of **uint16**.

closed 2020-09-26 by MartyG-RealSense