realsenseai/librealsense #7280: GL CallBack Error after running realsense-viewer

arwindersing, 2020-09-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         

| Camera Model                       |  D435 | 
| Kernel Version (Linux Only)    |  Linux Raspberrypi 5.4.51-v7 |                                         
| Platform                                 | Raspberry Pi 3  |
| SDK Version                            | 2.38.1             | 

### Issue Description
1) I went through this installation guide mentioned here https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md & after running realsense-viewer in terminal I get error
     **GL CALLBACK: ** GL ERROR ** type = 0x826c, severity = 0x9146 , message = GL_INVALID_VALUE in glTexImage2D( invalid width=4096 or height=512 or depth=1)** .

2) My Realsense Viewer looks like this
[image]
[image]
[image]

I am really stuck at this for couple of days now & don't know what to do anymore, **please HELP.**

dorodnic, 2020-09-08:
If you are able to navigate to Settings (rightmost icon, second to last menu item) please try reducing Font Samples slider (the first slider)
[image]
The error indicate there is not sufficient GPU memory to load the font atlas.

arwindersing, 2020-09-08:
> If you are able to navigate to Settings (rightmost icon, second to last menu item) please try reducing Font Samples slider (the first slider)
> [image]
> The error indicate there is not sufficient GPU memory to load the font atlas.

Hi @dorodnic , Thanks for your prompt reply, I tried the solution which you suggested , navigated and reduce the size but that didn't help.

dorodnic, 2020-09-08:
Font atlas was last increased in 2.36, so 2.35.2 should work. This only a function of GPU memory, so all camera examples should work correctly. RPi4, if viable, does not have this problem and has USB3 support.

arwindersing, 2020-09-09:
How do i go back and install a previous version (e.g 2.35 ) of Real Sense . Is there a git hub repo for that? I want to try installing the older version and see if that helps.

arwindersing, 2020-09-09:
Hi @dorodnic  I was able to run the capture & other examples successfully, but I do have one issue my Pi freezes after 10 seconds irreseptive of any example I run . Is it related to high CPU utilization or the power supplied to Pi itself ( I am supplying power from Laptop USB port).

closed 2020-09-09 by arwindersing