realsenseai/librealsense #7281: L515 Depth Image Rotated

bae9875, 2020-09-08:
[image]
[image]

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 |
| Operating System & Version |   Ubuntu 18.04  | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1  | 

### Issue Description
Hi,

I'm using L515 to acquire both depth image and color image. The color image comes out fine, but the equivalent depth from aligned_depth_to_color comes out rotated 90 degrees clockwise. Also, whenever the cable is agitated during acquisition, the whole screen on my laptop rotates randomly. Could that be related to this issue? Thanks.

jinbae9875, 2020-09-09:
I fixed the laptop screen rotating by running "sudo apt-get remove iio-sensor-proxy". It's strange that it only happened to L515 and not Kinect or D435. Anyways, I still need help with getting the depth image right. Thanks.

jinbae9875, 2020-10-01:
I also fixed the rotation issue with aligned_depth_to_color. I deleted some lines that relate to enabling IR and apparently that results in the picture being rotated. Weird, isn't it?

closed 2020-10-01 by jinbae9875