realsenseai/librealsense #7282: setup_udev_rules.sh

scottrpaterson, 2020-09-09:
Camera Model                -      D435i
Operating System & Version - Ubuntu 18.04

### Issue Description
I'm trying to install on Ubuntu per instructions (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). 

According to docs I'm suppose to run 'setup_udev_rules.sh' so that my camera can have the correct permissions. However this file no longer exists anywhere.

scottrpaterson, 2020-09-09:
Found the file - https://github.com/IntelRealSense/librealsense/blob/1bc211c147e536c725ebe76a1c8b221dd8062a83/scripts/setup_udev_rules.sh it seems to exist in the main librealsense SDK.

It is not mentioned anywhere in the ROS package that this needs to be install first!

MartyG-RealSense, 2020-09-09:
Hi @scottrpaterson   In the **Method 2** installation instructions on the front page of the RealSense ROS wrapper's GitHub, it suggests in Step 1 to install the RealSense SDK 2.0 (librealsense). 

https://github.com/IntelRealSense/realsense-ros#installation-instructions

It offers the choice of doing so with a Debian package or by building from source code.  For those who prefer the latter option, it provides a link to the installation.md instructions for building the SDK from source.  

This is directing the reader to go to the installation.md page and install librealsense from source before moving on to Step 2 of the ROS wrapper installation instructions for Method 2, which is to secondly build the ROS wrapper once librealsense installation is completed.

So the direction of flow of the Method 2 instructions is that librealsense should be installed first (Step 1) before moving on to installing the ROS wrapper (Step 2)..

MartyG-RealSense, 2020-09-18:
Hi @scottrpaterson   Do you require further assistance with this case, please?   Thanks!

scottrpaterson, 2020-09-21:
I don't, thank you.

closed 2020-09-21 by scottrpaterson