realsenseai/librealsense #7283: D455 RuntimeError: No device connected

malith1992, 2020-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 | PC |
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I got a new realsense D455 camera. It works perfectly on Realsense viewer. But it gives me 

> Traceback (most recent call last):
>   File "C:/Users/UH5190/PycharmProjects/untitled/opencv_viewer_example.py", line 19, in <module>
>     pipeline.start(config)
> RuntimeError: No device connected
> 

error when I run this example file [https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py](url)

sam598, 2020-09-09:
Same issue with the D455 and Python on a Raspberry Pi 4. No device connected.

Latest firmware on the D455 and built from the latest source today.

MartyG-RealSense, 2020-09-09:
Hi @malith1992  and @sam598   Do either of you own another RealSense 400 Series model such as D415 or D435 / D435i that you can test the same Python wrapper installation with, please?  This would indicate whether the problem was likely to be related to the D455 camera specifically if the wrapper that you have already built works with the other camera models.

sam598, 2020-09-09:
I’ve tested a D435 with the latest firmware and the script runs fine.

MartyG-RealSense, 2020-09-09:
@sam598   Okay, thank you very much for the information.  I will seek advice from Intel about this issue.

malith1992, 2020-09-10:
I have tested the script with D435i camera and the script runs fine.

RealSenseCustomerSupport, 2020-09-11:
----------------------------------------------

Hi @malith1992
Please check if your pyrealsense2 package is updated to support D455 as well
and python.exe  able to find it and use it (either you build it from source or use pip to install it)
the quickest way it use pip to install it
you can use
>> pip install pyrealsense2         #install it
>> pip install -U pyrealsense2   :#to update it to the latest version which is v2.38.1.2225 if install it before
>> pip list # to check the pyrealsense 2 version

C:\Users\xxxx>pip list
Package          Version
----------------- -----------
opencv-python    4.4.0.42
pip              20.2.2
pylint            2.6.0
pyrealsense2      2.38.1.2225
setuptools        41.2.0

sam598, 2020-09-11:
@RealSenseCustomerSupport this solution works for me on Windows, but Linux raspberrypi still does not connect. There are more details in this issue: https://github.com/IntelRealSense/librealsense/issues/7312

RealSenseSupport, 2020-09-14:
Great to know the solution works for windows,
we will look into the #7312 since there may different issue cause this issue on Raspberry Pi 4

closed 2020-09-14 by RealSenseSupport

malith1992, 2020-09-17:
> Hi @malith1992
> Please check if your pyrealsense2 package is updated to support D455 as well
> and python.exe able to find it and use it (either you build it from source or use pip to install it)
> the [...]

I have updated the pyrealsense package, but the issue remain same in Windows.

RealSenseCustomerSupport, 2020-09-18:
----------------------------------------------

Hi @malith1992

1. The latest wrapper and SDK should be able to support D455 on Win10 PC
since I am able to run the opencv_viewer_example.py example with D455 and Realsense SDK2.8,1 wrapper and library on Win10 PC

is there only one version of Python installed on your Win10 PC?

Under different version of Python folder, all of them have a site_packages folder that PIP install the pyrealsense under it,  please ensure the python version you used has the pyrealsense updated

`site-packages` is by default part of the Python _search path, need to ensure the pyrealsensexxx.pyd under that folder is updated to the one that support D455_

if you only have one python installed,  the pip list should show the version is 2.38.1.2225
>>pip list
Package Version
opencv-python 4.4.0.42
pip 20.2.2
pylint 2.6.0
pyrealsense2 2.38.1.2225

2. before running the script, you can use device manager to check if the D455 is enumerated or not

RealSenseCustomerSupport, 2020-09-18:
----------------------------------------------

for your information in case you have different version of python installed

below link has way to ensure you install the package under right python version
https://stackoverflow.com/questions/2812520/dealing-with-multiple-python-versions-and-pip

RealSenseSupport, 2020-10-05:
Dear @malith1992 
have you able to run the python with D455 now?

malith1992, 2020-10-05:
It's now working on Windows. But I have an issue with librealsense 2.38  update with Jetson Nano. I have opened a new issue for that.

tanakorn-kongyok, 2020-12-05:
i have intelrealsense d455 and How can i use skeleton tracking by not buy code of intelreasense

MartyG-RealSense, 2020-12-05:
Hi @tanakorn-kongyok   Do you mean that you want to do skeleton tracking with RealSense without buying commercial software such as **Nuitrack SDK** please?

If you are interested in Nuitrack SDK for skeleton tracking then you can find out more about it at the links below.  It has a free trial version and Nuitrack recently added support for D455 and L515.

https://nuitrack.com/
https://nuitrack.com/blog

If you would prefer to create your own skeleton tracking system then Intel have published a YouTube seminar about using the 400 Series cameras to train a body joint tracking data-set:

https://www.youtube.com/watch?v=VSHDyUXSNqY

Alternatively, if you are able to use ROS then a RealSense compatible skeleton tracking system suc as **OpenPose** may be helpful.

https://github.com/ravijo/ros_openpose