realsenseai/librealsense #7284: BUILD_DLIB_EXAMPLES Enabled: fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory

MBSadeghzadeh, 2020-09-09:
I am going to build librealsense version 2.38.1 with cmake version 3.18.0 with BUILD_DLIB_EXAMPLES enabled in Windows 10 but by cmake configuration I get this error:
**_RealSense\v2.38.1\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory_**
[CMakeError.log](https://github.com/IntelRealSense/librealsense/files/5193417/CMakeError.log)

Dlib version is 19.21 and my compiler is Visual Studio 2015.
I am sure that it is due to BUILD_DLIB_EXAMPLES, because when it is disabled, build process is doing successfully.

MartyG-RealSense, 2020-09-09:
Hi @MBSadeghzadeh   My research into your question indicates that this pthread.h error message is just a CMake configuration error that has occurred in numerous different software packages (not just librealsense) and is not a guide to where the problem is actually occurring.

In CMake, have you pointed **DLIB_DIR** to the location of your extracted dlib folder as point 3 of the dblib wrapper's Windows instructions requests, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib#windows

The instructions for Linux suggest that this directing of DLIB_DIR can also be done within the CMake instruction.  using the **-DDLIB_DIR** flag and pointing it to the folder location on your computer where dblib was extracted to.

MBSadeghzadeh, 2020-09-09:
Thank you @MartyG-RealSense. 
You guessed right. I pointed DLIB_DIR to the build directory of DLIB_DIR where _include_ and _lib_ folder exist.
But why should we point DLIB_DIR to the source root folder, not compiled(binary) directory? for example for openCV or PCL example I do later and it works!!

MartyG-RealSense, 2020-09-09:
I will direct your question about why DLIB_DIR points to the dlib wrapper developer @maloel

MartyG-RealSense, 2020-09-18:
Hi @maloel  Do you have any advice that you can kindly offer about the DLIB_DIR related question above, please?   Thanks!

MartyG-RealSense, 2020-09-26:
Hi @MBSadeghzadeh   Apologies that you did not receive a response.  I will try to get some assistance for your case.  Thank you very much for your patience.

RealSenseSupport, 2020-09-28:
The wrapper example helps to compile the dlib library as part of the whole libRealSense build hence DLIB_DIR is the path to the source of dlib. Please let us know if you have any other question. 
Thank you.

[image]

MBSadeghzadeh, 2020-10-03:
@MartyG-RealSense,  thank you so much.

closed 2020-10-03 by MBSadeghzadeh

MartyG-RealSense, 2020-10-03:
You are very welcome @MBSadeghzadeh  - thanks very much for the update, and thanks to @RealSenseSupport too for the assistance!