realsenseai/librealsense #7285: D415 depth stream error when disable auto-exposure

WeixiangZhang, 2020-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415                                                      | 
| Firmware Version                   |  5.12.7.100                                              |  
| Operating System & Version |  Win10                                                    | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 |  PC                                                          |
| SDK Version                            |  v2.38.1                                                   | 
| Language                            |  C/C++                                                       | 
| Segment			|  others                           | 

### Issue Description
I meet the same issue with [#2412](https://github.com/IntelRealSense/librealsense/issues/2412), and easy to reproduce.

Here is part of my code to disable auto-exposure of Color Sensor after pipeline start.
`
			if (sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE)) {
				std::cout << "sensor support RS2_OPTION_ENABLE_AUTO_EXPOSURE" << std::endl;
				sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);

				if (sensor.supports(RS2_OPTION_GAIN)) {
					sensor.set_option(RS2_OPTION_GAIN, mGain);
				}
				if (sensor.supports(RS2_OPTION_EXPOSURE)) {
					sensor.set_option(RS2_OPTION_EXPOSURE, mExposure);
				}
			}`

Run this code, things happen

1. Almost every time, RC IC2 Config error received.
1. With high probability, Depth stream start failure received
1. After several times, depth streams is abnormal. Depth Frame in RealSense Viewer is almost black. And from IR frame, we can see one of IR frame is not right which causing depth frame error. When this happens, only way to solve is hardware reset or unplug camera.

[image]

I test four D415 cameras, it happens always. And I'm sure it is not related to USB3.0 cable or operation temperature.

MartyG-RealSense, 2020-09-09:
Hi @WeixiangZhang   I would not rule out the temperature completely.  Your ASIC temperature is 67 degrees C.  I would expect late 20's to early 30's with the settings that you are using, unless there is a high environmental temperature outside of the camera (e.g hot weather) that raises the temperature of the camera's casing.  

[image]

In the data sheet document for the 400 Series cameras, the official recommended maximum temperature when powered is 35 degrees C, though it may be more like 42 in practical real-world conditions before problems with the image begin manifesting with temperatures in excess of that level.  If the ASIC is 67 then it is definitely too hot.  I would strongly recommend trying to decrease the temperature first so that it can be excluded as a cause of your problems.

WeixiangZhang, 2020-09-10:
@MartyG-RealSense Thank you for reply. But my test shows it is not a temperature issue. When it happens, ASIC temp is 33 and projector temp is 28.
[image]
[image]

MartyG-RealSense, 2020-09-10:
I re-read this case from the start to get some new insights.  There are some points that I would like some further information about, please.

Is the script that you provide in your first comment a new librealsense program that you have written, or have you modified the code of the RealSense Viewer program?

If it is two separate programs: please bear in mind that whilst it is possible for more than one application to access the camera at the same time, two programs cannot access the same stream type on the same camera.  For example, one program could access depth and one program could access IR, but both programs cannot access depth on the same camera at the same time.

Also, the stereo and color functions have their own separate depth and RGB auto-exposure.  When enabling / disabling them, the code for accessing each type of auto-exposure is almost the same but they have different index numbers.  Index '0' in **sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0)** refers to the **depth** auto-exposure.

https://github.com/IntelRealSense/librealsense/issues/3558#issuecomment-549405382

though you are writing your own librealsense script and running it, and then looking at what is happening in the RealSense Viewer program after the script is run.  Is that correct, please?

WeixiangZhang, 2020-09-11:
@MartyG-RealSense Firstly, I don't modify RealSense Viewer, that script is my own code. And there is only one program to access camera at the same time. 

Secondly, I change my code following #3558, the issue happens again.
[image]

And at last,  I can use RealSense Viewer (release on Github, no modification) to reproduce this issue.
1. Set both depth and color to 640x480 @ 60Hz. (This is resolution my program need, not necessary to reproduce this issue)
2. Disable auto-exposure of color sensor and keep auto-exposure of depth enabled.
3. Click to on both color and depth and after frame displayed click to off. Repeat several times, the issue happens.

[image]
[image]

MartyG-RealSense, 2020-09-11:
I followed your test method in the RealSense Viewer with a D415.  I was able to replicate the **synthetic stream** and the **exception was thrown during user processing callback** errors that you are experiencing when you perform the test, though not the RT IC2 error or the image corruption.

Given that the errors are only being generated when the mode is changed and not generating continuously, they are likely not a significant factor in what is occurring with your streams and are probably just 'hiccups' during the change of settings.  FPS speeds look normal in the images that display the speeds, even when the images are corrupted.

 If you start the Viewer without the camera plugged in, do you find that the RT IC2 errors occur in the Viewer's log as soon as you plug in the camera?  Or do they occur later once you have started and stopped the streams a few times?

WeixiangZhang, 2020-09-12:
@MartyG-RealSense 
For **RT IC2 Config error**,  it does not happen when start the Viewer without the camera plugged in and then plug in the camera. But as soon as I run a program like the **rs-show example**, stop it and then open RealSense Viewer the **RT IC2 Config error** happens. In my test, **RT IC2 Config error** seems not to cause the depth stream error.

For depth stream error, my test shows it is caused by **disable auto-exposure**. If I keep the auto-exposure enabled as default, depth stream error never happens (at least in my test). Another test  program which just gets and shows color frame and depth frame can prove this:
1. start pipeline with color and depth streams enabled, and do nothing to auto-exposure option, keep it as default.
2. after several frame received (e.g. 120 frames), disable the auto-exposure of color stream

Run this program several times, the depth stream error happens, depth frame almost black. Most importantly, this error happens at the exact time point when disable the auto-exposure of color stream.

 If color sensor and depth sensor are separate, why disable auto-exposure of color stream can cause depth stream error...

MartyG-RealSense, 2020-09-12:
I did further extensive testing with your test technique and was not able to recreate it on a USB 3 connction no matter what I did.  

However, I was able to trigger the RT IC2 Config error pop-up quite quickly when repeating the test on a USB 2 connection.   I was able to repeat it again in subsequent tests on USB 2.   However, I was only able to trigger it if I repeatedly clicked on the stream on-off toggle without pause, which does not reflect how the streams would be used in a real-world setting (it is something that some RealSense users do in "stress tests" during program testing).

I was not able to exactly replicate your test results of the error occurring as soon as the color auot-exposure was disabled.  With the depth stream with auto-exposure enabled, I was able to trigger it by slowly toggling the color on and off a few times (not toggling it rapidly).  I was not able to trigger it simply by tirning color auto-exposure off.

Ultimately, aside from demonstrating that it is possible to trigger the error and that it was more likely to occur if the camera was deteted as being on a USB2 connection, the results were too random to draw firm conclusions.

WeixiangZhang, 2020-09-14:
@MartyG-RealSense 
I change code based on the example rs-show. 
Run this code, several times, on my side about 20~30 times, the depth stream error happens.
And **RT IC2 Config error** and **Depth stream start failure** received with high probability.

Firmware : 5.12.7.100
OS :   Win10
SDK : v2.38.1
USB : 3.0

And both **1. disable auto-exposure before start** and  **2. disable auto-exposure after start** can reproduce
the issue.

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API

void MyNotificationCallback(const rs2::notification& n) {

	auto severity = n.get_severity();
	auto category = n.get_category();
	auto description = n.get_description();
	auto data = n.get_serialized_data();

	std::string message = "";
	message += "Notification [" + std::string(rs2_log_severity_to_string(severity));
	message += "] ";

	message += " Category: " + std::string(rs2_notification_category_to_string(category));

	if (!description.empty()) {
		message += ", Description: " + description;
	}

	if (!data.empty()) {
		message += ", Data: " + data;
	}

	std::cout << message << std::endl;
}

int main(int argc, char * argv[]) try
{
    rs2::context context;
    rs2::device_hub hub(context);
    rs2::device dev = hub.wait_for_device();

    // 1. disable auto-exposure before start
    auto colorSensor = dev.first < rs2::color_sensor >();
    colorSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    rs2::config config;
    config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 60);
    config.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 60);

    // Start streaming with default recommended configuration
    auto profile = pipe.start(config);
    // 2. disable auto-exposure after start
    //auto colorSensor = profile.get_device().first < rs2::color_sensor >();
    //colorSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);

    auto depthSensor = profile.get_device().first < rs2::depth_sensor >();
    depthSensor.set_notifications_callback(MyNotificationCallback);

    using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::frame depth = data.get_depth_frame().apply_filter(color_map);

        // Query frame size (width and height)
        const int w = depth.as<rs2::video_frame>().get_width();
        const int h = depth.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

[image]

MartyG-RealSense, 2020-09-14:
Apologies for the delay in responding further.   My earlier tests would indicate that it is a problem that is not specific to your computer only, since I was able to replicate the problem to a certain extent on my own computer too.

Could you provide more detail please about the way in which you feel that color auto-exposure being enabled negatively impacts the image?  This may provide insights about other settings that could be changed that could preserve image quality whilst allowing color auto-exposure to remain enabled, greatly reducing the possibility of the error occurring.

WeixiangZhang, 2020-09-15:
@MartyG-RealSense 
We have used D415 in our product, with color auto-exposure enabled as default, for several months, and never find this issue. There is no code to set exposure, just with default settings.

Recently, we need disable color auto-exposure in some scenario to get better performance and found this issue during test. During our debug, we found the key point is to disable color auto-exposure. After adding this code, depth stream error happens with high probability at program starts (about 20~30 test cycles). And if we remove this code and hardware reset D415 to default state, it never happens (about 100 test cycles). 
```
colorSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
```
Tested with five D415 cameras and different DELL laptops and PCs

MartyG-RealSense, 2020-09-15:
I re-researched your entire case from the beginning to seek further insights into possible solutions.

-  I considered whether it was a fault with one particular computer and whether using a different computer could provide better results.  I see in your latest comment above that you have already tried that though.

-  I considered whether putting depth and RGB on separate pipelines instead of having both on the same pipeline may be beneficial.  This may avoid any conflicts that arise from settings changes when both are sharing the same pipeline.

https://github.com/IntelRealSense/librealsense/issues/891#issuecomment-350001755

-  I would still encourage looking at ways to obtain the image that you want without turning off color exposure.  If you could provide details of why you need to turn off color exposure (i.e the conditions that you are trying to avoid from occurring) then I may be able to suggest alternative methods of achieving your goal.

WeixiangZhang, 2020-09-16:
@MartyG-RealSense 
I will try this.
> I considered whether putting depth and RGB on separate pipelines instead of having both on the same pipeline may be beneficial. This may avoid any conflicts that arise from settings changes when both are sharing the same pipeline.

We use D415 to do face/eyes tracking. But recently we find face/eyes in color frame is too dark in some situations (e.g. backlighting) with auto-exposure enabled. So we try to disable color auto-exposure and set gain&exposure manually. Unfortunately this issue was found in our test.

WeixiangZhang, 2020-09-16:
@MartyG-RealSense 
I update my test code using `rs2:context` + `rs2::device` + `rs2::sensor,` the issue happened again.
[#891 (comment)](https://github.com/IntelRealSense/librealsense/issues/891#issuecomment-350001755)

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include <thread>

void MyNotificationCallback(const rs2::notification& n) {

	auto severity = n.get_severity();
	auto category = n.get_category();
	auto description = n.get_description();
	auto data = n.get_serialized_data();

	std::string message = "";
	message += "Notification [" + std::string(rs2_log_severity_to_string(severity));
	message += "] ";

	message += " Category: " + std::string(rs2_notification_category_to_string(category));

	if (!description.empty()) {
		message += ", Description: " + description;
	}

	if (!data.empty()) {
		message += ", Data: " + data;
	}

	std::cout << message << std::endl;
}

int main(int argc, char * argv[]) try
{
	rs2::context context;
	rs2::device_hub hub(context);
	rs2::device dev = hub.wait_for_device();

	auto colorSensor = dev.first<rs2::color_sensor>();
	std::vector<rs2::stream_profile> stream_profiles = colorSensor.get_stream_profiles();
	//disable auto-exposure
	colorSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);

	colorSensor.open(stream_profiles[89]); //640x480@60Hz, BGR8
	colorSensor.start([&](rs2::frame frame) {
		auto videoFrame = frame.as<rs2::video_frame>();

		const int w = videoFrame.get_width();
		const int h = videoFrame.get_height();

		// Create OpenCV matrix of size (w,h) from the colorized depth data
		cv::Mat image(cv::Size(w, h), CV_8UC3, (void*)videoFrame.get_data(), cv::Mat::AUTO_STEP);

		// Update the window with new data
		imshow("Color Frame", image);
		cv::waitKey(1);
	});

	auto depthSensor = dev.first<rs2::depth_sensor>();
	rs2::colorizer colorizer(0);
	std::vector<rs2::stream_profile> depthSensorStreamProfiles = depthSensor.get_stream_profiles();
	depthSensor.open(depthSensorStreamProfiles[234]); //640x480@60Hz, Z16
	depthSensor.set_notifications_callback(MyNotificationCallback);
	depthSensor.start([&](rs2::frame frame) {
		auto depthFrame = frame.as<rs2::depth_frame>();

		const int w = depthFrame.get_width();
		const int h = depthFrame.get_height();

		rs2::video_frame colorizedDepth = colorizer.colorize(depthFrame);

		// Create OpenCV matrix of size (w,h) from the colorized depth data
		cv::Mat image(cv::Size(w, h), CV_8UC3, (void*)colorizedDepth.get_data(), cv::Mat::AUTO_STEP);

		// Update the window with new data
		imshow("Depth Frame", image);
		cv::waitKey(1);
	});

	int input;
	std::cin >> input; //just block main thread

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

[image]

MartyG-RealSense, 2020-09-16:
I ran some tests in the Viewer with color auto-exposure enabled to see whether I could improve the brightness of the image when auto-exposure is enabled.  I found that the image can be significantly brightened if the value of the **Gamma** setting in the RGB controls is maximized.  As this option is not connected to the color auto-exposure, changing it does not cause the auto-exposure to turn off.

WeixiangZhang, 2020-09-18:
@MartyG-RealSense 
Maybe that is a workaround, thanks.
However, I really hope RealSense team can fix this issue ASAP.

MartyG-RealSense, 2020-09-18:
@WeixiangZhang   As I was able to recreate the error myself during testing, I will file a bug report with Intel so that the RealSense team can investigate it.

WeixiangZhang, 2020-10-19:
@MartyG-RealSense How is this issue going ? Does RealSense team have a schedule to fix it ?

MartyG-RealSense, 2020-10-19:
@WeixiangZhang  There is no information or schedule to provide about a bug-fix for the reported issue unfortunately.  It is best not to make your plans dependent on a fix arriving.

Nir-Az, 2025-05-15:
Hi all, we believe this issue is resolved is latest major release 2.55.1
Please try it, if the issue remain please reopen with a reproduction steps.
Thanks!

closed 2025-05-15 by Nir-Az