realsenseai/librealsense #7286: Denoise of depth image

Ted-EE, 2020-09-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |Win10 64-bit| 
| Kernel Version (Linux Only)    |  N/A| 
| Platform                                 | PC|
| SDK Version                            | 2.0| 
| Language                            |  python  | 
| Segment			|  image processing | 

### Issue Description
Hi, there are many noise points in the depth image though I perform closing morphological operations on the raw depth image. How can I denoise such small noise points?
[image]

Thanks!

MartyG-RealSense, 2020-09-09:
Hi @Ted-EE  They look like the kind of holes that would be straightforward for a **Hole-Filling Filter** post-processing filter to close.  Intel's Python post-processing tutorial offers code for setting up a range of filters, including a hole-filling one.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

Ted-EE, 2020-09-09:
Thanks! I successfully filled holes using Hole Filling filter.

MartyG-RealSense, 2020-09-18:
Case closed due to successful outcome and no further comments received.

closed 2020-09-18 by MartyG-RealSense