realsenseai/librealsense #7287: USB cameras not working anymore after librealsense-dkms update

roccopietrini, 2020-09-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.38.1) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (4.4.0-189-generic)                                         | 
| Platform                                 | PC |

### Issue Description
Hi, after last update of librealsense-dkms my usb cameras are not working anymore (internal webcam as well as any USB camera). I have both USB2 and USB3 ports. Cameras are listed under lsusb but programs like cheese/skype/etc. cannot find them, there are no /dev/video* ... 
My issue is similar to [https://github.com/IntelRealSense/librealsense/issues/5868](url).
If I uninstall librealsense-dkms, everything works as expected (cameras works again), but this is not an option. Maybe at the time of update secure boot was enabled (sometimes windows update re-enable it, I have a dual boot), but now is disabled. Re-installing the package again does not produce any improvement and breaks the usb cameras again.. 

```
sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 7.826 kB of archives.
After this operation, 61,8 MB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-dkms all 1.3.12-0ubuntu1 [7.826 kB]
Fetched 7.826 kB in 47s (164 kB/s)                                                                                                                                                                         
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 362883 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.12-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.12-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.12-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.12 DKMS files...
First Installation: checking all kernels...
Building only for 4.4.0-189-generic
Building initial module for 4.4.0-189-generic
Secure Boot not enabled on this system.
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

depmod....

DKMS: install completed.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.12, 4.4.0-189-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.12
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
```

Any help ?

MartyG-RealSense, 2020-09-10:
Hi @roccopietrini   Does it make any difference if you update the **udev rules** after installing the librealsense packages, please?

**Refresh the local packages cache.**
sudo apt-get update

**Update udev rules**
sudo apt-get upgrade/install librealsense2-udev-rules

roccopietrini, 2020-09-10:
Hi,
I just tried to uninstall/reinstall librealsense2-udev-rules and update the udev rules with 
`udevadm control --reload-rules ; udevadm trigger`
but nothing changes..

MartyG-RealSense, 2020-09-10:
Thank you for trying the rule update.  May I confirm that when you are doing a clean package installation after having uninstalled packages from a previous attempt with a purge, you are installing only the two packages below and no others:

sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils

roccopietrini, 2020-09-10:
Yes, I've just tried this.. 
Uninstall everything first:

` dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

USB Cameras (except realsense of course) are working now, but then when I do the following:  

```
sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libatk-bridge2.0-dev libepoxy-dev libgmp-dev libgmpxx4ldbl libgtk-3-dev nettle-dev
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/7.833 kB of archives.
After this operation, 61,8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 362700 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.12-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.12-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.38.1-0~realsense0.3377_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.38.1-0~realsense0.3377) ...
Setting up librealsense2-dkms (1.3.12-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.12 DKMS files...
First Installation: checking all kernels...
Building only for 4.4.0-189-generic
Building initial module for 4.4.0-189-generic
Secure Boot not enabled on this system.
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-189-generic/updates/dkms/

depmod.........

DKMS: install completed.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.12, 4.4.0-189-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.12
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
Setting up librealsense2-udev-rules:amd64 (2.38.1-0~realsense0.3377) ...
```

USB cameras KO.

MartyG-RealSense, 2020-09-10:
I have not personally encountered this particular problem with librealsense before.  Your kindly provided log confirms **Secure Boot not enabled on this system** though.

The final line of the log is **Setting up librealsense2-udev-rules:amd64 (2.38.1-0~realsense0.3377) ...**, after which you report that the USB cameras are all disabled.  That would seem to throw suspicion back on the librealsense udev rules, if there is nothing else in your log after that point

@ev-mp is more experienced with the inner workings of udev than I am, so I will refer the question to them.  

Hi @ev-mp do you think it is possible that an issue with udev could be causing all the USB cameras to become undetectable, given that the log provided in this case confirms that secure boot is disabled?

ev-mp, 2020-09-10:
@roccopietrini , please attach the kernel log to review plug/unplug devices scenario - there is probably a conflicting driver/kernel issue

roccopietrini, 2020-09-10:
Here you are, thanks
[kernel_log.txt](https://github.com/IntelRealSense/librealsense/files/5201507/kernel_log.txt)

ev-mp, 2020-09-10:
@roccopietrini , thank t you for the input - there is an error in several kernel modules reported. 
Please disconnect cameras, restart the PC and provide a fresh log from boot till camera plugin

roccopietrini, 2020-09-10:
No.. I installed Ubuntu 16.04 LTS 4 years ago alongside win10 on a brand new laptop.
Uname produce 

`Linux rocco 4.4.0-189-generic #219-Ubuntu SMP Tue Aug 11 12:26:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

I want to add that everything was working fine couple of months ago. I have been using realsense for the last year without any problem..

roccopietrini, 2020-09-14:
In addition, I just tried to boot with kernel 4.4.0-187-generic and everything works as expected, so something happened with kernels 188 or 189..

MartyG-RealSense, 2020-09-14:
Hi @roccopietrini   If you are able to boot successfully with kernel 4.4.0-187-generic, is it possible to use librealsense with that previous kernel, please?

roccopietrini, 2020-09-15:
Yes, this can be a workaround at the moment.. but I guess there is anyway a problem somewhere since looking online there are other user affected by this [issue](https://askubuntu.com/questions/1267484/new-kernel-broke-the-usb-cameras-on-ubuntu-desktop-16-04/1272697#1272697).. Thanks

MartyG-RealSense, 2020-09-15:
@roccopietrini   I did further careful research, but it seems that this case and the non-RealSense forum that you linked to in the above comment are the only RealSense users that have reported being affected by this problem.

It may be worth re-testing later with subsequent kernel releases to see whether the problem fixes itself.  This was a strategy that worked in the days when Windows updates would stop RealSense cameras from working, only for them to work again when a new build of Windows was later released.

roccopietrini, 2020-09-16:
Thank you @MartyG-RealSense for the support! Let's see what happen with the new kernels, I'll keep you updated.

gachiemchiep, 2020-09-25:
@MartyG-RealSense 
I meet the same problem. Realsense camera used to work fine with 2.32 version. 
Now with the new 2.38.1, realsense can't be detected by realsense-viewer. 

Required Info |  
-- | --
Camera Model | { D435 }
Firmware Version | (2.38.1)
Operating System & Version | {Linux (Ubuntu 16)
Kernel Version (Linux Only) | (4.4.0-190-generic)
Platform | PC

[dmesg.log](https://github.com/IntelRealSense/librealsense/files/5280596/dmesg.log)

Here's that part of dmesg log when I unplug-replug the camera

```bash
[  776.115131] usb 8-4: new SuperSpeed USB device number 2 using xhci_hcd
[  776.131548] usb 8-4: New USB device found, idVendor=8086, idProduct=0b3a
[  776.131555] usb 8-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  776.131559] usb 8-4: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  776.131563] usb 8-4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  776.131567] usb 8-4: SerialNumber: 845523050611
[  776.132327] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  776.132330] ------------[ cut here ]------------
[  776.132342] WARNING: CPU: 8 PID: 130 at /build/linux-UNiqWp/linux-4.4.0/drivers/media/media-device.c:380 __media_device_register+0xf0/0x140 [media]()
[  776.132344] Modules linked in: xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables br_netfilter bridge stp llc aufs overlay binfmt_misc nvidia_uvm(OE) skx_edac edac_core x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_realtek(OE) snd_hda_codec_generic(OE) coretemp i915_bpo kvm_intel kvm hid_sensor_gyro_3d(OE) hid_sensor_accel_3d(OE) hid_sensor_trigger industrialio_triggered_buffer uvcvideo(OE) kfifo_buf videobuf2_vmalloc snd_hda_codec_hdmi(OE) industrialio videobuf2_memops videobuf2_v4l2 videobuf2_core hid_sensor_iio_common v4l2_common videodev(OE) intel_ips media irqbypass i2c_algo_bit crct10dif_pclmul crc32_pclmul snd_hda_intel(OE)
[  776.132409]  joydev snd_hda_codec(OE) snd_hda_core(OE) ghash_clmulni_intel snd_seq_midi snd_hwdep snd_seq_midi_event snd_rawmidi aesni_intel snd_pcm snd_seq dell_wmi sparse_keymap dcdbas video aes_x86_64 lrw gf128mul dell_smm_hwmon xillybus_pcie xillybus_core snd_seq_device 8250_fintek glue_helper snd_timer ablk_helper cryptd input_leds mei_me snd serio_raw wmi mei ioatdma shpchp soundcore mac_hid parport_pc ppdev lp parport autofs4 btrfs xor raid6_pq hid_sensor_custom uas usb_storage hid_sensor_hub hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm nvidia(POE) psmouse ipmi_msghandler igb(OE) e1000e dca ptp ahci pps_core pata_acpi libahci fjes
[  776.132488] CPU: 8 PID: 130 Comm: kworker/8:1 Tainted: P        W  OE   4.4.0-190-generic #220-Ubuntu
[  776.132490] Hardware name: Dell Inc. Precision 7920 Tower/0RN4PJ, BIOS 1.8.0 08/17/2018
[  776.132500] Workqueue: usb_hub_wq hub_event
[  776.132503]  0000000000000286 bc8e0fc833aa6547 ffff88084520f710 ffffffff8140edbb
[  776.132509]  0000000000000000 ffffffffc17f1380 ffff88084520f748 ffffffff810864b2
[  776.132514]  ffff88084f4bf870 ffff88084f4bf870 ffff88072460b000 ffff88084f4bf800
[  776.132519] Call Trace:
[  776.132529]  [<ffffffff8140edbb>] dump_stack+0x6d/0x92
[  776.132539]  [<ffffffff810864b2>] warn_slowpath_common+0x82/0xc0
[  776.132544]  [<ffffffff810865fa>] warn_slowpath_null+0x1a/0x20
[  776.132550]  [<ffffffffc17efa40>] __media_device_register+0xf0/0x140 [media]
[  776.132559]  [<ffffffffc18bcb0b>] uvc_probe+0x48b/0x2a70 [uvcvideo]
[  776.132566]  [<ffffffff81649a9d>] usb_probe_interface+0x11d/0x300
[  776.132574]  [<ffffffff8157d752>] driver_probe_device+0x1c2/0x4b0
[  776.132579]  [<ffffffff8157db44>] __device_attach_driver+0x74/0xa0
[  776.132585]  [<ffffffff8157dad0>] ? __driver_attach+0x90/0x90
[  776.132590]  [<ffffffff8157b499>] bus_for_each_drv+0x69/0xb0
[  776.132596]  [<ffffffff8157d48c>] __device_attach+0xdc/0x170
[  776.132601]  [<ffffffff8157dbb3>] device_initial_probe+0x13/0x20
[  776.132607]  [<ffffffff8157c735>] bus_probe_device+0x95/0xa0
[  776.132612]  [<ffffffff8157a23e>] device_add+0x3de/0x640
[  776.132617]  [<ffffffff81647956>] usb_set_configuration+0x516/0x8e0
[  776.132622]  [<ffffffff8165294e>] generic_probe+0x2e/0x80
[  776.132627]  [<ffffffff81649948>] usb_probe_device+0x38/0x70
[  776.132632]  [<ffffffff8157d752>] driver_probe_device+0x1c2/0x4b0
[  776.132638]  [<ffffffff8157db44>] __device_attach_driver+0x74/0xa0
[  776.132643]  [<ffffffff8157dad0>] ? __driver_attach+0x90/0x90
[  776.132648]  [<ffffffff8157b499>] bus_for_each_drv+0x69/0xb0
[  776.132653]  [<ffffffff8157d48c>] __device_attach+0xdc/0x170
[  776.132659]  [<ffffffff8157dbb3>] device_initial_probe+0x13/0x20
[  776.132664]  [<ffffffff8157c735>] bus_probe_device+0x95/0xa0
[  776.132668]  [<ffffffff8157a23e>] device_add+0x3de/0x640
[  776.132675]  [<ffffffff81541900>] ? add_device_randomness+0x90/0x150
[  776.132681]  [<ffffffff8163c547>] usb_new_device+0x277/0x4b0
[  776.132686]  [<ffffffff8163dd27>] hub_port_connect+0x5e7/0xaf0
[  776.132692]  [<ffffffff8163e901>] hub_event+0x6d1/0xb20
[  776.132698]  [<ffffffff810a0d5b>] process_one_work+0x16b/0x4e0
[  776.132702]  [<ffffffff810a111e>] worker_thread+0x4e/0x590
[  776.132706]  [<ffffffff810a10d0>] ? process_one_work+0x4e0/0x4e0
[... cut ...]
[  776.133604]  [<ffffffff81869da5>] ret_from_fork+0x55/0x80
[  776.133609]  [<ffffffff810a7770>] ? kthread_create_on_node+0x1e0/0x1e0
[  776.133612] ---[ end trace 0182a722d7593f0f ]---
[  776.191007] hid-sensor-hub 0003:8086:0B3A.0005: No report with id 0xffffffff found
[  776.191661] hid-sensor-hub 0003:8086:0B3A.0005: No report with id 0xffffffff found
[  776.247091] hid-sensor-hub 0003:8086:0B3A.0005: No report with id 0xffffffff found
[  776.247911] hid-sensor-hub 0003:8086:0B3A.0005: No report with id 0xffffffff found

```

MartyG-RealSense, 2020-09-25:
Hi @gachiemchiep   Does the Depth Quality Tool function okay on your computer in 2.38.1 please?

gachiemchiep, 2020-09-26:
Hello @MartyG-RealSense . 
Unfortunately, I switch back to 2.32.1 so I couldn't check that. 
As I remember, realsense-viewer couldn't detect any connected camera so I think the Depth Quality Tool will not work

roccopietrini, 2020-09-26:
Also for me with kernel 4.4.0-190-generic problem is still there, looking in dmseg I find this on boot:

```
[   17.665824] videodev: loading out-of-tree module taints kernel.
[   17.665912] videodev: module verification failed: signature and/or required key missing - tainting kernel
[   17.668273] Linux video capture interface: v2.00
[   18.271345] uvcvideo: Found UVC 1.00 device HP Truevision HD (064e:930b)
[   18.271348] ------------[ cut here ]------------
[   18.271352] WARNING: CPU: 1 PID: 412 at /build/linux-UNiqWp/linux-4.4.0/drivers/media/media-device.c:380 __media_device_register+0xf0/0x140 [media]()
[   18.271354] Modules linked in: uvcvideo(OE+) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev(OE) kvm media irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel arc4 rtl8188ee aesni_intel rtl_pci aes_x86_64 rtlwifi lrw mac80211 gf128mul glue_helper ablk_helper snd_hda_codec_realtek snd_hda_codec_hdmi cryptd cfg80211 snd_hda_codec_generic joydev snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm input_leds serio_raw snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq intel_pch_thermal snd_seq_device snd_timer shpchp snd soundcore processor_thermal_device int340x_thermal_zone int3400_thermal intel_soc_dts_iosf acpi_thermal_rel mei_me mei hp_wireless lpc_ich acpi_pad mac_hid parport_pc ppdev lp parport autofs4 hid_generic usbhid amdkfd amd_iommu_v2
[   18.271392]  i915 radeon i2c_algo_bit ttm drm_kms_helper psmouse syscopyarea sysfillrect sysimgblt fb_sys_fops ahci libahci r8169 drm mii wmi video i2c_hid hid fjes
[   18.271403] CPU: 1 PID: 412 Comm: systemd-udevd Tainted: G           OE   4.4.0-190-generic #220-Ubuntu
[   18.271404] Hardware name: HP HP Notebook/80C2, BIOS F.30 12/24/2019
[   18.271406]  0000000000000286 8d36a01b1c62b7c1 ffff88024ff579b0 ffffffff8140edbb
[   18.271408]  0000000000000000 ffffffffc07ce380 ffff88024ff579e8 ffffffff810864b2
[   18.271410]  ffff88008e26f870 ffff88008e26f870 ffff88008e26f800 0000000000000003
[   18.271412] Call Trace:
[   18.271417]  [<ffffffff8140edbb>] dump_stack+0x6d/0x92
[   18.271420]  [<ffffffff810864b2>] warn_slowpath_common+0x82/0xc0
[   18.271423]  [<ffffffff810865fa>] warn_slowpath_null+0x1a/0x20
[   18.271425]  [<ffffffffc07cca40>] __media_device_register+0xf0/0x140 [media]
[   18.271429]  [<ffffffffc0871b30>] uvc_probe+0x490/0x2aa0 [uvcvideo]
[   18.271432]  [<ffffffff81649a9d>] usb_probe_interface+0x11d/0x300
[   18.271436]  [<ffffffff8157d752>] driver_probe_device+0x1c2/0x4b0
[   18.271438]  [<ffffffff8157dac7>] __driver_attach+0x87/0x90
[   18.271440]  [<ffffffff8157da40>] ? driver_probe_device+0x4b0/0x4b0
[   18.271443]  [<ffffffff8157b3c2>] bus_for_each_dev+0x72/0xc0
[   18.271445]  [<ffffffff8157cf6e>] driver_attach+0x1e/0x20
[   18.271447]  [<ffffffff8157caa2>] bus_add_driver+0x1e2/0x280
[   18.271449]  [<ffffffff8157e3e0>] driver_register+0x60/0xe0
[   18.271451]  [<ffffffff81648454>] usb_register_driver+0x84/0x140
[   18.271453]  [<ffffffffc0888000>] ? 0xffffffffc0888000
[   18.271456]  [<ffffffffc0888024>] uvc_init+0x24/0x1000 [uvcvideo]
[   18.271459]  [<ffffffff810021f5>] do_one_initcall+0xb5/0x200
[   18.271462]  [<ffffffff811fbca5>] ? kmem_cache_alloc_trace+0x185/0x1f0
[   18.271465]  [<ffffffff81199cf5>] do_init_module+0x5f/0x1cf
[   18.271468]  [<ffffffff81113dfa>] load_module+0x16aa/0x1c50
[   18.271470]  [<ffffffff811102f0>] ? __symbol_put+0x60/0x60
[   18.271472]  [<ffffffff81224c10>] ? kernel_read+0x50/0x80
[   18.271475]  [<ffffffff811145e4>] SYSC_finit_module+0xb4/0xe0
[   18.271477]  [<ffffffff8111462e>] SyS_finit_module+0xe/0x10
[   18.271481]  [<ffffffff8186991b>] entry_SYSCALL_64_fastpath+0x22/0xd0
[   18.271482] ---[ end trace b32dc362b885a90d ]---
```

any idea?

MartyG-RealSense, 2020-09-26:
@roccopietrini @gachiemchiep  So the problem seems to be that the 4.4 kernel works okay up to  **4.4.0-187** but does not work in kernel releases in the 4.4 series that come after that (188, 189, 190).

Do either of your projects have the option of using the **RSUSB backend** method of building librealsense that is not dependent on kernel patches?

roccopietrini, 2020-10-02:
I don't know much of this method.. can you give me some details about pros/cons and how to do it ? Thank you

MartyG-RealSense, 2020-10-02:
Details of this method for building with the source-code folder using **RSUSB** can be found at the link below:

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

The link below discusses the advantages and disadvantages of using the backend method compared to kernel patching. Scroll down the comment to the section of it headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

MartyG-RealSense, 2020-10-12:
Hi @roccopietrini   Do you still require assistance with this case, please?   Thanks!

roccopietrini, 2020-10-12:
Hi @MartyG-RealSense .. I didn't tried the RSUSB backend method, I would rather prefer to solve my issue in a canonical way, given the fact that I'm running a supported kernel and I am not the only one experiencing this problem. You can close at the moment, we'll see what happens with new kernels.

MartyG-RealSense, 2020-10-12:
Okay, thank you very much @roccopietrini - good luck!

closed 2020-10-12 by MartyG-RealSense

roccopietrini, 2020-10-14:
Just wanted to add that installing hwe and going to kernel 4.15.0-120-generic solved my problem! Thank you

matbord, 2023-04-18:
I have the same problem with kernel 5.15.0-69-generic. what should I do to let my usb webcam work again?

MartyG-RealSense, 2023-04-19:
Hi @matbord   Which RealSense SDK version are you using, please?  Support for kernel 5.15 was only added in SDK version **2.53.1**.

If you are experiencing a possible kernel related conflict then you could build the SDK from source code with the **RSUSB backend** installation method, which bypasses the kernel.  This makes an RSUSB build of the SDK not dependent on Linux versions or kernel versions.

matbord, 2023-04-19:
Hello, these are the version I installed:
 $ dpkg -l | grep realsense
 
 librealsense2-dkms                           1.3.19-0ubuntu1                     all          Modified kernel modules for librealsense2
 
 librealsense2-udev-rules:amd64             2.53.1-0~realsense0.8250            amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules 

Moreover where can I find the **RSUSB backend** installation method?

MartyG-RealSense, 2023-04-19:
You are using SDK 2.53.1.  Thanks very much for the confirmation.

Instructions for an RSUSB backend source code build of the SDK can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692