realsenseai/librealsense #7288: “Frame didn't arrive within 15000” after 24 hours run with D430

hustwb, 2020-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   | 05.12.01.00| 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |     4.9.140-tegra                                 | 
| Platform                                 | NVIDIA Jetson  Nano|
| SDK Version                            |    2.36.0                        | 
| Language                            |  C++                        | 
| Segment			|  others                           | 

### Issue Description
I get a video stream of depth frames from D430,but it happened "Frame didn't arrive within 15000 "after 24 hours ,my code is as follows:

rs2::config cfg;
rs2::pipeline pipe;

bool OpenCamera()
{
    cfg.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
 
    pipe.start(cfg);

    return true;
}

int ReadImage(cv::Mat &DepthImg)
{
    rs2::frameset frameset = pipe.wait_for_frames();

    auto depth = frameset.get_depth_frame();

    const int w = depth.as<rs2::video_frame>().get_width();
    const int h = depth.as<rs2::video_frame>().get_height();

    cv::Mat dMat(cv::Size(w, h), CV_16UC1, (void*)depth.get_data(), cv::Mat::AUTO_STEP);

    if(dMat.empty())
    {
        return ReadImageEmpty;
    }

    dMat.copyTo(DepthImg);

    return ReadCameraOk;

}

int main()
{
   
    cv::Mat DepthImg;

    OpenCamera();   
    
    while(1)
    {
       
            try
            {

                ReadImage(DepthImg); 

            }

            catch(const rs2::error & e)
            {

                std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;                        
                

            }

            cout<<p->tm_hour<<":"<<p->tm_min<<":"<<p->tm_sec<<": read image success"<<endl;

    }

    return 0;
}

MartyG-RealSense, 2020-09-09:
Hi @hustwb   I would recommend running another long-run test and monitoring memory usage in Ubuntu this time.  This should show whether there is a "memory leak" (memory that is not being released properly) that is consuming the Nano's available memory over time and degrading performance, leading eventually to a failure of the program.

The 400 Series cameras can run indefinitely so long as they remain within recommended temperature tolerances, so if problems occur in a long-run session then it is likely to be due to a problem with the USB cable / USB port, the computer hardware or its operating system.

hustwb, 2020-09-10:
Hi @MartyG-RealSense I just ran this program on the nano development board,only 45% of the memory is used,When this problem occurs, other functions can still be used and there is no crash,It's just that the program crashed

hustwb, 2020-09-10:
Hi @MartyG-RealSense  the USB is 3.0 ,I changed other USB3.0 interfaces, but this problem still occurs

MartyG-RealSense, 2020-09-10:
If you are getting the "Frame didn't arrived within 150000" error thrown, you could try using **try_wait_for_frames()** instead of **wait_for_frames()**.  The two instructions work almost the same, except that **try_wait_for_frames()** just returns a "False" status if there is a timeout instead of throwing an error.  This does not guarantee that the program will not cease streaming depth though and just return False continuously.

Perhaps the **hardware_reset()** instruction could be used as part of an 'If' logic check ... if try_wait_for_frames = false then initiate the hardware_reset() function to reset the camera automatically without having to reset it manually with a USB unplug-replug.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

hustwb, 2020-09-11:
Hi @MartyG-RealSense If the "Frame didn't arrived within 150000" error thrown, I use  **hardware_reset()** ,But it still can't be recovered,the code as follows:

bool ReadCamera::OpenCamera()
{
       
    cfg.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
    dev= pipe.start(cfg).get_device();

    return true;
}

void ReadCamera::ResetCamera()
{

    dev.hardware_reset();

}

int ReadImage(cv::Mat &DepthImg)
{
    rs2::frameset frameset = pipe.wait_for_frames();

    auto depth = frameset.get_depth_frame();

    const int w = depth.as<rs2::video_frame>().get_width();
    const int h = depth.as<rs2::video_frame>().get_height();

    cv::Mat dMat(cv::Size(w, h), CV_16UC1, (void*)depth.get_data(), cv::Mat::AUTO_STEP);

    if(dMat.empty())
    {
        return ReadImageEmpty;
    }

    dMat.copyTo(DepthImg);

    return ReadCameraOk;
}

void main()
{
   cv::Mat DepthImg;

   OpenCamera();   

    while(1)
    {

       try
      {
           ReadImage(DepthImg); 
      } 
      catch(const rs2::error & e)
      {
           ReadCamera.ResetCamera()
           std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;                        
      }  

    }

}

MartyG-RealSense, 2020-09-11:
Okay, to recap the points so far:

-  The program stops after 24 hours, reporting "Frame didn't arrived within 15000"
-  There does not appear to be a memory leak
-  The crash is only in the RealSense program, and other functions on the Nano are not affected
-  Performing a hardware_reset() instruction in the program does not correct the problem
-  The program probably is not at fault if it can run successfully for 24 hours before freezing

Should **bool ReadCamera::OpenCamera** instead be **void ReadCamera::OPenCamera**?  I get void and bool mixed up sometimes when programming.

gaopeipeiok, 2020-09-11:
when get "Frame didn't arrived within 15000" error, I call pipe.stop() my program  will be choked .Then I did not call stop,  just call hardware_reset();dev = hub.wait_for_device(); it will failed with error. Then I call exit() to exit my program, but my program will choked too. So I think the SDK is dead clocked, cannot be stop nor exit;

MartyG-RealSense, 2020-09-21:
Hi @hustwb   Do you still require assistance with this case, please?   Thanks!

hustwb, 2020-09-22:
Hi@MartyG-RealSense ,the problem has been solved

hustwb, 2020-09-22:
I reinstall the latest SDK libaray

MartyG-RealSense, 2020-09-22:
Great news - thanks for the update!  As you have found a solution, I will close the case.

closed 2020-09-22 by MartyG-RealSense