realsenseai/librealsense #7290: Realsense Viewer has immediately a segmentation fault

aha-bd-de, 2020-09-09:
------------------------------------------------------------------------

| Required Info                         |                                                       |
|---------------------------------|-------------------------------------|
| Camera Model                       | D435                                              | 
| Firmware Version                   |  05.12.06.00                                   | 
| Operating System & Version |  Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  Linux raspberrypi 5.4.61-v71+     | 
| Platform                                 | Raspberry Pi 4 8GB                        |
| SDK Version                           |  2.38.1                                            | 
| Language                               |  c++                                              | 
| Segment			        |  -                                                    | 

### Issue Description
As soon as I start the realsense viewer, I can see that a window appears, but it disappears at the same moment as soon as I get a segmentation error.
```
$ realsense-viewer 
Segmentation fault
```

My steps of installation were:
```
git clone https://github.com/IntelRealSense/librealsense.git
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
cd librealsense
sudo ./scripts/setup_udev_rules.sh
sudo ./scripts/patch-realsense-ubuntu-lts.sh
sudo apt-get install gcc-5 g++-5

sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5
sudo update-alternatives --set gcc "/usr/bin/gcc-5"
mkdir build && cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release
sudo make uninstall && make clean && make -j4 && sudo make -j4 install
```

aha-bd-de, 2020-09-11:
I followed the installation steps from the "dev.intelrealse.com" site, but that did not work for me well. (new pi with new sd card)

The guide from "acrobotic" works for me, thank you!

closed 2020-09-11 by aha-bd-de

MartyG-RealSense, 2020-09-11:
Great news - thanks so much for the update!