realsenseai/librealsense #7291: RGB Camera Crash when recording in Real sense viewer

Petervries, 2020-09-09:
Hi,

Inside Realsense viewer, I can see pointcloud + RGB texture working just fine, But it seems to crash when I press record.
When loading a recorded sequence I can no longer select the RGB camera as texture. Only the depth is available as source.

I'm trying to make a pointcloud recording with the RGB camera channels to convert as a sequence of .ply files using rs-convert.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | V2.36.0 | 
| Operating System & Version |  Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
How can I record a point cloud sequence with RGB color data from the RGB camera?

RealSenseSupport, 2020-09-14:
Hi @Petervries 

The initial post stated that you are using older SDK, have you attempted to upgrade the SDK to the latest 2.38.1 and updated the camera FW to the latest?

RealSenseSupport, 2020-09-22:
Hi @Petervries 

Were you able to get things working with all the latest SW and FW?

*If we don’t hear from you in 7 days, this issue will be closed.

Petervries, 2020-09-24:
Hi @RealSenseSupport  

That seem to work, thank you!

Best, Peter R de Vries

closed 2020-09-25 by RealSenseSupport