realsenseai/librealsense #7292: how to use D415 as a standard UVC device

WeixiangZhang, 2020-09-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |   Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | ARM  |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C                          | 
| Segment			| others }                          | 

### Issue Description
I want to use D415 as a standard UVC device. Where can I find a detail example? Thanks.

dorodnic, 2020-09-10:
[Cheese](https://wiki.gnome.org/Apps/Cheese) and [V4L2 API example](https://kernel.readthedocs.io/en/latest/media/uapi/v4l/capture.c.html)

WeixiangZhang, 2020-09-10:
@dorodnic I'm not familiar with V4L2 API and want to know if I can get essential info via V4L2, such as

1. color frame and depth frame
2. do alignment between color frame and depth frame
3. map between image and world coordinate

RealSenseSupport, 2020-09-14:
@WeixiangZhang  You can achieve the frames, alignment and mapping by librealsense API. It doesn't depend on V4L2. 
For alignment, please refer to https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp
For mapping, please refer to https://github.com/IntelRealSense/librealsense/blob/master/src/proc/pointcloud.cpp

WeixiangZhang, 2020-09-14:
@RealSenseSupport Thank you.

RealSenseSupport, 2020-09-15:
@Weixiangzhang You're welcome! Any other questions about this? Looking forward to your reply. Thanks!

hywang-66, 2020-09-18:
@RealSenseSupport 

Hello!
1. How can I get the depth frame using the v4l2?
2. How to synchronize between the RGB frame and depth frame?

When I use the realsense camera on the Ubuntu, there are three /dev/video* found. I found only /dev/video2 can use the v4l2 to get RGB data. Whether the /dev/video0 and /dev/video1 can use the v4l2. What's the function of /dev/video0 and /dev/video1? Below is my code. Thank you!

Firmware version:05.12.05.00
OS:Ubuntu
USB:2.0

#include <iostream>
#include <linux/videodev2.h>
#include <string.h>
#include <sys/ioctl.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <sys/mman.h>
#include <stdlib.h>
#include <opencv2/opencv.hpp>
#include <opencv2/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <chrono>

#define DEV_NAME_LEN 64
#define MAX_BUFFER_CNT 5
#define WIDHT 424
#define HEIGHT 240
#define BUFFER_CNT 1
#define VLIB_ERROR -1
#define VLIB_SUCCESS 0

struct buffer {
    unsigned int index;
    int dbuf_fd;
    void *v4l2_buff;
    unsigned int v4l2_buff_length;
};

struct v4l2_dev {
    char devname[DEV_NAME_LEN];
    int fd;
    enum v4l2_buf_type buf_type;
    enum v4l2_memory mem_type;
    struct v4l2_pix_format format;
    struct buffer vid_buf[ MAX_BUFFER_CNT];
    struct video_pipeline *setup_ptr;
};

struct buffer *v4l2_dequeue_buffer(struct v4l2_dev *dev, struct buffer *buffers) {
    struct v4l2_buffer buf;
    int ret;
    memset(&buf, 0, sizeof(buf));
    buf.type = dev->buf_type;
    buf.memory = dev->mem_type;
    ret = ioctl(dev->fd, VIDIOC_DQBUF, &buf);
    return &buffers[buf.index];
};

int v4l2_parse_node(const char *cardname, char *devnode) {
    int fd, i = 0, ret = 0;
    struct v4l2_capability cap;
    char newdev [DEV_NAME_LEN];

    snprintf(newdev, sizeof(newdev), "/dev/video%u", 2);
	if ((fd = open (newdev, O_RDWR)) < 0)
	{
[... 163 lines cut ...]
    dev_0.format.height = HEIGHT;
    dev_0.format.bytesperline = 2 * WIDHT;
    dev_0.format.colorspace = V4L2_COLORSPACE_SRGB;

    if(v4l2_parse_node("uvcvideo", dev_0.devname))
    {
        std::cout << "no device find !" << std::endl;
    }
    v4l2_init(&dev_0, BUFFER_CNT);
    while(true)
    {
        auto start = std::chrono::system_clock::now();
        v4l2_capture(&dev_0);
        cv::Mat frame_yuv(HEIGHT, WIDHT, CV_8UC2, dev_0.vid_buf[0].v4l2_buff);
        cv::Mat frame_rgb;
        cv::cvtColor(frame_yuv, frame_rgb, CV_YUV2BGR_YUYV);
        auto end1 = std::chrono::system_clock::now();
        auto duration1 = double((std::chrono::duration_cast<std::chrono::microseconds>(end1 - start)).count())*std::chrono::microseconds::period::num/std::chrono::microseconds::period::den;
        std::cout << "v4l2 costs " << duration1 << " times!" << std::endl;
        imshow("video", frame_rgb);
        munmap(dev_0.vid_buf[0].v4l2_buff,dev_0.vid_buf[0].v4l2_buff_length);
        cvWaitKey(1);
    }

}

RealSenseSupport, 2020-09-21:
@hywang-66 For /dev/video* , please refer to https://github.com/IntelRealSense/librealsense/issues/6584#issuecomment-643766433   And do you apply kernel patch as described in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md ?

hywang-66, 2020-09-21:
@RealSenseSupport 
When I run the `./scripts/patch-ubuntu-kernel-4.16.sh`, it shows this kernel version is not supported. My Linux kernel version is 4.9.0-xilinx which unable to install patch. Is there any other way to install the patch?

RealSenseSupport, 2020-09-22:
@hywang-66 What's your platform? According to https://github.com/IntelRealSense/librealsense/issues/3534#issuecomment-474444576 you should be able to get depth and color without any patches given that the kernel is 4.4+.  Did you apply udev rules by ./scripts/setup_udev_rules.sh according to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites?

RealSenseSupport, 2020-09-28:
@hywang-66 Any update from your side? Thanks!

hywang-66, 2020-09-28:
@RealSenseSupport 
Now I can get depth and rgb, but when I use dual threads to collect depth and rgb independently, rgb data sometimes will be abnormal. Is this problem related to the opening sequence of the camera? Whether to collect rgb once, then collect depth, again and again? Thanks!

RealSenseSupport, 2020-09-29:
@hywang-66 As your issue is another new one, could you please submit another issue to track? Please state more clearly about the abnormal. Snapshot and logs are helpful. Thanks!

hywang-66, 2020-09-29:
@RealSenseSupport 
When I use V4L2 to collect RGB and depth separately in a dual-thread mode, the two threads are independent of each other. I found that RGB data sometimes has problems. As the picture shown. Why?
[image]

RealSenseSupport, 2020-09-29:
@hywang-66 Any error logs when the issue happens?

RealSenseSupport, 2020-09-29:
@hywang-66 Also please check if the issue caused by incomplete frame received. Please add error check according to https://github.com/IntelRealSense/librealsense/blob/master/src/linux/backend-v4l2.cpp#L973 Thanks!

hywang-66, 2020-09-30:
@RealSenseSupport 
I tried your suggestion, and this error was not print. I found when the RGB data abnormal, it print `VIDIOIC_S_FMT` error. I set the `pixelformat = V4L2_PIX_FMT_YUYV, width = 424, height = 240, bytesperline = 2 * 424, colorspace = V4L2_COLORSPACE_SRGB`. But sometimes RGB data is normal, sometimes RGB data is abnormal. It confused me.

RealSenseSupport, 2020-10-10:
@hywang-66 Did you try single thread to see if the same issue for RGB?

hywang-66, 2020-10-10:
@RealSenseSupport 
If the program only uses a single thread to collect RGB, this problem will not occur. But if the program uses dual-thread to collect RGB and DEPTH separately, this problem will occur.

RealSenseSupport, 2020-10-28:
@hywang-66 Can single-thread acceptable for you?

RealSenseSupport, 2020-11-06:
@hywang-66 Could you please update if single-thread applicable for you? Thanks!

hywang-66, 2020-11-06:
@RealSenseSupport 
If I use a single thread to collect RGB data first, and then collect depth data, set the frame rate to 60, the actual RGB data is only 30 frames. Because RGB data consumes 16.6ms, depth consumes 16.6ms.

RealSenseSupport, 2020-11-11:
@hywang-66 Could you please check the calling order and parameters of VIDIOC_REQBUFS, VIDIOC_QBUF and VIDIOC_STREAMON? Here's one reference link - https://gist.github.com/allenk/a54e9dfc0cba07a670fff406f1c6328d#file-capture_raw_frames-c  especially please note https://gist.github.com/allenk/a54e9dfc0cba07a670fff406f1c6328d#file-capture_raw_frames-c-L254  and https://gist.github.com/allenk/a54e9dfc0cba07a670fff406f1c6328d#file-capture_raw_frames-c-L331

RealSenseSupport, 2020-11-17:
@hywang-66 Any other questions about this ticket? Thanks!

hywang-66, 2020-11-17:
@RealSenseSupport 
I check the parameters of VIDIOC_REQBUFS, VIDIOC_STREAMON, they are right. But if I set VIDIOC_QBUF to 4, whether the currently displayed frame is the latest frame?

RealSenseSupport, 2021-01-08:
if you care the latest frame, I guess 2 or 3 buffers for each stream is better. and please DQBUF the frames ASAP.

There are some webpage may help you.
https://stackoverflow.com/questions/48392317/v4l2-what-happens-when-the-queue-is-full

"Tip 1: Manage your buffers!" in this page.
https://lightbits.github.io/v4l2_real_time/

RealSenseSupport, 2021-05-06:
@hywang-66 Did you get chance to try the above? Any other questions? Looking forward to your update. Thanks!

RealSenseSupport, 2021-05-25:
@hywang-66 Sorry that we don't hear from your for weeks. Will close the ticket at this point. Please feel free to create another new ticket if you still have issues or questions. Thanks!

closed 2021-05-25 by RealSenseSupport