realsenseai/librealsense #7295: How to detect objects using the depth from the intel realsense in c#

Xander-Vreeswijk, 2020-09-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 10 |                                  | 
| Platform                                 | PC |
| SDK Version                            |  2.<?>.<?>                          | 
| Language                            |  C# | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I need to detect different kind of things in a crate like leaves. For that problem I need to detect the objects using the depth image form the intel realsense. I have seen different kinds of solutions for c++ but not one for c#. Can someone help me with this problem.

MartyG-RealSense, 2020-09-10:
Hi @Xander-Vreeswijk  During research of your question, I found that it is much easier to find relevant search results about c# object recognition when googling for **c# object detection**, as virtually nothing is found when googling for the term **c# object recognition.** 
 
There was a past case on this GitHub forum in which a RealSense C# user implemented automatic object detection by "converting the colour frame into a binary image and searching for contours using EMGU / OpenCV".  Details of how this was done were not provided however,

Xander-Vreeswijk, 2020-09-14:
Thank you for your response. I will look if it works.

MartyG-RealSense, 2020-09-21:
Hi @Xander-Vreeswijk    Do you require further assistance with this case, please?   Thanks!

Xander-Vreeswijk, 2020-09-21:
MartyG yes do you know an other way of detecting objects using intel realsense because the microsoft.ml nugget package doesn't work on my laptop. Do you perhaps have a way of using opencvsharp?

MartyG-RealSense, 2020-09-21:
There is a set of **OpenCVSharp** samples at the link below.  Sample number 16 is a program that describes itself as "train your own detector".

https://github.com/VahidN/OpenCVSharp-Samples/tree/master/OpenCVSharpSample16

A RealSense user also suggested the following:

********

You could try using a third-party library like OpenCV.  There's an example / tutorial for object tracking here: 

https://www.learnopencv.com/object-tracking-using-opencv-cpp-python/

You can use OpenCVSharp to get it working in dot net.

Xander-Vreeswijk, 2020-09-22:
thank you for your respons MartyG. I will try it

Xander-Vreeswijk, 2020-09-22:
quick question but in the example they use a video file and not direct from the camera. And if I try to use the frame from the intel realsense wrapper I get the cannot convert error. Can you help me with that?

Xander-Vreeswijk, 2020-10-05:
No it is clear thank you for your assistance

closed 2020-10-05 by Xander-Vreeswijk