realsenseai/librealsense #7300: D435 with new d4 will fuse depth and rgb in black pointcloud and dynamic calibr. tool doesn't work

FPSychotic, 2020-09-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 | 
| Firmware Version                   | 05.12.07.100 but tried the 3 last | 
| Operating System & Version |   doesnt work in w10,ubuntu 1804,ubuntu arm64 or Android  | 
| Kernel Version (Linux Only)    |  (4.16,4.19 or 5,02)                                         | 
| Platform                                 | PC/Raspberry Pi 4/ NVIDIA Jetson  nx/
| SDK Version                            |  { 2}                          | 
| Language                            | ros or sdk                        | 
| Segment			|   }                          | 

### Issue Description
My Realsense d435 didn't boot and I bought a new D4, swaped and booted, but now with the new one when it fuses depth with rgb the point cloud is black, will work properly with IR. I thought it is due to calibration, as it got a new d4 module, it flashed ok from the factory one to last 2 FW versions, but same problem.It is recognized as d435 properly and via usb3.2 and I installed the last version of dynamic calibration tool, 2.11 I think, but send error (9901) , it say run Intel.Realsense.CustomRW.exe -g to get factory valors, I do and it say valors restored, after I try again the calibration with same error. I tested different arm boards, PC, linux, w10...

MartyG-RealSense, 2020-09-11:
Hi @FPSychotic  Could you try using the CustomRW tool (which you have already used to attempt a gold reset) to run the command **CustomRW -r ** in order to check the camera's calibration tables and post the log results in this discussion please?

FPSychotic, 2020-09-11:
It sent me the same error than calibration tool I think

PS C:\Users\fpsyc> Intel.Realsense.CustomRW.exe -r
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again.
check CustomRW tool.

but there is a option on Realsense viewer that allow e get in a .json file calibration data, no idea it is what you need

thanks by the help, I'm struggle with this and I cannot afford other camera right now

FPSychotic, 2020-09-11:
Hi as I wrote Intel.Realsense.CustomRW.exe -r, send the same error than the last version of calibration tool. We are at the same point than the first message

FPSychotic, 2020-09-11:
it doesn't work. I can make calibration on chip, extrinsic and intrinsic with realsense viewer, I can get data table with on it too, in a .json file that I cannot upload, I paste here
{
  "baseline": "-50.0477",
  "intrinsic_left.x.x": "0.499041",
  "intrinsic_left.x.y": "0.796534",
  "intrinsic_left.x.z": "0.500129",
  "intrinsic_left.y.x": "0.5042",
  "intrinsic_left.y.y": "-0.0544546",
  "intrinsic_left.y.z": "0.06197",
  "intrinsic_left.z.x": "0.00011231",
  "intrinsic_left.z.y": "-0.0010748",
  "intrinsic_left.z.z": "-0.0200161",
  "intrinsic_right.x.x": "0.501188",
  "intrinsic_right.x.y": "0.799566",
  "intrinsic_right.x.z": "0.495462",
  "intrinsic_right.y.x": "0.481221",
  "intrinsic_right.y.y": "-0.0546522",
  "intrinsic_right.y.z": "0.0636175",
  "intrinsic_right.z.x": "-0.0012763",
  "intrinsic_right.z.y": "-0.00143045",
  "intrinsic_right.z.z": "-0.0206787",
  "rectified.0.fx": "983.37",
  "rectified.0.fy": "983.37",
  "rectified.0.height": "1080",
  "rectified.0.ppx": "956.814",
  "rectified.0.ppy": "530.774",
  "rectified.0.width": "1920",
  "rectified.1.fx": "655.58",
  "rectified.1.fy": "655.58",
  "rectified.1.height": "720",
  "rectified.1.ppx": "637.876",
  "rectified.1.ppy": "353.85",
  "rectified.1.width": "1280",
  "rectified.10.fx": "0",
  "rectified.10.fy": "0",
  "rectified.10.height": "0",
  "rectified.10.ppx": "0",
  "rectified.10.ppy": "0",
  "rectified.10.width": "0",
  "rectified.11.fx": "0",
  "rectified.11.fy": "0",
  "rectified.11.height": "0",
  "rectified.11.ppx": "0",
  "rectified.11.ppy": "0",
  "rectified.11.width": "0",
  "rectified.12.fx": "0",
  "rectified.12.fy": "0",
  "rectified.12.height": "400",
  "rectified.12.ppx": "0",
  "rectified.12.ppy": "0",
  "rectified.12.width": "640",
  "rectified.13.fx": "0",
  "rectified.13.fy": "0",
  "rectified.13.height": "576",
  "rectified.13.ppx": "0",
  "rectified.13.ppy": "0",
  "rectified.13.width": "576",
  "rectified.14.fx": "0",
  "rectified.14.fy": "0",
  "rectified.14.height": "720",
  "rectified.14.ppx": "0",
  "rectified.14.ppy": "0",
  "rectified.14.width": "720",
  "rectified.15.fx": "0",
  "rectified.15.fy": "0",
  "rectified.15.height": "1152",
  "rectified.15.ppx": "0",
  "rectified.15.ppy": "0",
  "rectified.15.width": "1152",
  "rectified.2.fx": "393.348",
  "rectified.2.fy": "393.348",
  "rectified.2.height": "480",
  "rectified.2.ppx": "318.725",
  "rectified.2.ppy": "236.31",
  "rectified.2.width": "640",
  "rectified.3.fx": "434.322",
  "rectified.3.fy": "434.322",
  "rectified.3.height": "480",
  "rectified.3.ppx": "422.593",
  "rectified.3.ppy": "235.925",
  "rectified.3.width": "848",
  "rectified.4.fx": "327.79",
  "rectified.4.fy": "327.79",
  "rectified.4.height": "360",
  "rectified.4.ppx": "318.938",
  "rectified.4.ppy": "176.925",
  "rectified.4.width": "640",
  "rectified.5.fx": "217.161",
  "rectified.5.fy": "217.161",
  "rectified.5.height": "240",
  "rectified.5.ppx": "211.296",
  "rectified.5.ppy": "117.963",
  "rectified.5.width": "424",
  "rectified.6.fx": "196.674",
  "rectified.6.fy": "196.674",
  "rectified.6.height": "240",
  "rectified.6.ppx": "159.363",
  "rectified.6.ppy": "118.155",
  "rectified.6.width": "320",
  "rectified.7.fx": "245.843",
  "rectified.7.fy": "245.843",
  "rectified.7.height": "270",
  "rectified.7.ppx": "239.203",
  "rectified.7.ppy": "132.694",
  "rectified.7.width": "480",
  "rectified.8.fx": "655.58",
  "rectified.8.fy": "655.58",
  "rectified.8.height": "800",
  "rectified.8.ppx": "637.876",
  "rectified.8.ppy": "393.844",
  "rectified.8.width": "1280",
  "rectified.9.fx": "491.685",
  "rectified.9.fy": "491.685",
  "rectified.9.height": "540",
  "rectified.9.ppx": "478.407",
  "rectified.9.ppy": "265.387",
  "rectified.9.width": "960",
  "world2left_rot.x.x": "0.999991",
  "world2left_rot.x.y": "0.000809247",
  "world2left_rot.x.z": "0.00409609",
  "world2left_rot.y.x": "-0.000810683",
  "world2left_rot.y.y": "1",
  "world2left_rot.y.z": "0.000349102",
  "world2left_rot.z.x": "-0.00409581",
  "world2left_rot.z.y": "-0.00035242",
  "world2left_rot.z.z": "0.999992",
  "world2right_rot.x.x": "0.999997",
  "world2right_rot.x.y": "-0.00216729",
  "world2right_rot.x.z": "0.00105412",
  "world2right_rot.y.x": "0.00216766",
  "world2right_rot.y.y": "0.999998",
  "world2right_rot.y.z": "-0.00034962",
  "world2right_rot.z.x": "-0.00105336",
  "world2right_rot.z.y": "0.000351904",
  "world2right_rot.z.z": "0.999999"
}

FPSychotic, 2020-09-12:
I got a Firmware backup from january, but the FW backup fail to load from realsense viewer in W10 and Ubuntu 18.04. In W10 the th FW app with .exe  wont run, the commands from the tutorial doesn't work. In ubuntu the same coommands works, but it fails too, with this error.
ubuntu@ubuntu:~$ rs-fw-update -s 832112073321 -f .832513024585.20200124_232513.bin

search for device with serial number: 832112073321

updating device: 
Name: Intel RealSense D435, serial number: 832112073321, update serial number: 939423023606, firmware version: 05.12.07.100, USB type: 3.2

failed to locate a device in FW update mode

I have a  a FW backup from when the camera was working with I guess it contents the original factory calibration, but non Intel tool works, I have all, even a clean w10 ad ubuntu installation, my PC is the refence one, a intel nuc i7 with intel gpu. 
So If have the original sensors that works, a new d4 board that works and was able to update to 5.12, a factory calibrated FW backup, w10 with full intel computer, can we make this work?

MartyG-RealSense, 2020-09-12:
The calibration is written to an EEPROM flash memory storage chip inside the camera and not the firmware driver, so using an old firmware file would not restore the original factory calibation.  The EEPROM that the calibration data is written to is on the camera depth module board (I confirmed this with the 400 Series data sheet document).

The data sheet says that the firmware driver is stored in flash memory on the Vision Processor D4 board.

When you say "D4", may I confirm this means that you changed the **Vision Processor D4** board please, and not the **D430 camera module** board with the imaging sensors on it?

Looking back at an earlier log that you provided, the camera was being correctly identified as a D435 after your board exchange, suggesting that the camera module board and the firmware are both okay.

Re-reading your original comment, it sounds as though the depth, IR and RGB work on their own but when mapping RGB onto depth in 3D point cloud mode you get a black cloud.  Is that correct, please?

Have you tried a **Targetless** type calibration with the Dynamic Calibrator tool yet?  

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-2-2-2-target-less-dynamic-calibration-rectification-calibration

This method does not calibrate RGB, just depth.  I have seen it suggested though that if there is a problem with the calibration tables then one can try a targetless calibration first and then follow it up with a targeted calibration.

FPSychotic, 2020-09-12:
> When you say "D4", may I confirm this means that you changed the **Vision Processor D4** board please, and not the **D430 camera module** board with the imaging sensors on it?

**Yes, I changed the Vision processor D4 module board**

> Re-reading your original comment, it sounds as though the depth, IR and RGB work on their own but when mapping RGB onto depth in 3D point cloud mode you get a black cloud. Is that correct, please?

**yes,  as it shown in the picture**
> 
> Have you tried a **Targetless** type calibration with the Dynamic Calibrator tool yet?
Not as the command Intel.Realsense.DynamicCalibrator.exe -m 0 that is in the guide for targer-less do nothing, doesn't work. v2.11
Dynamic Calibration Tool with Target doesn't work as send error (9901) as it is shown in the picture
Firmware tool or firmaware recovery from realsense viewer don't work 
Maybe intel should try if their tools are working properly, I'm using a i7 in a clean installation .

> This method does not calibrate RGB, just depth. I have seen it suggested though that if there is a problem with the calibration tables then one can try a targetless calibration first and then follow it up with a targeted calibration.
no way get works the tools.

Pease can you check test if they works (maybe they are broken by the last firmware),Is there some way to check if my eeprom works properly?
Should I try a previous calibration tool with a previous firmware that is proven works with it? if so, can you please give me the versions to download?

MartyG-RealSense, 2020-09-12:
The cased USB cameras are not meant to be taken apart, so I think that Intel will not be able to offer much technical support in regard to the hardware in this case, as disassembly would constitute a sales warranty invalidation.

There is a debugging function in the SDK called **RS2_CAMERA_INFO_CAMERA_LOCKED** that reports True if the eeprom is locked.  You would have to write a script to generate a readout of the status of the eeprom though.

It may be worth going to the **Settings** interface in the Viewer (left-click the gearwheel icon in the top corner), going to the **General** tab of the Settings window and setting the Viewer to default settings to eliminate the possibility that the black cloud is not caused by an errant setting.

FPSychotic, 2020-09-12:
I'm not asking by any camera warranty, I'm asking to make work the tools available for the users of a product, which doesn't work, or you still didn't show they works.Such tools  are needed to get the product fit to propose, the hardware that effectively has warranty, the D4 video processor need that tools working, if not you sold me only a PCB, I didn't pay a PCB. Such tools are in the price that I paid with the module, and they are necessary to make it work. that the tools don't work have nothing in common with my camera manipulation.
Also they are needed to detect if my module is faulty and it is under warranty.

FPSychotic, 2020-09-12:
the black point cloud is visible in 3 different realsense_viewer installations. Dynamic Calibration tool with pattern gives error (9901) in gui mode in command mode -r and -f wont work, so wont read the table or save it in a file. calibration target-less command do nothing in windows 10, doesn't work

FPSychotic, 2020-09-12:
Target-less doesn't work in linux  
ubuntu@ubuntu:~$ Intel.Realsense.DynamicCalibrator -cli -m 0

Result folder: DyCalibResult/832112073321/DC4

Serial Number: 832112073321
Device Name: Intel RealSense D435
Device PID:  0B07
FW version:  05.12.07.100

Calibration Type: Targetless
Laser Power: On
Auto Exposure: On
AE SetPoint: Auto
Ignore borders: No
Initializing GL ...
display size: 1280,720 with aspect ratio 1.77778
window size: 960,540 with aspect ratio 1.77778
Launching graphical view ...
Streamming Resolution: 1280x720 at 30(fps)
initialize dynamic calibration parameters ...
Start Initialize device ...
initialization failed.

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again.
check CustomRW tool.

ubuntu@ubuntu:~$

MartyG-RealSense, 2020-09-13:
You can query sensors and their information individually with the **sensor-control** sample program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

You can also try to retrieve calibration information with **rs-enumate-devices** using the command below:

**rs-enumerate-devices -c**

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

FPSychotic, 2020-09-13:
> You can query sensors and their information individually with the **sensor-control** sample program.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

**what should I check? dynamic calibration appear as not supported** , just I noted IR streamings won't be shown and the check box option is not in realsense viewer, in ubuntu I can see the IR streams and watch them**
``
======================================================

What would you like to do with the sensor?

0 : Control sensor's options
1 : Control sensor's streams
2 : Show stream intrinsics
3 : Display extrinsics

Select an action: 00

======================================================

Sensor supports the following options:

  0: Backlight Compensation is not supported
  1: Brightness is not supported
  2: Contrast is not supported
  3: Exposure
       Description   : Depth Exposure (usec)
       Current Value : 8500
  4: Gain
       Description   : UVC image gain
       Current Value : 16
  5: Gamma is not supported
  6: Hue is not supported
  7: Saturation is not supported
  8: Sharpness is not supported
  9: White Balance is not supported
  10: Enable Auto Exposure
       Description   : Enable Auto Exposure
       Current Value : 1
  11: Enable Auto White Balance is not supported
  12: Visual Preset
       Description   : Advanced-Mode Preset
       Current Value : 0
  13: Laser Power
       Description   : Manual laser power in mw. applicable only when laser power mode is set to Manual
       Current Value : 150
  14: Accuracy is not supported
  15: Motion Range is not supported
  16: Filter Option is not supported
  17: Confidence Threshold is not supported
  18: Emitter Enabled
       Description   : Emitter select, 0-disable all emitters, 1-enable laser, 2-enable laser auto (opt), 3-enable LED (opt)
       Current Value : 1
  19: Frames Queue Size
       Description   : Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
       Current Value : 16
  20: Total Frame Drops is not supported
  21: Auto Exposure Mode is not supported
  22: Power Line Frequency is not supported
  23: Asic Temperature is not supported
  24: Error Polling Enabled
       Description   : Enable / disable polling of camera internal errors
       Current Value : 1
  25: Projector Temperature is not supported
  26: Output Trigger Enabled
       Description   : Generate trigger from the camera to external device once per frame
       Current Value : 0
  27: Motion Module Temperature is not supported
  28: Depth Units
       Description   : Number of meters represented by a single depth unit
       Current Value : 0.001
  29: Enable Motion Correction is not supported
  30: Auto Exposure Priority is not supported
  31: Color Scheme is not supported
  32: Histogram Equalization Enabled is not supported
  33: Min Distance is not supported
  34: Max Distance is not supported
  35: Texture Source is not supported
  36: Filter Magnitude is not supported
  37: Filter Smooth Alpha is not supported
  38: Filter Smooth Delta is not supported
  39: Holes Fill is not supported
  40: Stereo Baseline
       Description   : Distance in mm between the stereo imagers
       Current Value : 50.0477
  41: Auto Exposure Converge Step is not supported
  42: Inter Cam Sync Mode
       Description   : Inter-camera synchronization mode: 0:Default, 1:Master, 2:Slave, 3:Full Salve, 4-258:Genlock with burst count of 1-255 frames for each trigger
       Current Value : 0
  43: Stream Filter is not supported
  44: Stream Format Filter is not supported
  45: Stream Index Filter is not supported
  46: Emitter On Off
       Description   : Alternating Emitter Pattern: 0:disabled(default), 1:enabled( emitter is toggled on/off on per-frame basis)
       Current Value : 0
  47: Zero Order Point X is not supported
  48: Zero Order Point Y is not supported
  49: Lld Temperature is not supported
  50: Mc Temperature is not supported
  51: Ma Temperature is not supported
  52: Hardware Preset is not supported
  53: Global Time Enabled
       Description   : Enable/Disable global timestamp.
       Current Value : 1
  54: Apd Temperature is not supported
  55: Enable Mapping is not supported
  56: Enable Relocalization is not supported
  57: Enable Pose Jumping is not supported
  58: Enable Dynamic Calibration is not supported
  59: Depth Offset is not supported
  60: Led Power is not supported
  61: Zero Order Enabled is not supported
  62: Enable Map Preservation is not supported
  63: Freefall Detection Enabled is not supported
  64: Receiver Gain is not supported
  65: Post Processing Sharpening is not supported
  66: Pre Processing Sharpening is not supported
  67: Noise Filtering is not supported
  68: Invalidation Bypass is not supported
  69: Ambient Light is not supported
  70: Sensor Mode is not supported
  71: Emitter Always On
       Description   : Emitter always on mode: 0:disabled(default), 1:enabled.
       Current Value : 0
  72: Thermal Compensation is not supported
  73: Trigger Camera Accuracy Health is not supported
  74: Reset Camera Accuracy Health is not supported

Select an option by index:

``
> You can also try to retrieve calibration information with **rs-enumate-devices** using the command below:
> 
> **rs-enumerate-devices -c**
> 
> https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

MartyG-RealSense, 2020-09-13:
On the **rs-enumerate-devices** attempt, it looks as though that instruction was entered without -c on the end to display the calibration information.  Could you try again with **rs-enumerate-devices -c** please and post here the log of what it says.

I tested **sensor-control**, and the options flow I would recommend using is:

Select '0' for the device by index.
Select '0' for the stereo module.
Select '3' to Display Extrinsics.
When asked to choose a sensor and then a stream, enter '0' for Stereo Module.  This produces a list of available **stream profiles**.

[image]

Scroll down through the list and find the number for a Depth stream on the list.  I chose profile 80 which is 848x480 at 60 Hz on my computer).  Type in the number of the profile to check.  

[image]

When asked to choose a sensor that will be used as the target of extrinsic transformation, select '0' for Stereo Module.

[image]

You are asked again to enter a number for a streaming profile.  Enter the same number as the last step (e.g 80).

The extrinsics for that particular stream mode (848x480 at 60 FPS in my computer's case).

[image]

The point of this exercise is to check whether the **sensor-control** tool can read the calibration details of your camera, whereas the Dynamic Calibrator tool refused to.

FPSychotic, 2020-09-14:
> On the **rs-enumerate-devices** attempt, it looks as though that instruction was entered without -c on the end to display the calibration information. Could you try again with **rs-enumerate-devices -c** please and post here the log of what it says.
> 
`ubuntu@ubuntu:~$ rs-enumerate-devices -c
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	832112073321
    Firmware Version              : 	05.12.07.100
    Recommended Firmware Version  : 	05.12.03.00
    Physical Port                 : 	1-1.4-33
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	D400
    Asic Serial Number            : 	939423023606
    Firmware Update Id            : 	939423023606

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 10Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  256x144	@ 90Hz	   Y8
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  848x480	@ 10Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  256x144	@ 90Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1280x720	@ 15Hz	   RGB8
    Color	  1280x720	@ 15Hz	   Y16
    Color	  1280x720	@ 15Hz	   BGRA8
    Color	  1280x720	@ 15Hz	   RGBA8
    Color	  1280x720	@ 15Hz	   BGR8
    Color	  1280x720	@ 15Hz	   YUYV
    Color	  1280x720	@ 10Hz	   RGB8
    Color	  1280x720	@ 10Hz	   Y16
    Color	  1280x720	@ 10Hz	   BGRA8
    Color	  1280x720	@ 10Hz	   RGBA8
    Color	  1280x720	@ 10Hz	   BGR8
    Color	  1280x720	@ 10Hz	   YUYV
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
[... 331 lines cut ...]
**I cannot get intrinsics of rgb either **
`======================================================

What would you like to do with the sensor?

0 : Control sensor's options
1 : Control sensor's streams
2 : Show stream intrinsics
3 : Display extrinsics

Select an action: 2

======================================================

Sensor consists of 3 streams: 
  - Depth #0
  - Infrared #1
  - Infrared #2
Sensor provides the following stream profiles:
0  : Infrared #1 (Video Stream: Y8 1280x800@ 30Hz)
1  : Infrared #2 (Video Stream: Y8 1280x800@ 30Hz)
2  : Infrared #1 (Video Stream: Y16 1280x800@ 25Hz)
3  : Infrared #2 (Video Stream: Y16 1280x800@ 25Hz)

`

MartyG-RealSense, 2020-09-14:
Thank you very much for your tests.  It is normal for all of the intrinsic and extrinsic **Coefficients** to have a value of '0' on the 400 Series cameras, so I do not see a problem there.

It does not seem as though the camera is having problems accessing intrinsics and extrinsics.  I expected this, since the RGB and depth individually does not seem to have a problem, only when mapping RGB to a point cloud.  I just wanted to confirm whether the calibration data could be displayed.  Thank you for that.

It could be that color sensor details are not listed because the selected device is Stereo Module (0) and the RGB details are under sensor option RGB Camera (1).

FPSychotic, 2020-09-14:
> Thank you very much for your tests. It is normal for all of the intrinsic and extrinsic **Coefficients** to have a value of '0' on the 400 Series cameras, so I do not see a problem there.
>
> It does [...]

thank to you by the patience and answer me the weekend. does mean a lot

i found that the streaming color now appear but it is in black, I think that is the issue
======================================================

What would you like to do with the sensor?

0 : Control sensor's options
1 : Control sensor's streams
2 : Show stream intrinsics
3 : Display extrinsics

Select an action: 1

======================================================

Sensor consists of 1 streams: 
  - Color #0
Sensor provides the following stream profiles:
0  : Color #0 (Video Stream: RGB8 1920x1080@ 30Hz)
1  : Color #0 (Video Stream: RAW16 1920x1080@ 30Hz)
2  : Color #0 (Video Stream: Y16 1920x1080@ 30Hz)
3  : Color #0 (Video Stream: BGRA8 1920x1080@ 30Hz)
4  : Color #0 (Video Stream: RGBA8 1920x1080@ 30Hz)
5  : Color #0 (Video Stream: BGR8 1920x1080@ 30Hz)
6  : Color #0 (Video Stream: YUYV 1920x1080@ 30Hz)
7  : Color #0 (Video Stream: RGB8 1920x1080@ 15Hz)
8  : Color #0 (Video Stream: Y16 1920x1080@ 15Hz)
9  : Color #0 (Video Stream: BGRA8 1920x1080@ 15Hz)
10 : Color #0 (Video Stream: RGBA8 1920x1080@ 15Hz)
11 : Color #0 (Video Stream: BGR8 1920x1080@ 15Hz)
12 : Color #0 (Video Stream: YUYV 1920x1080@ 15Hz)
13 : Color #0 (Video Stream: RGB8 1920x1080@ 6Hz)
14 : Color #0 (Video Stream: Y16 1920x1080@ 6Hz)
15 : Color #0 (Video Stream: BGRA8 1920x1080@ 6Hz)
16 : Color #0 (Video Stream: RGBA8 1920x1080@ 6Hz)
17 : Color #0 (Video Stream: BGR8 1920x1080@ 6Hz)
18 : Color #0 (Video Stream: YUYV 1920x1080@ 6Hz)
19 : Color #0 (Video Stream: RGB8 1280x720@ 30Hz)
20 : Color #0 (Video Stream: Y16 1280x720@ 30Hz)
21 : Color #0 (Video Stream: BGRA8 1280x720@ 30Hz)
22 : Color #0 (Video Stream: RGBA8 1280x720@ 30Hz)
23 : Color #0 (Video Stream: BGR8 1280x720@ 30Hz)
24 : Color #0 (Video Stream: YUYV 1280x720@ 30Hz)
25 : Color #0 (Video Stream: RGB8 1280x720@ 15Hz)
26 : Color #0 (Video Stream: Y16 1280x720@ 15Hz)
27 : Color #0 (Video Stream: BGRA8 1280x720@ 15Hz)
28 : Color #0 (Video Stream: RGBA8 1280x720@ 15Hz)
29 : Color #0 (Video Stream: BGR8 1280x720@ 15Hz)
30 : Color #0 (Video Stream: YUYV 1280x720@ 15Hz)
31 : Color #0 (Video Stream: RGB8 1280x720@ 6Hz)
32 : Color #0 (Video Stream: Y16 1280x720@ 6Hz)
33 : Color #0 (Video Stream: BGRA8 1280x720@ 6Hz)
34 : Color #0 (Video Stream: RGBA8 1280x720@ 6Hz)
35 : Color #0 (Video Stream: BGR8 1280x720@ 6Hz)
36 : Color #0 (Video Stream: YUYV 1280x720@ 6Hz)
37 : Color #0 (Video Stream: RGB8 960x540@ 60Hz)
38 : Color #0 (Video Stream: Y16 960x540@ 60Hz)
39 : Color #0 (Video Stream: BGRA8 960x540@ 60Hz)
40 : Color #0 (Video Stream: RGBA8 960x540@ 60Hz)
41 : Color #0 (Video Stream: BGR8 960x540@ 60Hz)
42 : Color #0 (Video Stream: YUYV 960x540@ 60Hz)
43 : Color #0 (Video Stream: RGB8 960x540@ 30Hz)
44 : Color #0 (Video Stream: Y16 960x540@ 30Hz)
45 : Color #0 (Video Stream: BGRA8 960x540@ 30Hz)
46 : Color #0 (Video Stream: RGBA8 960x540@ 30Hz)
[... 128 lines cut ...]
175: Color #0 (Video Stream: RGB8 320x180@ 60Hz)
176: Color #0 (Video Stream: Y16 320x180@ 60Hz)
177: Color #0 (Video Stream: BGRA8 320x180@ 60Hz)
178: Color #0 (Video Stream: RGBA8 320x180@ 60Hz)
179: Color #0 (Video Stream: BGR8 320x180@ 60Hz)
180: Color #0 (Video Stream: YUYV 320x180@ 60Hz)
181: Color #0 (Video Stream: RGB8 320x180@ 30Hz)
182: Color #0 (Video Stream: Y16 320x180@ 30Hz)
183: Color #0 (Video Stream: BGRA8 320x180@ 30Hz)
184: Color #0 (Video Stream: RGBA8 320x180@ 30Hz)
185: Color #0 (Video Stream: BGR8 320x180@ 30Hz)
186: Color #0 (Video Stream: YUYV 320x180@ 30Hz)
187: Color #0 (Video Stream: RGB8 320x180@ 6Hz)
188: Color #0 (Video Stream: Y16 320x180@ 6Hz)
189: Color #0 (Video Stream: BGRA8 320x180@ 6Hz)
190: Color #0 (Video Stream: RGBA8 320x180@ 6Hz)
191: Color #0 (Video Stream: BGR8 320x180@ 6Hz)
192: Color #0 (Video Stream: YUYV 320x180@ 6Hz)

Please select the desired streaming profile: 61

Streaming profile: Color. Close display window to continue...

[image]
it will happen in w10 too

FPSychotic, 2020-09-14:
Now is working

MartyG-RealSense, 2020-09-14:
Thank you.   The point cloud is not black any more?

FPSychotic, 2020-09-14:
yes the pointcloud is black and sometimes white
[image]
and aligned depth to color dont stream in ROS, that is equivalent

[image]

FPSychotic, 2020-09-14:
the tests are ok, but point cloud  is black, and I still cannot use the customWR and dynamic calibration tools, just like start ed or worse because now I know it doesn't work in ROS neither.

MartyG-RealSense, 2020-09-14:
I will seek further advice about your point cloud problem and get back to you.

FPSychotic, 2020-09-14:
> I will seek further advice about your point cloud problem and get back to you.

thanks very much, I understand how so weird is this issue.

MartyG-RealSense, 2020-09-15:
Hi again, I have received guidance from Intel about your problem.

On the D435, the RGB calibration table is stored on the ASIC board, so if the ASIC board stops working then it is hard to retrieve that information through the USB connection.

However, if you have performed a firmware update before, you should have backup firmware stored in your computer at a folder location that is something like:

C:\Users\<YourUserName>\AppData\Roaming\ 012345678901.20200914_143743.bin

If you have performed Dynamic Calibration with the camera before on your computer then there may also be a chance to retrieve the RGB calibration table.  Usually under a folder like “C:\CalibrationToolAPI\2.6.8.0\bin\DyCalibResult\802312060126\DC2”, there is a file called “RgbCalibrationTable.xml”.  This could be loaded into the camera that has the new D4 board to recover the old RGB calibration table.

The Intel.Realsense.CustomRW.exe command can help to load the above XML file.  It should be located under a folder like “C:\Program Files\Intel\CalibrationToolAPI\2.6.8.0\bin”

FPSychotic, 2020-09-16:
> On the D435, the RGB calibration table is stored on the ASIC board, so if the ASIC board stops working then it is hard to retrieve that information through the USB connection.
> 
**ASIC board is the D4 processor, am I right?, when I removed it because the camera didn't boot, I lost the calibration of  rgb as the rgb camera is a extra over the module d430 (which keep the calibration data in the camera module). D4 image processor has not as target as spare of d435, it has as target d430 which has calibrated data in eeprom. As D4 has not calibration from factory, I need intrude it at hand, and the pointcloud is black because the software cannot get a calibration from d4 board. I understood right the issue? **
> However, if you have performed a firmware update before, you should have backup firmware stored in your computer at a folder location that is something like:
> 
**yes, I have an backup with I couln't upload with the recovery tool.I will try again upload it, but the problem is software dont work, reason I openned this case. dynamic cal. tull doesnt work, customRW doesnt work and realsense viewer works but fails to upoad the backup to the camera.  As alternative maybe I can extract the .bin firm file and if it content a .json or .xml with the rgb calibration data I can ingress it manually on realsense viewer, with the tool data calibration table, write, it is under calibration on chip option.
Please find attached a backup with production firm 5.10.06 . I needed add .txt, please remove it.
[832513024585.20200124_232513.bin.txt](https://github.com/IntelRealSense/librealsense/files/5231976/832513024585.20200124_232513.bin.txt)
I have other backup from a different firmware.**

> If you have performed Dynamic Calibration with the camera before on your computer then there may also be a chance to retrieve the RGB calibration table. Usually under a folder [...]
**I never used windows until this moment to try fix the issue. I wil look by such file in ubuntu, but probaby is not**
**I never performed a calibration sadly, can I or Intel extract that “RgbCalibrationTable.xml” from my firm backup**
> The Intel.Realsense.CustomRW.exe command can help to load the above XML file. It should be located under a folder like “C:\Program Files\Intel\CalibrationToolAPI\2.6.8.0\bin”

**the tool you refer will not work in windows or ubuntu in my case, this days is 2.11 version,, there is not version 2.06 never more, I tried find 2.06 to try, but I couldnt find it. Maybe it doesnt work because there is not calibration in ASIC, and that is the problem, maybe If I give a file with calibration, it works.
While I try upload the backup would be great if you can help to extract the calibration data from the file I uploaded and check if the CustomRW 2.11 tool works with you**

After this I will not ask you more and quit you more time, as it is not problem of 4D and I understand the situation. the reason to swap the d4 processor and do not use warranty was the camera was send to me from Intel, and is Intel who have the invoice and who bought the camera. Just  recognize your patience and help. I'm sorry if I was  a little frustated by the situation of any tool works, but Inow understand that probaby is because the tools are designed for a d435 with calibration data, which is not my case.

MartyG-RealSense, 2020-09-16:
Thanks for the detailed update.  

D4 is the ASIC, yes.

I cannot carry out the firmware task that you requested though as I cannot risk making a camera unusable.  I will refer the matter back to Intel for further advice about your file.

MartyG-RealSense, 2020-09-18:
Hi again @FPSychotic   Intel gave your problem further attention and provided a response.  A RealSense team member was able to extract the calibration table from the backup firmware file that you provided.  I have attached the file - **832513024585.20200124_232513.bin** -  as a zipped file at the bottom of this message.  Please try copying it into your new ASIC board using this command: 

**Intel.Realsense.CustomRW.exe -r -f 832513024585_CAL_table.xml**

Let us know what happens, please.

The RealSense team member who extracted the backup firmware said that it is likely firmware version **5.12.1** rather than the **5.10.06** quoted earlier in this discussion, as 5.12.1 was shown when they applied it to their camera.

*********************

[832513024585_CAL_table.zip](https://github.com/IntelRealSense/librealsense/files/5246916/832513024585_CAL_table.zip)

FPSychotic, 2020-09-18:
> Hi again @FPSychotic Intel gave your problem further attention and provided a response. A RealSense team member was able to extract the calibration table from the backup firmware file that you [...]
Hi, thanks by the effort, I copied the file into my user folder root and used the command above, which is read, I think should be -w of write, anyway I used both and same error of always
`PS C:\Users\fpsyc> Intel.Realsense.CustomRW.exe -r -f 832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator 
software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and
 try again.
check CustomRW tool.

PS C:\Users\fpsyc> Intel.Realsense.CustomRW.exe -w -f 832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator 
software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and
 try again.
check CustomRW tool.  ` 
[image]
[image]
**using documents directory thinking in avoid permissions issue with -r and -w **
`PS C:\Users\fpsyc> Intel.Realsense.CustomRW.exe -w -f C:\Users\fpsyc\OneDrive\Documents\832513024585_CAL_table\832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator 
software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and
 try again.
check CustomRW tool.
PS C:\Users\fpsyc> Intel.Realsense.CustomRW.exe -r -f C:\Users\fpsyc\OneDrive\Documents\832513024585_CAL_table\832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator 
software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and
 try again.
check CustomRW tool.
`
> 
> The RealSense team member who extracted the backup firmware said that it is likely firmware version **5.12.1** rather than the **5.10.06** quoted earlier in this discussion, as 5.12.1 was shown when they applied it to their camera.
**probably is right I don't remember why I thought it  was that version, maybe a wrong search in the logs or intel firm website
I'm going to try in linux same commands **

It is proven my ASIC can update firmware, is it possible make a 5.12.07.100 with that configuration data and give me in file .bin or through AWS as alternative update server? or maybe unlocking the camera, just by try something more before need scrap the camera.

FPSychotic, 2020-09-18:
After try in Ubuntu, same error
`ubuntu@ubuntu:~$ Intel.Realsense.CustomRW -r -f 832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again.
check CustomRW tool.

ubuntu@ubuntu:~$ Intel.Realsense.CustomRW -w -f 832513024585_CAL_table.xml
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 832112073321
  Firmware version: 05.12.07.100

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. 
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again.
check CustomRW tool.
[image]
`

MartyG-RealSense, 2020-09-19:
Thanks for the detailed report @FPSychotic   I will pass it on to the RealSense team member who handled the calibration extraction.

MartyG-RealSense, 2020-09-29:
Hi again @FPSychotic   After discussions with Intel, it was concluded that there is nothing else that can be done, unfortunately.  Do you have any further questions please?

FPSychotic, 2020-09-30:
So calibration tools wont work for me, so finally what is the reason? I cannot calibrate the color camera with the stardar tools. Intel sell D4 ASIC module to work with D430. Can be the D430 calibrated with that standard tools? If I remove the RGB camera, so know I will have a 430, will work the tools? So I remove the rgb camera, I calatibrate it as a d430 or load my previous calibration backup, now I have a stereo camera calibrated for slam, I use the color stream for other uses, as AI, computer vision, video streaming...

MartyG-RealSense, 2020-09-30:
A complicating factor in this case is that the new ASIC is not OEM-calibrated, so it is not pairing well with the D430 board inside the D435 that **did** undergo factory calibration.  

In theory, if the RGB sensor in the D435 has its cable disconnected then the firmware should identify the camera as being D430 instead of D435.  I'm not sure that it is going to change anything but now that you have come this far, you may as well try.

FPSychotic, 2020-09-30:
Well I will try and I will tell you the results, maybe helps others

MartyG-RealSense, 2020-10-08:
Hi @FPSychotic  do you have an update for us please about your tests?   Thanks!

MartyG-RealSense, 2020-10-17:
Case closed due to no further comments received.

closed 2020-10-17 by MartyG-RealSense

qqice, 2024-09-02:
I'm having exactly the same problem, that the D435 camera can receive both RGB and Stereo streams, but fail to fuse depth and RGB together. I've tried to execute Intel.Realsense.CustomRW.exe -r , and I also received the error:
```
CustomRW for Intel RealSense D400, Version: 2.13.1.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 035422071337
  Firmware version: 05.16.00.01
  Type: 3.2
  Physical port: \\?\usb#vid_8086&pid_0b07&mi_00#6&21764678&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

Error (9901): calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software.
If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again.
check CustomRW tool.
```
So, after close to four years of effort, the issue remains unsolved? Or do you have any new approaches to solving the problem?

MartyG-RealSense, 2024-09-02:
Hi @qqice  If you have access to the RealSense Viewer tool then you could try resetting your camera to its factory-new default calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

qqice, 2024-09-03:
> Hi @qqice  If you have access to the RealSense Viewer tool then you could try resetting your camera to its factory-new default calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

It doesn't work. After I successfully reset the camera to factory-new calibration, the fuse is still in a mess. I think maybe I need access to something like "RotationLeftRGB" to get it fixed. Do you have further information about how to do it?

MartyG-RealSense, 2024-09-03:
@qqice  Which version of the RealSense SDK / RealSense Viewer are you using, please?  If you have firmware 5.16.0.1 installed in your camera then the SDK / Viewer version that should be used with it is **2.55.1**.

If you are using 2.55.1 and are also using the RealSense Viewer, its 2D mode does not have depth-color alignment.  Depth and color can be mapped together though in its 3D pointcloud mode (accessible by clicking on '3D' in the top corner of the Viewer window) by enabling depth first and RGB second.

**********

Could you use the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 to check whether you have a pair of **GLSL** options called 'Rendering' and 'Processing' under the 'Performance' section of the Settings interface of the RealSense Viewer, please.  If you do have these options and they are both enabled - shown as a blue box beside them - then please try disabling Rendering and Processing to see whether depth and RGB can then be successfully mapped together in the Viewer's 2D mode.

*********

If RGB is not mapping to depth in 3D mode then an alternative mode of texturing the pointcloud is to enable Depth and Infrared in the 'Stereo Module' section of the side-panel instead of Depth and RGB.  Then go to the 'Texture' option at the top of the Viewer window and select 'Infrared 1' from the drop-down to texture the cloud with a black and white RGB-like infrared image instead of color RGB.

@FPSychotic was able to texture depth and infrared in the pointcloud mode like this, but depth and RGB resulted in a black pointcloud.

[image]

[image]

[image]

qqice, 2024-09-03:
> If you are using 2.55.1 and are also using the RealSense Viewer, its 2D mode does not have depth-color alignment. Depth and color can be mapped together though in its 3D pointcloud mode (accessible by clicking on '3D' in the top corner of the Viewer window) by enabling depth first and RGB second.

Yes, I'm using realsense-viewer with version 2.55.1, and firmware 5.16.0.1. I've learned that depth and color can be mapped together though in its 3D pointcloud mode, but my camera couldn't map them together, only left me something like this:
[image]

> Could you use the instructions at [#8110 (comment)](https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023) to check whether you have a pair of **GLSL** options called 'Rendering' and 'Processing' under the 'Performance' section of the Settings interface of the RealSense Viewer, please. If you do have these options and they are both enabled - shown as a blue box beside them - then please try disabling Rendering and Processing to see whether depth and RGB can then be successfully mapped together in the Viewer's 3D mode.

I've tried to turn on and turn off both of them, but nothing happened.

> If RGB is not mapping to depth in 3D mode then an alternative mode of texturing the pointcloud is to enable Depth and Infrared in the 'Stereo Module' section of the side-panel instead of Depth and [...]

Yes. I am going through the same situation as @FPSychotic. Depth and infrared fused correctly, but depth and RGB didn't.
[image]

MartyG-RealSense, 2024-09-03:
@qqice  Are you able to test alignment with the RealSense SDK's **rs-align** depth-color alignment example program to see whether alignment also has a problem there too?  If you installed the SDK with tools and examples included then a pre-built executable version of the rs-align program should be in the **usr/local/bin** folder of Linux.

qqice, 2024-09-03:
> @qqice Are you able to test alignment with the RealSense SDK's **rs-align** depth-color alignment example program to see whether alignment also has a problem there too? If you installed the SDK with [...]

Sure. I will post the test pictures below:
[image]
[image]
[image]
[image]
[image]
[image]

MartyG-RealSense, 2024-09-03:
@qqice  Thank you for your rs-align images.  They definitely do not look normal for what would be expected from that program's output.

You tried the -r mode of CustomRW to read the camera's calibration values.  Have you also tried the -g (Gold Reset) flag that @FPSychotic tried during their own case?  Similar to the calibration reset in the Viewer, the Gold Reset function attempts to recover the camera's calibration table.

`Intel.Realsense.CustomRW.exe -g `

qqice, 2024-09-03:
> @qqice Thank you for your rs-align images. They definitely do not look normal for what would be expected from that program's output.
> 
> You tried the -r mode of CustomRW to read the camera's calibration values. Have you also tried the -g (Gold Reset) flag that @FPSychotic tried during their own case? Similar to the calibration reset in the Viewer, the Gold Reset function attempts to recover the camera's calibration table.
> 
> `Intel.Realsense.CustomRW -g `

Yes. It does say "Calibration on device successfully reset to default gold factory settings." However, things haven't changed at all. I also tried -w mode, and the error is the same as -r mode.

MartyG-RealSense, 2024-09-03:
@qqice  Please go back to the 3D mode of the Viewer with both depth and RGB disabled.  Then enable **depth first** and RGB second and see whether the two can align correctly with this method.  

Enabling in this order is important because if RGB is enabled first and depth second then it can cause the RGB image to disappear and only show the depth image.

If this method does not work then it is unfortunately likely that there is nothing further that can be done.

qqice, 2024-09-04:
> @qqice Please go back to the 3D mode of the Viewer with both depth and RGB disabled. Then enable **depth first** and RGB second and see whether the two can align correctly with this method. [...]

It still doesn't make my camera work, but thanks for the advice and help anyway!

JAY-Yaser, 2025-07-03:
I have encountered exactly the same problem. My camera is a D435. The color image and the point cloud cannot be aligned. I wonder if there is a solution now.