realsenseai/librealsense #7302: L515: failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS

RessenTINTINmen, 2020-09-11:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Mac OS|
|**Name**|Intel RealSense L515|
|**Serial Number**|ffffffffffffffff|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|ffffffffffffffff|
|**Firmware Update Id**|ffffffffffffffff|

Hi,
I started realsense viewer, and after connecting L515, the following errors happened. In this way, the installation of the firmware is the situation that there is not, too.

<<error code>>
14:55:49 [Error] /Users/siglab/Downloads/librealsense-2.38.1/src/libusb/handle-libusb.h:95 - failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:55:49 [Error] /Users/siglab/Downloads/librealsense-2.38.1/src/error-handling.cpp:68 - Error during polling error handler: failed to set power state
14:55:50 [Error] /Users/siglab/Downloads/librealsense-2.38.1/common/notifications.cpp:514 - Fatal error occur and device is unable 
to run depth stream
15:01:18 [Error] /Users/siglab/Downloads/librealsense-2.38.1/src/libusb/handle-libusb.h:95 - failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
15:01:19 [Error] /Users/siglab/Downloads/librealsense-2.38.1/src/libusb/handle-libusb.h:95 - failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
15:01:19 [Error] /Users/siglab/Downloads/librealsense-2.38.1/src/error-handling.cpp:68 - Error during polling error handler: failed to set power state
15:01:20 [Error] /Users/siglab/Downloads/librealsense-2.38.1/common/notifications.cpp:514 - Fatal error occur and device is unable 
to run depth stream

Please provide a description of the problem!

ishanivyas, 2020-09-12:
I am having the same issue with MAC OS.

RealSenseSupport, 2020-10-22:
Hi @RessenTINTINmen @ishanivyas 

With the latest SDK 2.39 and latest FW, are you still seeing this issue?

MotokiShiga, 2020-10-24:
Using the same hardware RessenTINTINmen used (we are in the same lab), the trouble occurred in Window 10 during the FW update.  I have met it with both the latest SDK and an previous version (Intel.RealSense.SDK-WIN10-2.35.2.1897). RGB camera and Motion module work by the Viewer. But L500 depth sensor does not work. (The viewer crashed when the depth sensor button was turned on.)  Serial number displayed by the software, "ffffffffffffffff", is also strange. Do you think whether this trouble is on software or hardware?

Borzborz, 2020-11-06:
@ishanivyas I have the same problem, I'm using a 10.15.7 OS

buenos-dan, 2020-11-19:
@ishanivyas I have the same problem too, I'm using a 10.15.7 OS

RealSenseSupport, 2021-01-12:
Hi 

We are still investigating the issues of the serial number turning to all "f".  Unfortunately we do not have a resolution to this at this time.  You can investigate the RMA process with the site or distributor that the units were purchased from.  Once we understand the issue of serial number turning "f", we will make sure we outline next steps.

RealSenseSupport, 2021-03-22:
Regarding the MacOS and IMU part of the thread, there was a PR related that should be looked at here:  https://github.com/IntelRealSense/librealsense/pull/8029

closed 2021-03-30 by RealSenseSupport