realsenseai/librealsense #7303: Assets\Rs\RealSenseSDK2.0\Scripts\RsARBackgroundRenderer.cs(14,13): error CS0246: The type or namespace name 'ARBackgroundRenderer' could not be found (are you missing a using directive or an assembly reference?)

MillSatoshi, 2020-09-11:
Hello.
when I Open the Unity2020 1.4f demo in librealsense-2.10.3 on windows10 connecting d435,there are some errors.Most errors are missing an assembly reference.how can I issue these problems?
I usually use Point Cloud in RealSenseSDK2.0. Suddenly an error occurred.
[image]

MartyG-RealSense, 2020-09-11:
Hi @MillSatoshi   Another RealSense user who was using Unity 2020 encountered this same error with **ARBackgroundRenderer**.  Advice is provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/6949#issuecomment-666166439

MillSatoshi, 2020-09-12:
I add a postscript to question.If you go downgrade to the old Unity version,（More specifically,Version is Unity2019.4.10f1)that  error didn't occur.
Maybe it's a version issue?

MartyG-RealSense, 2020-09-12:
Unity 2019 versions are the most proven and stable for use with the RealSense SDK wrapper.  Unity 2020 is still new and so using existing programs with it instead of starting an empty project from the beginning can have unexpected results.  Unity's own notes about 2020.1 states that "Unity 2020.1 includes a wide range of features and improvements that make Unity workflows even more intuitive and make you more productive, while Unity 2019 LTS remains the recommended version for projects in production".

MartyG-RealSense, 2020-09-21:
Case closed due to answer provided and no further comments received.

closed 2020-09-21 by MartyG-RealSense

sliwowitz, 2022-04-09:
Almost 2 years later, Unity 2020 LTS is the go-to stable version but this bug hasn't been resolved yet so RealSense cameras can't be used in newer projects. Maybe the issue should be re-opened?

MartyG-RealSense, 2022-04-09:
Hi @sliwowitz  RealSense cameras can be used in modern Unity projects and I have successfully tested with a range of different camera models all of the Unity sample programs in Unity 2021.1.16f **except** for ARBackground.  This is because the Unity wrapper documentation contains a note stating that this specific example will not work on Unity 2020 (or 2021 either, my tests confirmed).

[image]

On my own Windows computer that I tested with, there is no error message but the camera streams are not rendered.

[image]

sliwowitz, 2022-04-17:
@MartyG-RealSense Ah! you're right of course. I didn't realize Unity just compiles everything even if it's not actually used in the scene, so it didn't occur to me I could just remove ARBackground to get RealSense working. Thank you for the clarification!

MartyG-RealSense, 2022-04-17:
You are very welcome  :)  Good luck with RealSense in Unity!