realsenseai/librealsense #7305: what is a Z Accuracy for sr305 depth camera

dsmahalle, 2020-09-11:
i need to measure height of around 10 to 20mm. and i need to know what accuracy can realsense provide.

MartyG-RealSense, 2020-09-11:
Hi @dsmahalle   The accuracy of the SR300 cameras has been quoted in the past as being around +- 5%.  I am not aware of an official document source to confirm this, though it was discussed unofficially some years ago when the SR300 camera was new.   The SR300 is the camera circuit board that is inside the SR305 camera model. 

The link below discusses in detail users' experiences with recording the accuracy of the old F200 model, which was the SR300's direct predecessor that had very similar hardware.

https://community.intel.com/t5/Software-Archive/Depth-Accuracy-and-reliability-across-different-f200-sensors/td-p/1020076

The Z-accuracy of the camera can be tested in the **Depth Quality Tool** from the RealSense SDK 2.0, though this requires having an actual camera to attach to the computer.

dsmahalle, 2020-09-11:
is it possible to achieve 99.5%  OR more z accuracy.
if yes, the which camera can give this accuracy?

MartyG-RealSense, 2020-09-11:
The 400 Series cameras can achieve less than 1% error under optimal conditions.  Camera configuration, the environment that the camera is in and other factors can influence accuracy though.   If you are doing very close range sensing, the D435 or D435i model would be suitable as they have has a minimum depth sensing distance (distance of the camera from the observed object) of 0.1 meters.

Generally, the closer that the camera is to the object, the less the error should be, as the error factor in depth measurements on the 400 Series cameras increases linearly over distance.  This phenomenon is known as **RMS Error**.

The table below is taken from the data sheet document for the 400 Series cameras.

[image]

dsmahalle, 2020-09-11:
Thank you very much for your reply. 
your response is helpful for me.

closed 2020-09-11 by dsmahalle