realsenseai/librealsense #7306: D435i camera stop working with my apllication

akhmeevaam, 2020-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10   | 
| Platform                                 | PC |
| SDK Version                            |  { librealsense 2.36.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I got a realsense D435i camera. It works perfectly until the beginning of this week. Now my app cannot connect to camera but when I open RealSense Viewer I can get images from camera.
Also if I try to run hello-realsense from rs-examples.sln the program '[15112] hello-realsense.exe' has exited with code 1 (0x1). But if i try to run rs-hello-realsense.exe it work.
I already try to reinstall RealSense SDK
Do you have any idea how to solve it?

akhmeevaam, 2020-09-11:
If i run hello-realsense.exe which I was built from rs-examples.sln it doesn't work.
It gives me this error:
`C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64>align.exe -c
RealSense error calling rs2_pipeline_wait_for_frames(pipe:00000263C25E9A10):
    Frame didn't arrive within 15000`

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense