realsenseai/librealsense #7307: Incorrect depth image

xapymam6ypoji, 2020-09-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  Linux raspberrypi 5.4.51-v7l+       | 
| Platform                                 | Raspberry Pi |
| SDK Version                            | 2.36.0                         | 
| Language                            |  C#                      | 
| Segment			|  Robot                    | 

### Issue Description

I have a D435i camera. Connected to the Raspberry Pi (Rasbian). Realsenseviewer works correctly. I am using Intel.RealSenseWithNativeDll library (v2.36.0)  in my c # project:

			var cameraContext = new Context();
            var pipeline = new Pipeline(cameraContext);
            using var frames = pipeline.WaitForFrames();
            var depthFrame = frames.DepthFrame.DisposeWith(frames);
            var colorizer = new Colorizer();
            var colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames);
            var image = new Bitmap(640, 1480, 1920, PixelFormat.Format24bppRgb, colorizedDepth.Data);
            var ms = new MemoryStream();
            image.Save(ms, ImageFormat.Jpeg);
			
I get this image:
[image]

Image via Realsenseviewer on Rasbian:
[image]

When I run this code on Windows, I get the correct image.
What am I doing wrong?

MartyG-RealSense, 2020-09-12:
Hi @xapymam6ypoji   My research of your case found a past case with the same conditions: written in C#, works fine in the Viewer and in Windows but not on Linux, and has a strange section on the right side that is a duplication of the left side of the image.

https://github.com/IntelRealSense/librealsense/issues/3796

That project was running on an Odroid device with the Ubuntu operating system, so Raspberry Pi / Raspbian may not be a factor in why it is occuring in your case.  That case also occurred in SDK 2.20,0, so your use of SDK 2.36.0 may not be a cause either.

The RealSense user in that case and a RealSense team member who answered it both thought that it may be related to use of .NET Core, as that project was using .NET Core instead of .NET Framework, and the RealSense team member did not experience the problem when testing in .NET Framework.

Is your project using .NET Core please @xapymam6ypoji ?

xapymam6ypoji, 2020-09-12:
Yes, i use .NET Core 3.1

xapymam6ypoji, 2020-09-12:
Now I managed to get the same problem on Windows (.NET Core 3.1 application). If I set the Pixelformat to Format32bppRgb I get the wrong image:

var image = new Bitmap(640, 480, 1920, PixelFormat.Format32bppRgb, colorizedDepth.Data);

xapymam6ypoji, 2020-09-12:
Maybe there is a way to save the image to a file without using System.Drawing, to localize the problem? I think the problem is in libgdiplus

MartyG-RealSense, 2020-09-12:
Please do keep updating your progress in the comments.  Good luck!

xapymam6ypoji, 2020-09-12:
The problem is in system.drawing, or libgdiplus. I used imagesharp library for rendering and got the correct image.
Sample code:
            using var context = new Context();
            var pipeline = new Pipeline(context);

            var cfg = new Config();
            cfg.EnableStream(Stream.Depth, 640, 480);
            pipeline.Start(cfg);

            using var frames = pipeline.WaitForFrames();
            var depthFrame = frames.DepthFrame.DisposeWith(frames);

            var colorizer = new Colorizer();

            var colorizedDepth = colorizer.Process(depthFrame);

            var bytes = 1920 * 480; //dtride * height
            byte[] rgbValues = new byte[bytes];

            Marshal.Copy(colorizedDepth.Data, rgbValues, 0, bytes);

            var image = SixLabors.ImageSharp.Image.LoadPixelData<Rgb24>(rgbValues, 640, 480);

            var ms = new MemoryStream();
            image.Save(ms, new JpegEncoder());

xapymam6ypoji, 2020-09-12:
Also, this rendering option helped (system.drawing):

```
public static Bitmap ToBitmap(this Frame self, int width, int height)
        {
            var bytes = 1920 * 480;
            byte[] rgbValues = new byte[bytes];
            Marshal.Copy(self.Data, rgbValues, 0, bytes);
            var bitmap = new Bitmap(width, height, PixelFormat.Format24bppRgb);
            var boundsRect = new Rectangle(0, 0, width, height);
            BitmapData bmpData = bitmap.LockBits(boundsRect,ImageLockMode.WriteOnly,bitmap.PixelFormat);
            IntPtr ptr = bmpData.Scan0;
            int skipByte = bmpData.Stride - width * 3;
            byte[] newBuff = new byte[bytes + skipByte * height];
            for (int j = 0; j < height; j++)
            {
                Buffer.BlockCopy(rgbValues, j * width * 3, newBuff, j * (width * 3 + skipByte), width * 3);
            }
            Marshal.Copy(newBuff, 0, ptr, newBuff.Length);
            bitmap.UnlockBits(bmpData);
            return bitmap;
        }
```
Taken from here:
https://stackoverflow.com/questions/47918451/creating-a-bitmap-from-rgb-array-in-c-result-image-difference/47919041

MartyG-RealSense, 2020-09-12:
Great, thanks for sharing your code with the RealSense community!   Do you require further assistance please or is your problem solved?

xapymam6ypoji, 2020-09-13:
This is not a solution, but a workaround. This is missing in the examples. In the current task, I will apply this solution, but I cannot assess its correctness

MartyG-RealSense, 2020-09-13:
As it is an apparent problem with .NET Core that does not occur when using .NET Framework and it has been replicated by others in https://github.com/IntelRealSense/librealsense/issues/3796 I will file a bug report with Intel so that the RealSense team can investigate it.  This case should remain open in the meantime.

xapymam6ypoji, 2020-09-13:
.NET Core application works correctly on Windows. Something is wrong when building System.Drawing for linux, the library behavior is different. On Linux, system.drawing requires an additional libgdiplus library

xapymam6ypoji, 2020-09-13:
On windows (.NET Core):
var image = new Bitmap(640, 480, 1920, PixelFormat.Format32bppRgb, colorizedDepth.Data); // Not correct image, Like on Linux.

var image = new Bitmap(640, 480, 1920, PixelFormat.Format24bppRgb, colorizedDepth.Data); // Correct image

On Linux, both options give an incorrect image

abhijaysingh, 2021-08-24:
Hi, is there any update on this? I am facing a similar issue with saving color images on Linux using .NETCore.

MartyG-RealSense, 2021-08-24:
Hi @abhijaysingh   Intel updated a week ago about this issue that it is not a fault and is an issue relating to the C# runtime libraries on Linux.

MartyG-RealSense, 2021-12-22:
Case closed due to Intel conclusion that the issue is not a fault.

closed 2021-12-22 by MartyG-RealSense