realsenseai/librealsense #7309: get aligned frames intrinsic (fx, fy, cx, cy) from recorded bag file

anguyen216, 2020-09-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.36.0) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| Language                            |  {python }                          | 

### Issue Description

I have a bag file with code to align color frames to depth frames.  I want to write code in python to extract intrinsic parameters of each aligned frame (in this case, I assume it'd be the intrinsic of depth frame since I'm aligning color to depth), namely fx, fy, cx, cy.  How do I do this? I need this information to extract X, Y, Z coordinates from disparity/depth maps

anguyen216, 2020-09-12:
**Update**
Following the code in this [link](https://github.com/IntelRealSense/librealsense/issues/2869?fbclid=IwAR2hbE15zE8JoIYLGXp4DtngLTixM5NH8aZwUDDcw-rQLjtuL_Xr4pI0HiE) I was able to extract the intrinsic of each frame.  In the case of the example bag file I'm using, the intrinsic parameters are the same for all frame; is it reasonable?

One weird thing, the width and height parameters returned are only half the value of the true sizes.  For example, the observed width and height of the frames I'm extracting is 480x848 (row by column).  However, `get_intrinsics()`  returned 240x424 for height and width, respectively.  Is there something wrong or these values are expected?

MartyG-RealSense, 2020-09-12:
Hi @anguyen216   The decimation filter does indeed scale down the image, doing so with both the width and height in order to maintain the **aspect ratio** of the adjusted image compared to the original.  The amount that the image is scaled down depends on the **magnitude** that the filter is set to.  The default is '2', causing the scale to be halved.

The **Decimation Filter** section of the Intel documentation on post-processing in the link below describes the decimation filter scaling.

https://dev.intelrealsense.com/docs/post-processing-filters#section-decimation-filter

Regarding how the ply export in rs-convert works: in a past question about this, Dorodnic the RealSense SDK Manager pointed to an SDK file called **archive.cpp**:

https://github.com/IntelRealSense/librealsense/blob/master/src/archive.cpp#L48

I've edited the line in the script that the link points to as the script will likely have changed a lot since the original link to it was created in November 2018.  The original comment from which the above information was sourced is here:

https://github.com/IntelRealSense/librealsense/issues/2671#issuecomment-435777432

anguyen216, 2020-09-13:
Hi @MartyG-RealSense, thank you for responding.  I have another question; I was under the impression that each frame would have slightly different intrinsic values, especially if the camera was not mounted fixed.  Is my assumption wrong, and it's reasonable that all frames have exactly the same intrinsic values?

MartyG-RealSense, 2020-09-13:
Official documentation about librealsense's intrinsic parameters can be found in the link below:

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#intrinsic-camera-parameters

Regarding the question of whether intrinsics can change, I will refer that query to @dorodnic ... intrinsics can be calibrated, though it requires a special OEM version of the Dynamic Calibrator tool supplied with the 1500 dollar **OEM Calibration Target** system. 
The normal version of the Dynamic Calibrator software only calibrates extrinsics, as they have the most influence on the image output.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

anguyen216, 2020-09-15:
@MartyG-RealSense Thank you for the documentation.  I have a follow-up question after reading the documentation about librealsense's intrinsic parameters.  There are only 3 supported distortion models (including None) according to the documentation.  However, the intrinsic parameters returned to me (shown again below) have `model: 4`.  What does that mean?
```
>>> width: 424, height: 240, ppx: 212.405, ppy: 121.909, fx: 212.523, fy: 212.523, model: 4, coeffs: [0, 0, 0, 0, 0]
```

MartyG-RealSense, 2020-09-15:
Hi @anguyen216   I researched your question about model:4 extensively but could not find any information, unfortunately.

closed 2020-09-26 by MartyG-RealSense