realsenseai/librealsense #7310: Failed Installing librealsense on Jetson Nano

danielcjacobs, 2020-09-11:
I am trying to get a realsense D435 camera running with ROS on a Jetson Nano.

The Jetson is running Ubuntu 18. I tried following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md), but the dkms package can't be found when I try to install it.  I saw the same issue in #7082.  I thought the Jetson was running an x64 architecture and should work, but it does not.  

Furthermore, after successfully running the rest of the installation instructions, when I run `modinfo uvcvideo | grep "version:"` , the output doesn't show a `realsense` string.

I also tried another method of installation, using the bash scripts at [this](https://github.com/JetsonHacksNano/installLibrealsense) repo to install librealsense.  It seemed to work, so I installed the ROS wrapper using [this](https://github.com/JetsonHacksNano/installRealSenseROS) repo.  I tried to test [this](https://github.com/IntelRealSense/librealsense/tree/master/examples/C/depth) example by making a ROS package and node, but when I run `catkin_make`, it can't find any of the function definitions (i.e. `rs2_create_sensor`, located in librealsense/src/rs.cpp), only the declarations.  

I don't see rs.cpp anywhere on my machine, and I'm wondering why it wasn't installed, or if something else is wrong.

MartyG-RealSense, 2020-09-11:
Hi @danielcjacobs  Jetson boards use the **Arm** processor architecture instead of x86 / x64.

I recommend trying the librealsense and ROS installation guides for Nano from the **JetsonHacks** website, as they make use of scripts to simplify the installation process.  You could first use the librealsense guide, and then the ROS guide once the librealsense install has been successful.

**Librealsense**
https://www.jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano/

**ROS**
https://www.jetsonhacks.com/2019/10/23/install-ros-on-jetson-nano/

danielcjacobs, 2020-09-14:
This issue is resolved.  I had used the jetsonhacks guides before but encountered issues.  We reflashed the drive and went through the guides again and it works now!

closed 2020-09-14 by danielcjacobs

MartyG-RealSense, 2020-09-14:
Great news, thanks for the update!