realsenseai/librealsense #7311: L515 unrecoverable illegal HW state error

jlam-rtr, 2020-09-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.03.15.100 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-115-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0 / 2.38.1 | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description

The L515 worked fine for a few weeks, but suddenly stopped functioning with the following error when opening Realsense Viewer:
`Invalid Value in rs2_get_option(options:0x7f6894000f40, option:Apd Temperature):
hwmon command 0x6a( 0 0 0 0 ) failed (response -38= Illegal HW state)`

There was no change to the sensor or the environment before this error, although the PC was cycled on / off. The serial number is displayed as ffffffffffffffff. Power cycling, firmware update and recovery have no effect.
It sometimes works briefly after a Hardware Reset, but this has only been observed once.

cja-515, 2020-09-23:
same problems as me.
Did you solve this problem?

RealSenseSupport, 2020-12-09:
Hi @jlam-rtr and @cja-515 

Are you still experiencing this issue?  Do you still have the failing HW with you to try a few things?

Have you changed out the USB cable?  Or tried other computer systems?
Is the device showing up in device manager?

Have you tried the RealSense Viewer of older SDK versions or latest SDK versions?

Camayam, 2021-01-20:
I have the same issue

depth_sensor1.set_option(rs.option.inter_cam_sync_mode, 1)
RuntimeError: hwmon command 0x19( 1 0 0 0 ) failed (response -38= Illegal HW state)

closed 2021-06-17 by RealSenseSupport