realsenseai/librealsense #7312: D455 not connecting | Python | Raspberry Pi 4

sam598, 2020-09-11:
| Camera Model                       | { D455 } | 
| Firmware Version                   | { 5.12.7.100 } | 
| Operating System & Version | { Linux raspberrypi }  | 
| Kernel Version (Linux Only)    | { 5.4.59-v7l+ } | 
| Platform                                 | { Raspberry Pi 4 } |
| SDK Version                            |  { 2.38.1 } | 
| Language                            |  {Python 3.7.3} | 
| Segment			|  { others } | 

### Issue Description

This script:
```
import pyrealsense2 as rs

ctx = rs.context()
devices = []
for i in range(len(ctx.devices)):
    sn = ctx.devices[i].get_info(rs.camera_info.serial_number)
    print(sn)
    devices.append(sn)
```

Prints a serial number when a D435 is connected to the USB 3 port.

However when a D455 is connected to the USB 3 port no devices or serial numbers are detected.

This was previously mentioned in a similar issue for Windows: https://github.com/IntelRealSense/librealsense/issues/7283

MartyG-RealSense, 2020-09-11:
Hi @sam598   The best place to post your kindly provided information will be on https://github.com/IntelRealSense/librealsense/issues/7283 as that case is the focus of the RealSense team's investigation into the reported issue of non-connection in Python with the D455.  Having all the information in one place will make it easier to analyse.  Thanks!

sam598, 2020-09-11:
Thanks Marty, I felt it would be better to open a new issue since the solution presented is #7283 is not applicable for Linux raspberrypi

MartyG-RealSense, 2020-09-11:
Okay, no problem.  If you wish to have a separate case for your particular situation, please post a message on https://github.com/IntelRealSense/librealsense/issues/7283 with a link to here so that the RealSense team members who are investigating are aware of it and can read it.  Thanks again!

sam598, 2020-09-17:
@RealSenseCustomerSupport are there any updates? Here are some more things I have tried.

- I've tried a second D435 (connects) and D455 (does not connect) and received the same results. So it is not an issue with the camera hardware.

- Running `lsusb` in the terminal lists `Bus 002 Device 005: ID 8086:0b07 Intel Corp.` when a D435 is connected and `Bus 002 Device 006: ID 8086:0b5c Intel Corp.` when a D455 is connected. So it is not the USB connection.

- Running realsense-viewer on the Raspberry Pi results in the same behavior. D435 cameras connect, while the D455 cameras do not.

- If I run the realsense-viewer through command line this warning appears whenever the D455 is connected. The D435 does not display this warning.

`16/09 23:34:58,837 WARNING [3023590464] (ds5-factory.cpp:1119) DS5 group_devices is empty.`

MartyG-RealSense, 2020-09-17:
Hi @sam598    Judging by your link to https://github.com/IntelRealSense/librealsense/issues/7283 I think that you meant to tag @RealSenseSupport instead of @RealSenseCustomerSupport  ?  :)

sam598, 2020-09-18:
@MartyG-RealSense both @RealSenseSupport and @RealSenseCustomerSupport have responded to #7283 so I assume they are interchangeable.

I have also tried compiling with the additional cmake flags:
`cmake .. -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DFORCE_LIBUVC=true`

But realsense-viewer displays the following error in the terminal when any camera (D435 or D455) is connected:
```
17/09 20:50:41,697 ERROR [2890912832] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 17/09 20:50:41,717 WARNING [2917364800] (rs.cpp:304) null pointer passed for argument "device"
```
and this error appears in a pop-up message:
```
UNKNOWN in rs2_create_device(info_list:0x2b5d4f8, index:0):
failed to set power state
```

RealSenseSupport, 2020-09-18:
Hi @sam598 
1. I upgraded my Raspbian from stretch to buster that has newer version to 5.4.64
and build the SDK from source using below cmake command

cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_NETWORK_DEVICE=ON

*-DFORCE_RSUSB_BACKED replace DFORCE_LIBUVC=true, so you can remove it
I add -DNETWORK_DEVICE=ON, because i need to do some test on rs-server later, you can remove it

before cmake, you need to install some prerequisites as mentioned at 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

under build folder
>>cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_NETWORK_DEVICE=ON

>> make -j4   # Raspberry has 4 core, this may help to shorten the compilation time
>> sudo make install
>> which realsense_viewer
>>/usr/locla/bin/realsense-viewer 

the viewer is able to detect either D435 or D455 and streaming

sam598, 2020-09-18:
@RealSenseSupport it seems to work now after `sudo rpi-update` from kernel 5.4.59-v7l+ to 5.4.65-v7l+. It is surprising that a minor Kernel bump fixes the issue (it was upgraded to 5.4.59 in August).

> I upgraded my Raspbian from stretch to buster that has newer version to 5.4.64

Are you sure you didn't upgrade to Raspberry Pi OS? Raspbian Buster has been deprecated.

> make -j4 # Raspberry has 4 core, this may help to shorten the compilation time

In my experience `make -j4` is extremely unstable on Raspberry Pis, often resulting in crashes or hour long system freezes. `make -j1` has been stable and consistent on several Pis.

RealSenseSupport, 2020-09-21:
Hi @sam598 
Great to know you are able to work now..

Buster seems the latest version , not sure why it had been deprecated? 
https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/
Thanks for sharing your experience as to the -Jx option

As to -J4, when i use it, normally I need to remove the case of my Raspberry Pi 4 to cool it down  while compiling the code

next time, I would use -J1 if I run into the crash issue or freeze issue

closed 2020-10-05 by MartyG-RealSense

sam598, 2020-10-06:
@RealSenseSupport The latest version is now called "Raspberry Pi OS"