realsenseai/librealsense #7313: Again getting "terminate called after throwing an instance of 'rs2::backend_error'" error

milan-r-shah, 2020-09-11:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.06.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-28-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.36.0)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description
In the past, I was already having the following error when I was trying to run **pcl-color** example:
```bash
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)
```

and for that, I had created #6940 . At that time, somehow I got rid of that error
1. By installing the SDK again (through installing from the source method).
2. Just to be safe, every time when my Linux 18.04 LTS boots up, manually selecting the **5.3.0-28-generic**.

But, now, I'm again getting the same error. Is there any permanent way to fix this? Is uninstalling the current SDK and reinstalling it using the **backend** method is the only option now?

milan-r-shah, 2020-09-11:
One more weird thing(s)  that I noticed/observed:
1. Generally, I'm getting the above-mentioned error every time I try to run **pcl-color** example
2. But, if I open up **realsense-viewer** --> turn on **Stereo Module** --> turn on **RGB Camera** --> turn off those two --> close the viewer and then if I try to run the "pcl-color" example then it works (I think most of the times). 
3. If I just open up realsense-viewer and close it (i.e. without turning on.off Stereo Module and RGB Camera) then running the "pcl-color" example gives the same error!

It looks like step 2 resets some sort of internal flag!

milan-r-shah, 2020-09-11:
One more thing is why running **pcl-color** example gives error most of the time except for a few random exceptions? If there is something wrong then it should give error every single time, right?

milan-r-shah, 2020-09-12:
Another point: I noticed that this error occurs frequently if I open or have opened **realsense-viewer** recently. 
Afterward, I keep getting this error multiple times, and then later on at some point in time, randomly, it would run without any error for 2-3 times and would again start giving the same error! 
At least, as of now, it looks like somehow **realsense-viewer** is affecting or changing something. But, later on, how it gets fixed... automatically... for very few numbers of times?

milan-r-shah, 2020-09-12:
Update: I noticed that after getting the above-mentioned error for the **pcl-color** example, if I run & stop any other example (I have tried "rs-hello-realsense") in a different terminal for a few times and then if I try again running the "pcl-color" example in the original terminal, it works (for 2-3 times).
In case, if it doesn't then I try doing the first thing i.e. running & stopping any other example code for a few times and then try again for "pcl-color" and it works!

But, as you can imagine, this is quite boring/frustrating and not an actual solution. 

Please suggest any way to fix this **rs2::backend_error** problem permanently. Thanks in advance!

RealSenseSupport, 2020-09-18:
Hi @milan-r-shah 

Can you please update the versions of FW and SDK to the latest ones and let us know if you still see the same issue?

Thank you!

RealSenseSupport, 2020-09-25:
@milan-r-shah 

Do you still need help with this question?

milan-r-shah, 2020-09-25:
Hi @RealSenseSupport ,

For updating the FW, I know the steps but could you please let me know how can I update the SDK? 

I found steps in [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) but those steps are for pre-built packages. On the other hand, I had installed the SDK using [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) i.e. building from source. Over there, I couldn't find any steps on how to upgrade the SDK.

RealSenseSupport, 2020-12-05:
Hi,

Are you still experiencing issues with this? Have you tried starting fresh by removing all RealSense™ SDK-related packages with:

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

and then install the latest SDK as per here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Thanks

RealSenseSupport, 2020-12-15:
Hi,

Were you able to give that a try? Did that work for you or will you be needing further help?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

milan-r-shah, 2020-12-16:
Hi @RealSenseSupport 

Apologies for the late reply. After updating the firmware as well as the SDK (through **backend** method), I'm no longer getting the above error. However, is this backend method not very stable? Because sometimes while trying to launch the SDK from the terminal, I get **Segmentation fault (core dumped)** error:
```
$ realsense-viewer
15/12 21:21:18,376 ERROR [140267424298752] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 15/12 21:21:18,378 WARNING [140267441084160] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 15/12 21:21:18,378 WARNING [140267441084160] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 15/12 21:21:18,379 WARNING [140267441084160] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
Segmentation fault (core dumped)
```
---

One more thing: now, I'm facing a somewhat similar issue with **Intel RealSense LiDAR Camera L515**. Upon running `rs-pcl-color` for **L515**, I'm getting this error:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Aborted (core dumped)
```

or sometimes, this one:
```
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
Aborted (core dumped)
```
For the errors with L515, I would create another issue!

RealSenseSupport, 2021-03-24:
Hi,

Apologies for the late reply as well. It sounds like it might be easier to clear your system of all previous installs and start fresh than to try and root cause what might have went wrong. Is this viable?

Have you tried to follow the method here?
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Or build the SDK from source?

Thank