realsenseai/librealsense #7314: Got this error after adding a button

hduonggithub, 2020-09-11:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - D:\dev\librealsense\common\viewer.cpp:1932

Hi All,

I have modified the code to add 1 button next to the Export button in GUI (viewer.cpp). I ran in a debug mode. My button executed python code on an external window smoothly. Everything went fine, until I stopped my visual studio and re-ran again, it shows me that error. I also tried to use **the original code**, it still gives me the same error.

Any help/suggestion is really appreciated! Thanks!
```

MartyG-RealSense, 2020-09-12:
Hi @hduonggithub   The **Assertion failed** dialog box seems to be a Visual Studio debugging function.

https://docs.microsoft.com/en-us/visualstudio/debugger/assertion-failed-dialog-box?view=vs-2019

As suggested in the link above, does clicking the **Ignore** button on the dialog box enable the program to continue running normally?

If your program is being built as a Debug build, does the problem still occur if it is built as a Release build?

https://docs.microsoft.com/en-us/visualstudio/debugger/how-to-set-debug-and-release-configurations?view=vs-2019

hduonggithub, 2020-09-12:
Hi @MartyG-RealSense 

Thanks for your prompt reply. 

It works in Release mode and now switch back to debug mode, it also works. 

I just want to add more info that there could be a bug when RealSense Window is in "zero" size as in picture.
[image]

if continue to hit Ignore, this is 2nd error 
[image]

MartyG-RealSense, 2020-09-12:
I'm glad to hear that you have had succes with your application.  Regarding your theory that having the Viewer window minimized may cause an error pop-up, I will tag the RealSense SDK Manager into this session.

@dorodnic  Do you think that having the Viewer window minimized n the background could have an effect on an application that is using a button custom-added into the RealSense GUI interface?

hduonggithub, 2020-09-12:
Hi @MartyG-RealSense and @dorodnic 

Sorry for another question (I can make another ticket if you prefer). Is there any better way to read and display point cloud from files (.pcd, .ply, etc) files into 3D RealSense viewer. I know we feed frame and calculate point cloud to display in 3D mode in real time. I am thinking to add a such function that can display point cloud into 3D RealSense viewer as well as processing later on. Thanks

MartyG-RealSense, 2020-09-12:
@hduonggithub   Whilst you can drag and drop bag files into the center panel of the Viewer in 2D mode, doing so in 3D mode with ply files produces an error as it is not supported.

[image]

Ply files are recognised by 3D modelling software.  They sometimes require a compatibility plugin to do so.  On Windows though you can double-leftclick on a ply file to open it as a 3D model in the Microsoft **3D Viewer** app (which should be built into Windows but can be obtained for free online from the Microsoft Store if it isn't).  An open source program called MeshLab can also read ply files created in RealSense.

https://www.meshlab.net/
https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

RealSense users are encouraged to make use of third-party processing libraries such as PCL to perform further work on ply files, as those kind of functions are their primary focus and so those platforms can create and support such tools very effectively.  Having said that, RealSense users are certainly welcome and encouraged to develop their own expansions for RealSense.

hduonggithub, 2020-09-12:
Hi @MartyG-RealSense 

I know the RealSense viewer does not support it yet as I have asked for a guideline to extend it. As RealSense already has 3D viewer, we may just need a) function to read point cloud file (ply) which can be used from any open source library and b) draw point cloud on the 3D viewer or c) convert to the right format in a RealSense workflow for visualization.

MartyG-RealSense, 2020-09-12:
I was working on a case about how the **rs-convert** tool converts bag data to ply format.  Apparently a section of the **archive.cpp** SDK file handles the ply export process.  Working through the process in reverse might provide some useful insights about ply to RealSense data conversion.

https://github.com/IntelRealSense/librealsense/blob/master/src/archive.cpp#L48

hduonggithub, 2020-09-14:
Hi @MartyG-RealSense 

Realsense Viewer is good for capturing & displaying data by frame for sure. I just feel regret that we cannot import point cloud from file and displayed in 3D RealSense Viewer like Open3D Viewer from http://www.open3d.org/. More processing results can also be displayed if we can utilize this viewer.

MartyG-RealSense, 2020-09-14:
Yes, Open3D is apparently useful for importing and rendering ply files that were generated in RealSense.  Here are some example discussions about this:

https://github.com/IntelRealSense/librealsense/issues/4640
https://github.com/intel-isl/Open3D/issues/473

MartyG-RealSense, 2020-09-21:
Hi @hduonggithub   Do you require further assistance with this case, please?   Thanks!

hduonggithub, 2020-09-22:
Hi @MartyG-RealSense 
Still waiting to hear any advice to import pointcloud into Realsense Viewer. Thanks for your suggestion anyway. I do not mind if you want to close it.

MartyG-RealSense, 2020-09-22:
The developers of dedicated libraries such as PCL and Open3D are laser-focused on creating the best possible tools that are dedicated to the specific purpose of mesh processing and nothing else.  By providing compatibility wrappers for these tools, it allows RealSense to be focused on what it does best whilst integrating the capabilities of the third-party mesh processing libraries.  

An official ply importer built into the RealSense Viewer is therefore unlikely to be implemented.  There is nothing to prevent the RealSense community from modifying the open-source Viewer code to implement such a ply viewer in a custom Viewer program themselves, as you have been hoping to do.  

There is an OpenCV module called **3D Viewer** for visualizing data that has a **cv::viz::Mesh** function for loading a mesh from a ply file.  As the RealSense SDK has full OpenCv compatibility, that may be a possible route for integrating the **3D Viewer** into a custom RealSense Viewer.

https://docs.opencv.org/master/d1/d19/group__viz.html

MartyG-RealSense, 2020-10-02:
Hi @hduonggithub   Do you still require assistance with this case, please?  Thanks!

hduonggithub, 2020-10-02:
@MartyG-RealSense  You can close this. Thanks for help!

MartyG-RealSense, 2020-10-02:
Thanks for the update!

closed 2020-10-02 by MartyG-RealSense