realsenseai/librealsense #7316: rs-gl when should I use it?

arothenberg, 2020-09-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
Dorodnic's rs-gl, When do I want to use it? 

I'm pretty deep into the sdk code at this point and I know what I want to do (I already did some of it in the old kinect platform) but I have some big gaps in my knowledge about the low level stuff. 
 I need to take a frame , analyze the depths, render and show a  texture related to the depth values. I need to do this quickly because ,ideally, it is live feedback reading "fast" motion (we'll see how fast is allowed). 

I'm wondering if it worthwhile for me to utilize the gpu as per dorodnics extension(?) ?  Would this be something to learn?

MartyG-RealSense, 2020-09-11:
Hi @arothenberg   The link below has a very good pros and cons analysis of whether to use GLSL processing blocks and where it can be used.

https://github.com/IntelRealSense/librealsense/pull/3654

arothenberg, 2020-09-11:
That helped. It looks easy to implement so I might as well try it and benchmark it. Thanks Marty.

closed 2020-09-12 by arothenberg