realsenseai/librealsense #7317: D455 mounting points are M4 not M3

sam598, 2020-09-11:
The technical specs listed on the store page here:

https://www.intelrealsense.com/depth-camera-d455/

state:

```
Mounting Mechanism:
– One 1/4‑20 UNC thread mounting point.
– Two M3 thread mounting points.
– Tripod
```

On the actual hardware the two rear mounting points are M4 threads.

MartyG-RealSense, 2020-09-12:
Hi @sam598   I have sought advice from Inrel about your question.  Thank you for your patience.

MartyG-RealSense, 2020-09-21:
Thanks for your patience in waiting for a response to this question about the mount hole size @sam598

MartyG-RealSense, 2020-09-22:
Hi @sam598  I was provided with confirmation that the mounting holes of the D455 are larger in order to support the greater weight of the camera compared to previous models but did not receive information about the M3 / M4 size question.  The best advice I can offer is that since all of the information sources for D455 states that the mounting points are M3 then it is the official position that they are M3.

sam598, 2020-09-22:
Hi @MartyG-RealSense, thank you for the response, however I think you misunderstand the issue. It is not a question, it is a statement.

M3 and M4 are a type of standard screw thread:
https://en.wikipedia.org/wiki/ISO_metric_screw_thread#:~:text=The%20ISO%20metric%20screw%20threads,general%2Dpurpose%20screw%20thread%20worldwide.&text=The%20%22M%22%20designation%20for%20metric,outer%20diameter%20of%206%20millimetres).

The threads on the previous D400 series were M3. The threads on the D455 are M4.

I am posting this issue because the technical specifications on the D455 store page are incorrect and need to be update, so that your customers do not make mistakes while buying or creating support systems for their camera mounts.

RealSenseSupport, 2020-09-22:
Hi @sam598 

Thanks for pointing this out.  We will definitely review the datasheets and the order site to make sure the data is correct as per the D455 camera.  We will make the modifications as needed and notify you when updates have been done.

MartyG-RealSense, 2020-10-02:
Hi @sam598   I will label this case with the documentation tag to ensure that it is kept open until Intel confirm their review to be completed.

MartyG-RealSense, 2020-11-07:
Hi @sam598   I followed up on this case, and the RealSense Store and data sheet details for the D455 now show the hole size as M4.

**Store**

[image]

**Data Sheet**

[image]

MartyG-RealSense, 2020-11-15:
Case closed due to answer provided and no further comments received.

closed 2020-11-15 by MartyG-RealSense

victoriathomasinvia, 2022-01-24:
Ok but what is the thread pitch on the M4 screws? 
Is it M4-0.7?

@sam598 Did you figure out the thread pitch?

MartyG-RealSense, 2022-01-24:
Hi @victoriathomasinvia   The 'M' screws are an **ISO** standard, meaning that M4 screws should always have a 0.7 mm thread pitch even if it is not stated.

https://www.accu.co.uk/en/p/117-iso-metric-thread-dimensions