realsenseai/librealsense #7318: wait_for_device can't detect camera plugin

WeixiangZhang, 2020-09-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            | 2.38                           | 
| Language                            |  C/C++                          | 
| Segment			|  others                           | 

### Issue Description
First run the code and then plugin camera, but wait_for_device doesn't return, it blocks all the time.
```c++
rs2::context context; 
rs2::device_hub hub(context);
hub.wait_for_device();
```

How does RealSense Viewer detect camera plugin??

WeixiangZhang, 2020-09-12:
Use `query_devices` instead
```c++
        rs2::device_list deviceList;
	while (deviceList.size() == 0) {
		deviceList = mRSContext.query_devices();
	}
```

MartyG-RealSense, 2020-09-12:
Hi @WeixiangZhang   If you wanted to look for any connected camera then you could try the code below:

**rs2::context context;
rs2::device dev = get_device_from_context(context);
rs2::device_hub hub(context);
dev = hub.wait_for_device();** 

If you wanted to insert a hardware_reset() instruction then the code would look like this:

**rs2::context context;
rs2::device dev = get_device_from_context(context);
dev.hardware_reset();
rs2::device_hub hub(context);
dev = hub.wait_for_device();**

WeixiangZhang, 2020-09-14:
@MartyG-RealSense Thank you.

closed 2020-09-14 by WeixiangZhang