realsenseai/librealsense #7321: L515 low color fps

suhangpro, 2020-09-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-117-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1-0~realsense0.3378                          | 

### Issue Description

The framerate of the color stream reported in various applications (guvcview, realsense viewer, etc.) is only around 18fps no matter what the target fps it is set at. depth and infrared fps is steady at 30 fps. According to the specs, the color stream should be able to achieve 60/30 fps depending on the resolution. Any suggestion what might be the the issue? 

This was discovered as the cause of #7298

RealSenseSupport, 2020-09-16:
Hi @suhangpro 

I wonder if you're actually experiencing this...

https://github.com/IntelRealSense/librealsense/issues/1957

Can you try turning off auto-exposure and see if that changes things?

Thanks

suhangpro, 2020-09-16:
Thanks for the pointer. It is exactly the cause! Disabling "Auto Exposure Priority" gets me the target 30/60 fps with a slight drop in image quality.

RealSenseSupport, 2020-09-16:
Fantastic. Can we call this issue closed?

suhangpro, 2020-09-16:
Issue resolved.

closed 2020-09-16 by suhangpro