realsenseai/librealsense #7324: I have so depth files saved in .png format by SR300, how can I restore the true depth from these png files

chky, 2020-09-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / **SR300** / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-09-13:
Hi @chky   Saving depth data into PNG image format causes depth detail to be lost, unfortunately.  A **.raw** file is the appropriate format for preserving depth detail in an image file.

The link below has scripting for obtaining depth information from a .raw file if you have the opportunity to recreate the scenes for the SR300 camera and re-save them as .raw.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331

chky, 2020-09-14:
Thanks for your reply @MartyG-RealSense. Actually I do not have a realsense sr300 in hand, what I have are some .png files. Is it possible to restore true depth without .RAW files? An approximate depth will be OK.

MartyG-RealSense, 2020-09-14:
I researched your question extensively in case there was a possible way, but every source says it will not work.  Even if you could convert a color on the image to a depth value representing that color, you could not be sure that it is accurate.

https://github.com/IntelRealSense/librealsense/issues/3640

chky, 2020-09-14:
would you mind provide me with links to these possible ways? much appreciation.

MartyG-RealSense, 2020-09-14:
I mean that I searched for possible ways but found that it was not practical.  Nonetheless, the best of the resources that I found other than https://github.com/IntelRealSense/librealsense/issues/3640 is below.  You probably will not get a meaningful depth value though since a PNG does not store depth data.

https://github.com/IntelRealSense/librealsense/issues/2200#issuecomment-411754323

Alternatively, the link below suggests a somewhat complex way to generate a depth map from a single image using a 'deep network'.

https://stackoverflow.com/a/41830992

chky, 2020-09-14:
got it! thanks a lot

MartyG-RealSense, 2020-09-21:
Hi @chky  Do you require further assistance with this case, please?   Thanks!

chky, 2020-09-22:
> Hi @chky Do you require further assistance with this case, please? Thanks!

Your reply had solved my problem, thank you!

MartyG-RealSense, 2020-09-22:
Great news, thanks for the update!   I will close this case now that you have a solution.

closed 2020-09-22 by MartyG-RealSense