realsenseai/librealsense #7328: Ubuntu 18.04 with SR300, turning camera off then back on causes "Frames didn't arrived within 5 seconds"

Belval, 2020-09-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |   5.4.0-47-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.38.1 }                          | 
| Language                            | N/A and Python | 
| Segment			| N/A | 

### Issue Description

When using the SR300 I cannot activate/deactivate it without having it enter an error state which requires me to powercycle (i.e. unplug/replug) the device.

[image]

#### How to reproduce

1. On a clean install of Ubuntu 18.04, follow the installation instructions [here]().
2. Open Intel RealSense Viewer
3. Turn on the camera/coded-light depth sensor 
4. The image is visible and the output is good
5. Turn off the camera/coded-light depth sensor
6. Turn on the camera/coded-light depth sensor 
7. No image, with error `(backend-v4l2.cpp:1034) Frames didn't arrived within 5 seconds`

#### What I have tried to fix the issue

1. Initially I was on Ubuntu 20.04. I downgraded to Ubuntu 18.04 to be "officially supported". Both platform had the same issue.
2. Tried to use the SR300 with the pyrealsense2 Python module, same error happens, whether I pipeline.stop() or not.

#### Temporary fix

If I unplug and replug the USB cable the camera can be reactivated and will work as expected without rebooting. However this is very painful when debugging.

MartyG-RealSense, 2020-09-14:
Hi @Belval   I carefully considered your case.  It does not seem like a problem that is likely to be related to erratic power, as it sounds as though your camera is stable other than when this problem occurs specifically when toggling the depth sensor on-off-on.

On point 1 of your kindly provided 'How to reproduce' guide, the link to the installation method that you used goes to this page here instead.  Could you confirm which method you used to install librealsense, please?

Because your log was checking the status every 5 seconds it made me wonder if it was the **RSUSB** method, as it checks device changes every 5 seconds.  

https://github.com/IntelRealSense/librealsense/issues/6921#issuecomment-668652875

Belval, 2020-09-14:
Sorry about the wrong link, here is the right one (I also updated it): https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

I will try the possible solutions that you mentioned.

Belval, 2020-09-14:
I can confirm that adding:

```py
    # Reset all devices
    ctx = rs.context()
    devices = ctx.query_devices()
    for dev in devices:
        dev.hardware_reset()
```

Fixes the issue for my python script, thank you for your help.

Feel free to close the issue, but it does persist in RealSense viewer.

MartyG-RealSense, 2020-09-14:
Thanks for the update - I'm very pleased that you made progress!  

To avoid unplug-replug in the remaining issue with the Viewer, you could see if you can do a hardware reset in the Viewer at the start of each session of using it.  This is peformed by going to the **More** option at the top of the options side-panel and selecting the hardware reset option from its menu.

Belval, 2020-09-15:
That works, thank you for your awesome support!

closed 2020-09-15 by Belval

MartyG-RealSense, 2020-09-15:
You are very welcome - I'm glad I could help.  :)   Thanks for the update!