realsenseai/librealsense #7333: kernel problem

gaotao19890725, 2020-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            | 2.35.2                      | 
| Language                            |  C++                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Build Instruction
$ mkdir build
$ cd build
$ cmake
$ make 

When I excute the followting command: ./scripts/patch-realsense-ubuntu-lts.sh, then the following error occurs 

Unsupported kernel version 4.9.140-tegra . The Bionic patches are maintained for Ubuntu LTS with kernels 4.15, 4.18, 5.0 and 5.3.

I do not know how to solve the problem, can you give me some suggestions. thanks a million

MartyG-RealSense, 2020-09-14:
Hi @gaotao19890725   There is an official guide for installing librealsense on Jetson boards that include TX2 in the link below.  I recently used this guide to successfully install librealsense on a TX2.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

gaotao19890725, 2020-09-14:
Dear MartyG-RealSense:
       Thanks for your quick reply. And I make it according to your suggestion.
       
Tao Gao

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

MartyG-RealSense, 2020-09-14:
Okay @gaotao19890725   Please let us know whether you were successful in installing librealense on TX2 using the official guide that I linked to.  Thanks!

gaotao19890725, 2020-09-14:
Dear MartyG-RealSense：
       In the terminal, I input the following commands one by one:
      (1)sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80            --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE

     (2)sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u

     (3)sudo apt-get install librealsense2-utils
     (4)sudo apt-get install librealsense2-dev

     (5)realsense-viewer

     Finally, I get the following picture
Tao Gao

MartyG-RealSense, 2020-09-14:
Sorry, your picture is not shown in the comment posted to the GitHub by email.  Could you describe what is in the picture of what happens when you launch the RealSense Viewer?

gaotao19890725, 2020-09-14:
When I launch the realsense-viewer, the camera realsense D435i runs and in the  picture it is me, for i am in front of the camera.

MartyG-RealSense, 2020-09-14:
Thank you very much.  Does this mean that the camera is now working correctly for you, please?

gaotao19890725, 2020-09-14:
But I meet another question, the project ElasticFusion can not work as before after my installation of RealSense2, and I do not know whether the RealSense2 conflicts with the project ElasticFusion or not.

MartyG-RealSense, 2020-09-14:
ElasticFusion should not conflict with librealsense, since it has optional compatibility with librealsense that is enabled by just having librealsense installed.

The ElasticFusion documentation suggests that dependencies (such as librealsense) should be installed first and when you have all of the dependencies installed, you should build the **Core** of ElasticFusion and then build the **GUI** afterward.

So maybe installing librealsense when ElasticFusion was already installed overwrote something important and so the ElasticFusion Core and GUI need to be built again.

MartyG-RealSense, 2020-09-21:
Hi @gaotao19890725   Do you require further assistance with this case, please?  Thanks!

gaotao19890725, 2020-09-23:
Dear MartyG-RealSense：
     Thanks a million for your guide and assitance, and sorry for my late response. A few days ago, I flashed my TX2 and installed librealense on TX2 successfully again according to the following site
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md 
In additon, I rebuilt the ElasticFusion and it can deal with the .klg file provided by the author mp3guy. Now I am trying to connect realsenseD435i with ElasticFusion，but no result. In other words, I have not figured out how to make a connection.

MartyG-RealSense, 2020-09-23:
I hope that you can find useful information in the link below about connecting RealSense to Elasticfusion:

https://github.com/mp3guy/ElasticFusion/issues/164

gaotao19890725, 2020-09-26:
Dear MartyG-RealSense：
      I can roslaunch realsense2_camear rs_camera.launch, but can not roslaunch realsense2_camera rs_rgbd.launch. Can you give me some suggestions.
      Hope you are well.

MartyG-RealSense, 2020-09-26:
Could you try one of the install instructions below please depending on whether you are using ROS Kinetic or Melodic:

**Kinetic**
sudo apt install ros-kinetic-rgbd-launch

**Melodic**
sudo apt install ros-melodic-rgbd-launch

gaotao19890725, 2020-09-26:
Hi MartyG-RealSense：
    I use ROS Melodic, after sudo apt install ros-melodic-rgbd-launch, I can roslaunch realsense2_camera rs_rgbd.launch. But when I rosrun ORB_SLAM3 RGBD PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE, the following picture appears: nothing shows. 
   can you see the picture?
[image]

> 
[image]

gaotao19890725, 2020-09-26:
Hi Hi MartyG-RealSense：
By the way, the link below about connecting RealSense to ElasticFusion helps:

mp3guy/ElasticFusion#164

but when I typed ‘ccmake ../’, then the following picture appears. But I do not know the meaning of “turning on Realsense option”. Where is the realsense?
[image]

gaotao19890725, 2020-09-26:
Hi MartyG-RealSense：
     With your help, I make it. The following picture is about ORB-SLAM2 with online RealSenseD435i.
     
[image]

      Thanks again

MartyG-RealSense, 2020-09-26:
Great news - thanks so much for the update and sharing of your image!

gaotao19890725, 2020-09-27:
Hi MartyG-RealSense：
Another good news.  I make it about ElasticFusion with RealSenseD435i on TX2: The following picture can be proved, thanks for your  unselfish sharing and hope you are well. If there is something wrong, I will disturb you again.
[image]

MartyG-RealSense, 2020-09-27:
It is no trouble at all @gaotao19890725   :)

closed 2020-10-05 by MartyG-RealSense