realsenseai/librealsense #7336: displaying .Ply image in c#

Xander-Vreeswijk, 2020-09-14:
I have the point cloud from the c# cookbook exported to a ply file. Can someone help me to display this ply image in the code from the csharp wrappers from librealsense.

MartyG-RealSense, 2020-09-14:
Hi @Xander-Vreeswijk   Your best option may be to create a non-RealSense function within your C# program that reads the data from the RealSense ply file and converts it, so that you can generate an image within your program.  The link below provides some guidance about how to do this.

https://stackoverflow.com/questions/58916236/reading-binary-ply-file-face-indices-using-c-sharp

Xander-Vreeswijk, 2020-09-17:
Thank you for your response, but I got it to work using the helix toolkit.

closed 2020-09-17 by Xander-Vreeswijk

MartyG-RealSense, 2020-09-17:
Great news - thanks for the update!