realsenseai/librealsense #7337: D435 Not working on realsense-viewer and pyrealsense2

wikpunGitHub, 2020-09-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.07.100) | 
| Operating System & Version |   Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  (5.4.0-47)                                         | 
| Platform                                 | Udoo x86.  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {/python }                          | 
| Segment			|  {Robot/ }                          | 

### Issue Description
Build from source

**Even realsense-viewer is not working.**
UNKNOWN in rs2_create_device(info_list:0x564e5301c330, index:0):
failed to set power state

**Pyrealsense Not Working**
Python 3.8.2 (default, Jul 16 2020, 14:00:26) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> pipeline = rs.pipeline()
>>> config = rs.config()
>>> config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
<stdin>:1: DeprecationWarning: an integer is required (got type pyrealsense2.pyrealsense2.format).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
>>> profile = pipeline.start(config)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D400
    Serial Number                 : 	ffffffffffff
    Firmware Version              : 	05.12.07.100
    Recommended Firmware Version  : 	05.12.07.100
    Physical Port                 : 	2-3-3
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD1
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	812113024424
    Firmware Update Id            : 	812113024424

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared	  1280x720	@ 30Hz	   UYVY
    Infrared	  1280x720	@ 30Hz	   BGRA8
    Infrared	  1280x720	@ 30Hz	   RGBA8
    Infrared	  1280x720	@ 30Hz	   BGR8
    Infrared	  1280x720	@ 30Hz	   RGB8
    Infrared	  1280x720	@ 15Hz	   UYVY
    Infrared	  1280x720	@ 15Hz	   BGRA8
    Infrared	  1280x720	@ 15Hz	   RGBA8
    Infrared	  1280x720	@ 15Hz	   BGR8
    Infrared	  1280x720	@ 15Hz	   RGB8
    Infrared	  1280x720	@ 6Hz	   UYVY
    Infrared	  1280x720	@ 6Hz	   BGRA8
    Infrared	  1280x720	@ 6Hz	   RGBA8
    Infrared	  1280x720	@ 6Hz	   BGR8
    Infrared	  1280x720	@ 6Hz	   RGB8
    Infrared	  848x480	@ 90Hz	   UYVY
    Infrared	  848x480	@ 90Hz	   BGRA8
[... 210 lines cut ...]
    Depth	  848x480	@ 30Hz	   Z16
    Depth	  848x480	@ 15Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  848x100	@ 100Hz	   Z16
    Depth	  640x480	@ 90Hz	   Z16
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
    Depth	  640x360	@ 6Hz	   Z16
    Depth	  480x270	@ 90Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  424x240	@ 90Hz	   Z16
    Depth	  424x240	@ 60Hz	   Z16
    Depth	  424x240	@ 30Hz	   Z16
    Depth	  424x240	@ 15Hz	   Z16
    Depth	  424x240	@ 6Hz	   Z16
    Depth	  256x144	@ 90Hz	   Z16

MartyG-RealSense, 2020-09-14:
Hi @wikpunGitHub    I notice that although the title says that you are using a D435 (a cased USB camera), the logs that you kindly provided state that the camera is a **D400**, which is a camera circuit board without a casing.  

**Name : Intel RealSense D400**

It should instead say "Intel RealSense D435" if it is a D435 camera.  A camera that is identified by the camera firmware as a D400 would have support missing for features such as RGB.

[image]

wikpunGitHub, 2020-09-15:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Hi Martin:I am using depth module D415. I also have a D435 and I mixed up the name. Sorry about it!Best Regards, Wik從我的 BlackBerry 傳送 - 最安全的行動裝置                                                                                                                                                                          寄件者： notifications@github.com已傳送： 2020年9月15日 上午1:52收件者： librealsense@noreply.github.com回覆： reply@reply.github.com副本： wikpun@gmail.com; mention@noreply.github.com主旨： Re: [IntelRealSense/librealsense] D435 Not working on realsense-viewer and pyrealsense2 (#7337)  
Hi @wikpunGitHub    I notice that although the title says that you are using a D435 (a cased USB camera), the logs that you kindly provided state that the camera is a D400, which is a camera circuit board without a casing.
Name : Intel RealSense D400
It should instead say "Intel RealSense D435" if it is a D435 camera.  A camera that is identified by the camera firmware as a D400 would have support missing for features such as RGB.

MartyG-RealSense, 2020-09-15:
Ok, thank you very much.  If you are using a caseless D415 depth module then you should be able to access the RGB and projector, as those components are included on the D415 depth module (the D415 is the only model in the depth module range that has an integrated RGB sensor).

The USB connection is clearly being detected as USB 3.2 and you have the full range of stream modes listed in the **rs-enumerate-devices** log that you would expect to see on a USB 3 connection.  So there does not seem to be a problem with USB bandwidth.

As the 400 Series cameras can work with any Intel or Arm processor, I do not think that your Udoo x86 board is the cause of the problem, and they seem ti have a good amount of memory RAM capacity.

I would like to eliminate the possibility of a conflict with the 5.4 kernel or Ubuntu 20.04 as the cause of the problem (as both can work with librealsense but are not officially validated to be used with it yet).  Could you try building librealsense and pyrealsense2 together from source code using the instructions in the link below, please?

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

wikpunGitHub, 2020-09-16:
Hi Marty:

Thanks for your help! I found out that it is the USB cable which causes
issue. My robot has very limited space and the cable is blended 90 degrees
for a very long time and it stops working. I built my robot 9 months ago
and I work on my desktop for the code. When I tried my new code on the
robot that I didn't use for 9 month, the realsense camera stop working.

Do you know where can I find the specification of the USB type C cable?
There are so many different configuration. I just bought one and it is not
working.

Thanks!

Best Regards,

Wik

MartyG-RealSense, 2020-09-16:
Great to hear that you made progress!

You can purchase USB cables that have a 90 degree angled connector" to help resolve problems caused by lack of insertion room with the typical straight connectors.  

A company called **Newnex** are the USB equipment supplier who are officially recommended by Intel in the data sheet document for the 400 Series cameras as a supplier for USB-C cables for use with RealSense, so you can have confidence that Newnex cables will work with your camera.  

Newnex can supply "industrial-grade" premium quality cables and have cables available with L-connectors and screw-locks for securing the cable firmly into the side of the camera via two small holes either side of the camera's micro-USB port.

https://www.newnex.com/realsense-3d-camera-connectivity.php
https://www.youtube.com/watch?v=GLQgR1jT04M

MartyG-RealSense, 2020-09-26:
Hi @wikpunGitHub   Do you require further assistance with this case, please?  Thanks!

wikpunGitHub, 2020-09-26:
Hi Marty:

No, thanks!

Best Regards,

Wik

closed 2020-09-26 by MartyG-RealSense